Nic30/hdlConvertor #1: Verilog parse causes Exeption: Variable has no type

amal-khailtash, 2016-08-10:
Using the example uart.v in tests directory causes this error:

```
>>> res = hdlConvertor.parse("uart.v", "verilog")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
Exception: Variable has no type

```

Nic30, 2016-08-10:
Yes,  parsing interface types in module body was disabled because there was problem with parser 
(which is now fixed).

I will fix it on Friday.

Nic30, 2016-08-14:
@amal-khailtash it is fixed but some of verilog rules are skipped (but now you can parse probably any module declaration).
I do not have much tests for verilog (https://github.com/Nic30/hwtLib/tree/master/hwtLib/tests).
Try if my fix works for you and I can close this issue.

Nic30, 2016-08-14:
Also in ANTLR cache is not working as it is supposed to work. 
This makes this parser ~10000x slower, but it is still usable (this will be fixed next week).

amal-khailtash, 2016-08-16:
This is working on the uart.v file.  I cannot use it on my own files yet for I do not have a good verilog preprocessor.  Thanks for the fix.  Hopefully SV and preprocessor support is coming.

Nic30, 2016-08-16:
Great! 

About preprocessors, SV etc.:
I would like, it is relatively easy, but I have no time to spare and it is getting worse day by day. 
If someone can help me with SV or anything else it would be great.

closed 2016-08-16 by Nic30