Nic30/hdlConvertor #2: New ANTLR C++ release

amal-khailtash, 2017-02-24:
I have not had any luck compiling with the latest ANTLR C++ runtime release.  Any chance you have time to update to latest ANTLR version?

Nic30, 2017-02-25:
Hello,
yes, but this weekend I have quite lot of work.

I will take look at it at 28. 3.
But any cooperation would be appreciated.

(And version in git contains buggy ANTRL that extremely slows down parsing,
this bug is already fixed.)

Thomasb81, 2017-02-25:
Hello

You can have look to my branch: https://github.com/Thomasb81/hdlConvertor/tree/devel
I already migrate to antlr 4.6 c++ runtime + personal patch submit 
Unfortunately I made other change : 
* support of python 2.7
* change in the build system to use cmake instead of python build system (the build system in place was a pain to use at least for me)

The initial wish was to add a verilog preprocessor... so far work is in progress.
I thought to do a push request once the job is mature enough.

Thomas

amal-khailtash, 2017-02-25:
I will try to contribute if I get some time.  Have you considered using CFFI instead? Maybe have ANTLR runtime compiled separately and use CFFI to make the python bindings.

Nic30, 2017-02-26:
@Thomasb81 I did not made python2.7 support because I was thinking nobody would use it.
I see I was wrong. I try to merge ours branches.

Nic30, 2017-02-26:
@amal-khailtash hdlConvertor uses python integers (we need arbitrary precision). I do not now if CFFI can pass these objects into python like integers and not like PyObject structure?

Thomasb81, 2017-02-26:
@Nic30 

It is quite complicate to move legacy code from Python2.7 to Python3 specially in front of a manager that would see added value from one side and cost time in another.
But I think everybody agree, since the release of python 3.5 the trend is to move to python 3. 
So for the moment if support of python 2.7 is possible at reasonable cost. It think it is good to have it.
cython ease this task.

It is not ready to merge. But I take feedbacks right now :)

Nic30, 2017-02-27:
I updated ANTRL to newest version available, then I found out that  master of ANTRL is not working
https://travis-ci.org/antlr/antlr4/builds
But it seems to be working just getChannelNames needs override in lexers.
Without it it ends up with 

```
g++ -std=c++1y -I"/home/nic30/Documents/workspace/hdlConvertor/src/antlr4-runtime-cpp" -I/usr/include/python3.5dm -O0 -g3 -Wall -c -fmessage-length=0 -MMD -MP -MF"src/convertor.d" -MT"src/convertor.o" -o "src/convertor.o" "../src/convertor.cpp"
In file included from ../src/convertor.h:24:0,
                 from ../src/convertor.cpp:1:
../src/parserContainer.h: In instantiation of ‘void ParserContainer<antlrLexerT, antlrParserT, hdlParserT>::initParser(antlr4::ANTLRFileStream*) [with antlrLexerT = vhdl::vhdlLexer; antlrParserT = vhdl::vhdlParser; hdlParserT = DesignFileParser]’:
../src/parserContainer.h:62:13:   required from ‘ParserErrors ParserContainer<antlrLexerT, antlrParserT, hdlParserT>::parseFile(const char*, bool, bool, std::function<void(antlrParserT*, hdlParserT*)>) [with antlrLexerT = vhdl::vhdlLexer; antlrParserT = vhdl::vhdlParser; hdlParserT = DesignFileParser]’
../src/convertor.cpp:42:67:   required from here
../src/parserContainer.h:37:9: error: invalid new-expression of abstract class type ‘vhdl::vhdlLexer’
   lexer = new antlrLexerT(input); 
...
```

Question is if this problem is caused by bug in antrl or by something else.

Nic30, 2017-02-27:
Ok, now hdlConverotor uses antrl4.7 it seems to be working but I did run only few tests.

Nic30, 2017-02-27:
It seems to be working except there is another bug in antrl.
(I have bad feeling that today was wrong day to do such a thing. ... https://travis-ci.org/antlr/antlr4/builds)
Lexer gets lost in some verilog modules.
I do not know If anyone from ANTRL knows about this issue can anyone post issue there? 
For example with file  tests/uart.v.
Or can anyone find out what is causing the issue? 

I double checked the grammar it should work and it was working.
And it is nearly the same grammar which they have in examples.

Thomasb81, 2017-02-27:
I use the official Antlrv4.6 release
http://www.antlr.org/download/antlr-4.6-complete.jar
http://www.antlr.org/download/antlr4-cpp-runtime-4.6-source.zip
There is some issue preventing the tool to work with the runtime : https://github.com/antlr/antlr4/issues/1620
Did not face any issue running tests/uart.v

Nic30, 2017-02-28:
I did used last version of master (commit b2012d9)

tests/uart.v works like this:

```
56:9:Error:no viable alternative at input 'if (reset)'
56:9:Error:mismatched input ')' expecting {'#', Escaped_identifier, Simple_identifier}
57:16:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
58:16:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
59:16:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
60:16:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
61:16:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
62:16:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
63:16:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
64:16:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
65:16:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
66:16:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
69:8:Error:extraneous input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
69:16:Error:mismatched input ';' expecting '('
70:8:Error:extraneous input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
70:16:Error:mismatched input ';' expecting '('
72:2:Error:extraneous input 'if' expecting {'endmodule', '(', 'defparam', 'localparam', 'integer', 'real', 'realtime', 'time', 'parameter', 'specparam', 'inout', 'input', 'output', 'reg', 'event', 'genvar', 'trireg', 'supply0', 'supply1', 'tri', 'triand', 'trior', 'tri0', 'tri1', 'wire', 'wand', 'wor', 'function', 'task', 'pulldown', 'pullup', 'cmos', 'rcmos', 'bufif0', 'bufif1', 'notif0', 'notif1', 'nmos', 'pmos', 'rnmos', 'rpmos', 'and', 'nand', 'or', 'nor', 'xor', 'xnor', 'buf', 'not', 'tranif0', 'tranif1', 'rtranif1', 'rtranif0', 'tran', 'rtran', 'generate', 'assign', 'initial', 'always', 'specify', Escaped_identifier, Simple_identifier}
72:17:Error:mismatched input ')' expecting {'#', Escaped_identifier, Simple_identifier}
73:13:Error:extraneous input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
73:22:Error:mismatched input ';' expecting '('
74:13:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
77:15:Error:mismatched input ')' expecting {'#', Escaped_identifier, Simple_identifier}
79:17:Error:mismatched input '&&' expecting {'#', Escaped_identifier, Simple_identifier}
79:26:Error:mismatched input ')' expecting {'#', Escaped_identifier, Simple_identifier}
80:20:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
81:20:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
82:20:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
85:15:Error:mismatched input ')' expecting {'#', Escaped_identifier, Simple_identifier}
86:21:Error:extraneous input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
86:38:Error:missing '(' at '+'
86:41:Error:missing ')' at ';'
88:7:Error:extraneous input 'if' expecting {'endmodule', '(', 'defparam', 'localparam', 'integer', 'real', 'realtime', 'time', 'parameter', 'specparam', 'inout', 'input', 'output', 'reg', 'event', 'genvar', 'trireg', 'supply0', 'supply1', 'tri', 'triand', 'trior', 'tri0', 'tri1', 'wire', 'wand', 'wor', 'function', 'task', 'pulldown', 'pullup', 'cmos', 'rcmos', 'bufif0', 'bufif1', 'notif0', 'notif1', 'nmos', 'pmos', 'rnmos', 'rpmos', 'and', 'nand', 'or', 'nor', 'xor', 'xnor', 'buf', 'not', 'tranif0', 'tranif1', 'rtranif1', 'rtranif0', 'tran', 'rtran', 'generate', 'assign', 'initial', 'always', 'specify', Escaped_identifier, Simple_identifier}
88:25:Error:mismatched input '==' expecting {'#', Escaped_identifier, Simple_identifier}
89:21:Error:mismatched input '==' expecting {'#', Escaped_identifier, Simple_identifier}
89:38:Error:mismatched input '==' expecting {'#', Escaped_identifier, Simple_identifier}
90:20:Error:mismatched input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
92:19:Error:extraneous input '<=' expecting {'#', Escaped_identifier, Simple_identifier}
92:29:Error:missing '(' at '+'
92:32:Error:missing ')' at ';'
94:12:Error:extraneous input 'if' expecting {'endmodule', '(', 'defparam', 'localparam', 'integer', 'real', 'realtime', 'time', 'parameter', 'specparam', 'inout', 'input', 'output', 'reg', 'event', 'genvar', 'trireg', 'supply0', 'supply1', 'tri', 'triand', 'trior', 'tri0', 'tri1', 'wire', 'wand', 'wor', 'function', 'task', 'pulldown', 'pullup', 'cmos', 'rcmos', 'bufif0', 'bufif1', 'notif0', 'notif1', 'nmos', 'pmos', 'rnmos', 'rpmos', 'and', 'nand', 'or', 'nor', 'xor', 'xnor', 'buf', 'not', 'tranif0', 'tranif1', 'rtranif1', 'rtranif0', 'tran', 'rtran', 'generate', 'assign', 'initial', 'always', 'specify', Escaped_identifier, Simple_identifier}
94:23:Error:mismatched input '>' expecting {'#', Escaped_identifier, Simple_identifier}
94:37:Error:mismatched input '<' expecting {'#', Escaped_identifier, Simple_identifier}
95:20:Error:extraneous input '[' expecting {'#', Escaped_identifier, Simple_identifier}
95:28:Error:missing '(' at '-'
95:31:Error:mismatched input ']' expecting ')'
96:12:Error:extraneous input 'end' expecting {'endmodule', '(', 'defparam', 'localparam', 'integer', 'real', 'realtime', 'time', 'parameter', 'specparam', 'inout', 'input', 'output', 'reg', 'event', 'genvar', 'trireg', 'supply0 [... cut ...]
                                       'value': None}},
                         {'direction': 'OUT',
                          'variable': {'name': 'rx_empty',
                                       'type': {'literal': {'type': 'ID',
                                                            'value': 'wire'}},
                                       'value': None}}]}],
 'imports': [],
 'packageHeaders': [],
 'packages': [],
 'variables': []}
```

= Interface of module is parsed but lexer gets lost in always statement.

Thomasb81, 2017-02-28:
I just update to give a trial and  get the same issue than you ...

If you run the demo from Cpp runtime. You will also notice "token recognition error" and "mismatched input" message too.

Yes it look line Cpp antlr runtime is currently broken.

Nic30, 2017-02-28:
I have hotfixed similar issues in antrl 4.5.3. It is caused cache like stucture in lexer.
Now I see this part was changed but bug is still there.
But problem is somewhere else I never had time to push tests into antrl test suite.
@mike-lischke maybe knows.

mike-lischke, 2017-02-28:
The demo only fails because I'm using Unicode input and my hack was removed in favor of the new (and much better) Unicode handling in ANTLR. This only requires a little [grammar change](mike-lischke/antlr4/[e368fb30ea3d801e78d89fcb0a0c50d4c5438e21]) and is not related to your problem here.

Thomasb81, 2017-03-01:
Ok
I track and try the various change and observe that before the following commit: the test is working as expected . And after those commit. The lexer failed and rise illegal error.
e1fb0b68c7c474b7c2429141e09ecc96aa747493 (C++ implementation of the LR-loop optimization.)
218ae7360858163826c53c0bce73e13c8bb131b3 (Make return states the same size as the parent array.)

mike-lischke, 2017-03-01:
Can you make a test case I can use to debug (e.g. change the demo app) and file an issue against ANTLR? I'll look at that then.

Thomasb81, 2017-03-01:
@mike-lischke  Hi Mike,

Is it ok to provide a testcase with the complete grammar ? 
The grammar is quite large (https://github.com/antlr/grammars-v4/blob/master/verilog/Verilog2001.g4) ? If not it will be difficult I think... 
I can probably try to reduce the input file trigging the issue.

mike-lischke, 2017-03-01:
No, grammar doesn't really matter. Just specify what you expect to come out for some simple input and what fails actually.

Thomasb81, 2017-03-01:
Testcase attach to following issue : https://github.com/antlr/antlr4/issues/1708
Thomas

closed 2017-07-03 by Nic30