Nic30/hdlConvertor #5: separate antrl4 from hdlConvertor, question to users

Nic30, 2017-07-03:
Hello,

ANTRL4 should always be separated from hdlConvertor but no package in ubuntu/respository existed.

Now when I see someone maybe using this I can create packages into linux respositories and also make an update.

Who does use this library?
On which distribution or system?

@stefan-titus-keysight 
@amal-khailtash
@Thomasb81

Thomasb81, 2017-07-03:
Hi

What do you suggest exactly ? 
I use this library. What would be useful would be to package antlrv4 c++ runtime and runtime-dev or provide script and procedure instead to recompile it. 
I use ubuntu.

Nic30, 2017-07-04:
I mean:

- antr4-dev package in ubuntu repository
- use it in hdlConvertor
- update hdlConvertor to use newest antrl

stefan-titus-keysight, 2017-07-05:
Hello,

We are not yet using this library, but the feature that requires it is coming up in the next month or so. 

I agree, ANTLR4 should be separate on principle. Our platforms are Windows (primary) and several flavors of Linux, so simply adding ANTLR4 to the Ubuntu package repo won't solve the problem for us. We will likely deal with this dependency in-house, so no need to make any accommodations for us.

Best regards,
--Stefan

Thomasb81, 2017-07-31:
What about using using CMake and using Cython (by the way it allows to easily support python2.7) to build this library ?
Antlr cpp runtime come already with a CMake infrastructure. It allows to generate Makefile in order to compile only modified file and not all file each time as current build system (not compatible as is with python2.7) ... specially when you compile the sv parser.

Nic30, 2017-08-01:
@Thomasb81 I would like to use cmake whole time.
Do you know if cmake can be called from disutils (setup.py basically) ?
Currently I do not know how to have pip compatible package which is build by cmake?
Is it possible, do you know about any example?

Thomasb81, 2017-08-02:
Not very familiar to pip, but we can make a trial with https://bitbucket.org/renficiaud/cmake-pip

amal-khailtash, 2017-08-03:
I am quite familiar with pip.  What exactly do you want to do?  Compile ANTLR runtime as a Python extension?

amal-khailtash, 2017-08-03:
Here is a project that uses cmake as well as cython.

https://github.com/symengine/symengine.py

This might be a better option that adds cmake capability to setuptools.

Nic30, 2017-08-13:
cca next week this repo will be updated with new ANTRL
Build will by managed by python and cmake will be used.
New package will also appear in pip repository.
And I at least try to write some description and documentation.

closed 2017-08-13 by Nic30