Nic30/hdlConvertor #7: g++ FATAL error

mohit162009, 2017-10-27:
Hello, I am not able to install. When I try to run the command: 
python3 hdlConvertor-master/hdlConvertor-master/setup.py install  

I get this error:
running install
running build
running build_ext
building 'hdlConvertor' extension
g++ -pthread -shared -o build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so
g++: fatal error: no input files
compilation terminated.

My gcc version- 5.4.0
Python version-3.5.1

I am not able to debug the issue. Could you please help?

Thanks,
Mohit

Nic30, 2017-10-27:
Hello,
my fault, https://github.com/Nic30/hdlConvertor/blob/master/setup.py cpp files are discovered by relative path, 

You can execute setup.py in folder where it is placed
or use pip to install hdlConvertor.

If I were you I would use pip3 install hdlConvertor. There will be plenty of new things (new ANTLR, tcl parser etc...). I just finding time to commit it and write some tests.

mohit162009, 2017-10-30:
Hello,
Thanks for your quick response. When I do pip3 install I get following error 

`~/hdlConvertor-master 708$ pip3 install hdlConvertor
Collecting hdlConvertor
  Using cached hdlConvertor-1.0.tar.gz
Installing collected packages: hdlConvertor
  Running setup.py install for hdlConvertor
    Complete output from command /home/nxf36037/3.5.1/x86_64-linux/bin/python3.5 -c "import setuptools, tokenize;__file__='/tmp/pip-build-mq70i2ys/hdlConvertor/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-ihewn8sx-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_clib
    building 'antlr4' library
    creating build
    creating build/temp.linux-x86_64-3.5
    ar rc build/temp.linux-x86_64-3.5/libantlr4.a
    running build_ext
    building 'hdlConvertor' extension
    creating build/temp.linux-x86_64-3.5/src
    creating build/temp.linux-x86_64-3.5/src/VhdlParser
    creating build/temp.linux-x86_64-3.5/src/hdlObjects
    creating build/temp.linux-x86_64-3.5/src/SVParser
    creating build/temp.linux-x86_64-3.5/src/baseHdlParser
    creating build/temp.linux-x86_64-3.5/src/vhdlConvertor
    creating build/temp.linux-x86_64-3.5/src/svConverter
    creating build/temp.linux-x86_64-3.5/src/VerilogParser
    creating build/temp.linux-x86_64-3.5/src/verilogConvertor
    creating build/temp.linux-x86_64-3.5/src/antlr4-runtime-cpp
    creating build/temp.linux-x86_64-3.5/src/antlr4-runtime-cpp/dfa
    creating build/temp.linux-x86_64-3.5/src/antlr4-runtime-cpp/misc
    creating build/temp.linux-x86_64-3.5/src/antlr4-runtime-cpp/atn
    creating build/temp.linux-x86_64-3.5/src/antlr4-runtime-cpp/support
    creating build/temp.linux-x86_64-3.5/src/antlr4-runtime-cpp/tree
    creating build/temp.linux-x86_64-3.5/src/antlr4-runtime-cpp/tree/xpath
    creating build/temp.linux-x86_64-3.5/src/antlr4-runtime-cpp/tree/pattern
    gcc -pthread -fno-strict-aliasing -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Isrc/antlr4-runtime-cpp/ -I/home/nxf36037/3.5.1/x86_64-linux/include/python3.5m -c src/syntaxErrorLogger.cpp -o build/temp.linux-x86_64-3.5/src/syntaxErrorLogger.o -std=c++11
    gcc -pthread -fno-strict-aliasing -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Isrc/antlr4-runtime-cpp/ -I/home/nxf36037/3.5.1/x86_64-linux/include/python3.5m -c src/hdlConvertorModule.cpp -o build/temp.linux-x86_64-3.5/src/hdlConvertorModule.o -std=c++11
    cc1plus: error: unrecognized command line option "-std=c++11"
    gcc -pthread -fno-strict-aliasing -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Isrc/antlr4-runtime-cpp/ -I/home/nxf36037/3.5.1/x86_64-linux/include/python3.5m -c src/notImplementedLogger.cpp -o build/temp.linux-x86_64-3.5/src/notImplementedLogger.o -std=c++11
    cc1plus: error: unrecognized command line option "-std=c++11"
    gcc -pthread -fno-strict-aliasing -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -Isrc/antlr4-runtime-cpp/ -I/home/nxf36037/3.5.1/x86_64-linux/include/python3.5m -c src/main.cpp -o build/temp.linux-x86_64-3.5/src/main.o -std=c++11
    cc1plus: error: unrecognized command line option "-std=c++11"
    cc1plus: error: unrecognized command line option "-std=c++11"
    error: command 'gcc' failed with exit status 1

    ----------------------------------------
Command "/home/nxf36037/3.5.1/x86_64-linux/bin/python3.5 -c "import setuptools, tokenize;__file__='/tmp/pip-build-mq70i2ys/hdlConvertor/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-ihewn8sx-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-mq70i2ys/hdlConvertor
`

Could you please help me here?

mohit162009, 2017-10-30:
Also, I want to know whether the  VHDL parser is capable of fetching out the values of constant from a package like the following:

`package example is
 constant MAX_NLBIST 		      : natural := 128;--changed from 16;
 constant MAX_NMCUT 		      : natural := 512;--changed from 96
 constant MAX_NRAM		      : natural := 512;--changed from 96
 constant MAX_NROM		      : natural := 512;--changed from 96
 -- Defines the reset value for each register
 constant STCU_MB_CTRL_RES  : mb_res_reg 
          := (
	    X"00000000", -- Index 88
	    X"00000000", -- Index 87
            X"00000000", -- Index 86
            X"00000000", -- Index 85
            X"00000000", -- Index 84
	    X"00000000", -- Index 83
            X"00000000", -- Index 82
            X"00000000", -- Index 81
            X"00000000", -- Index 80
            X"00000000", -- Index 79
	        X"00000000", -- Index 78
            X"00000000", -- Index 77
            X"00000000", -- Index 76
            X"00000000", -- Index 75
            X"00000000", -- Index 74
	        X"00000000", -- Index 73
            X"00000000", -- Index 72
            X"00000000", -- Index 71
            X"00000000", -- Index 70
            X"00000000", -- Index 69
            X"00000000", -- Index 68
            X"00000000", -- Index 67
            X"00000000", -- Index 66
            X"00000000", -- Index 65
	        X"00000000", -- Index 64
            X"00000000", -- Index 63
            X"00000000", -- Index 62
            X"00000000", -- Index 61
	        X"00000000", -- Index 60
            X"00000000", -- Index 59
            X"00000000", -- Index 58
            X"00000000", -- Index 57
	        X"00000000", -- Index 56
            X"00000000", -- Index 55
            X"00000000", -- Index 54
            X"00000000", -- Index 53
	        X"00000000", -- Index 52
            X"00000000", -- Index 51
            X"00000000", -- Index 50
            X"00000000", -- Index 49
	        X"00000000", -- Index 48
            X"00000000", -- Index 47
            X"00000000", -- Index 46
            X"00000000", -- Index 45
            X"00000000", -- Index 44
	        X"00000000", -- Index 43
            X"00000000", -- Index 42
            X"00000000", -- Index 41
            X"00000000", -- Index 40
            X"00000000", -- Index 39
	        X"00000000", -- Index 38
            X"00000000", -- Index 37
            X"00000000", -- Index 36
            X"00000000", -- Index 35
            X"00000000", -- Index 34
            X"00000000", -- Index 33
            X"00000000", -- Index 32
            X"00000000", -- Index 31
            X"00000000", -- Index 30
	        X"00000000", -- Index 29
[... 188 lines cut ...]
            X"00000000", -- Index 7
            X"00000000", -- Index 6
            X"00000000", -- Index 5
            X"00000000", -- Index 4
	        X"00000000", -- Index 3
            X"00000000", -- Index 2
            X"00000000", -- Index 1
            X"00000000"	 -- Index 0
          );

  component demo is
    generic (
      GENERIC1: boolean := false;
      GENERIC2: integer := 100
    );
    port (
      a, b : in std_ulogic := '1';
      c, d : out std_ulogic_vector(7 downto 0);
      e, f : inout unsigned(7 downto 0)
    );
  end component;
end package;
`

I want the full value s of constants here. Thanks,

mohit162009, 2017-10-30:
Hello, I was able to install it, Could you please help me with the above query? Thanks a lot.

Nic30, 2017-10-30:
Hello, yes.
But i have to finish something else first, it will take ~8 hours until i look into it.
As far as I know it should not be a problem, but in https://github.com/Nic30/hdlConvertor/blob/master/src/vhdlConvertor/packageParser.cpp#L36 I did not implement constant processing (it is parsed and there is no conversion to output for python), but it is very easy. I will implement it for you at night, I did not need that functionality so I did not noticed.

And please keep issues separated.

mohit162009, 2017-10-30:
Hello,
I need to parse constant-arrays. I am facing issues here. Also, do you support hierarchical vhdl files. I mean like if one file is using (`use example_pkg.all;`) and I parse the file, Can I get variables declared in the file example_pkg. Because I have so many cases like this. Could you please guide me here?

Nic30, 2017-10-30:
```python
res = hdlConvertor.parse(fileName, "vhdl", debug=True)
print(res["imports"])
```
You can read imports as well, external symbols are not required for parsing.
You will get HdlRef for each symbol. You can then search this symbol in imports.
I had SW which does this thing https://github.com/Nic30/hwtHdlParsers/tree/master/hwtHdlParsers/ but it is not maintained and not working anymore.

Nic30, 2017-10-30:
@mohit162009 now there should be working version in the git

output should look like
```
{
... 
 'packageHeaders': [{ 
 ....
                    'variables': [{'name': 'MAX_NLBIST',
                                    'type': {'literal': {'type': 'ID',
                                                         'value': 'natural'}},
                                    'value': {'literal': {'type': 'INT',
                                                          'value': 128}}},    
                               {'name': 'STCU_MB_CTRL_RES',
                                    'type': {'literal': {'type': 'ID',
                                                         'value': 'mb_res_reg'}},
                                    'value': {'literal': {'type': 'ARRAY',
                                                          'value': [{'bits': 32,
                                                                     'type': 'INT',
                                                                     'value': 0},
                                                                    {'bits': 32,
                                                                     'type': 'INT',
                                                                     'value': 0},
....
```

Check If this works for you and if so, I will upload it to PIP

Nic30, 2017-10-30:
Also next time put distinct issues to separate github issue please.

mohit162009, 2017-10-31:
Thanks a lot, I will try this.

mohit162009, 2017-10-31:
Hello Reagrding "imports", If I have a file like this - 

```
use example_pkg.all;
use abc.example_pkg_2.all;
```

I want to extract : example_pkg and abc.example_pkg_2

With 
```
res = hdlConvertor.parse(fileName, "vhdl", debug=True)
print(res["imports"])
```
I get 
```
{'binOperator': {'op0': {'literal': {'value': 'example_pkg', 'type': 'ID'}}, 'operator': 'DOT', 'op1': {'literal': {'value': None, 'type': 'ALL'}}}}
{'binOperator': {'op0': {'binOperator': {'op0': {'literal': {'value': 'abc', 'type': 'ID'}}, 'operator': 'DOT', 'op1': {'literal': {'value': 'example_pkg_2', 'type': 'ID'}}}}, 'operator': 'DOT', 'op1': {'literal': {'value': None, 'type': 'ALL'}}}}
```

I am trying something like:
```
for e in res['imports']:
  for b in e['binOperator']:
      print(b[0])
```
But I am unable to fetch the desired output. Could you please help?

Nic30, 2017-10-31:
Are you new to python?

``python

a = {'binOperator': {'op0': {'literal': {'value': 'example_pkg', 'type': 'ID'}}, 'operator': 'DOT', 'op1': {'literal': {'value': None, 'type': 'ALL'}}}}
b = {'binOperator': {'op0': {'binOperator': {'op0': {'literal': {'value': 'abc', 'type': 'ID'}}, 'operator': 'DOT', 'op1': {'literal': {'value': 'example_pkg_2', 'type': 'ID'}}}}, 'operator': 'DOT', 'op1': {'literal': {'value': None, 'type': 'ALL'}}}}

def getPath(item):
    try:
        binOp = item['binOperator']
    except KeyError:
        # is literal
        lit = item['literal']
        return [lit['value'], ]
    op0 = getPath(binOp['op0'])
    op1 = getPath(binOp['op1'])

    partsOfPath = op0
    partsOfPath.extend(op1)

    return partsOfPath

for item in [a, b]:
    p = getPath(item)
    p  = p[:len(p)-1] # cut off "all"
    print(p)

```

result:
```
['example_pkg']
['abc', 'example_pkg_2']
```

closed 2017-11-03 by Nic30