Nic30/hdlConvertor #10: Error in constant arrays

mohit162009, 2017-11-01:
If I have a constant array like this:
```
  CONSTANT XYZ  : xyz_ld_shift_state_value_type          
          := (  
            "11111", -- Index 31 NOT_USED_13
	    "11110", -- Index 30 NOT_USED_12
            "11101", -- Index 29 NOT_USED_11
	    "11100", -- Index 28 NOT_USED_10
            "11011", -- Index 27 NOT_USED_9
            "11010", -- Index 26 NOT_USED_8
            "11001", -- Index 25 NOT_USED_7
            "11000", -- Index 24 NOT_USED_6
            "10111", -- Index 23 NOT_USED_5
	    "10110", -- Index 22 NOT_USED_4
            "10101", -- Index 21 NOT_USED_3
            "10100", -- Index 20 NOT_USED_2
	    "10011", -- Index 19 NOT_USED_1
            "10010", -- Index 18 SEL_MB_RUN_TEST_IDLE
            "10001", -- Index 17 SHIFT_DR_MB_REGS
            "10000", -- Index 16 SEL_DR_MB_REGS 
            "01111", -- Index 15 SEL_DR_MB_BRANCH
	    "01110", -- Index 14 SHIFT_IR_MB_REGS
            "01101", -- Index 13 SEL_IR_MB_BRANCH
	    "01100", -- Index 12 SEL_IR_MB_REGS
            "01011", -- Index 11 LBIST_SEL_NEXT
            "01010", -- Index 10 LD_LB_REGS
            "01001", -- Index 9  WAIT_LBIST_UPD_TIME
            "01000", -- Index 8  SHIFT_DATA_LB_REGS
            "00111", -- Index 7  SEL_DATA_LB_REGS
	    "00110", -- Index 6  SHIFT_DSEL_LB_REGS
            "00101", -- Index 5  SEL_DSEL_LB_REGS
            "00100", -- Index 4  END_LOAD_MB_CUTSEL
	    "00011", -- Index 3  LD_MB_STATUS
            "00010", -- Index 2  CHECK_MB_STATUS
            "00001", -- Index 1  LOAD_MB_CUTSEL
            "00000"  -- Index 0  IDLE
          );
```
I get this error:
```
bsub -I -q gui -P eda python3.5 ~/hdlbigfile.py
Job <673511> is submitted to queue <gui>.
<<Waiting for dispatch ...>>
<<Starting on lc2407>>
*** glibc detected *** python3.5: double free or corruption (fasttop): 0x000000000953a4c0 ***
======= Backtrace: =========
/lib64/libc.so.6[0x3439c7247f]
/lib64/libc.so.6(cfree+0x4b)[0x3439c728db]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZNSt6vectorI6SymbolSaIS0_EE19_M_emplace_back_auxIJRKS0_EEEvDpOT_+0x109)[0x2b65ab158289]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN4Expr5ARRAYESt6vectorIPS_SaIS1_EE+0xdf)[0x2b65ab1580ef]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN10ExprParser14visitAggregateEPN4vhdl10vhdlParser16AggregateContextE+0x156)[0x2b65ab15f6e6]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN10ExprParser11visitFactorEPN4vhdl10vhdlParser13FactorContextE+0x16)[0x2b65ab15f856]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN10ExprParser9visitTermEPN4vhdl10vhdlParser11TermContextE+0x3b)[0x2b65ab15f95b]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN10ExprParser22visitSimple_expressionEPN4vhdl10vhdlParser24Simple_expressionContextE+0x3b)[0x2b65ab15faab]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN10ExprParser21visitShift_expressionEPN4vhdl10vhdlParser23Shift_expressionContextE+0x1a)[0x2b65ab15ff4a]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN10ExprParser13visitRelationEPN4vhdl10vhdlParser15RelationContextE+0x1a)[0x2b65ab16005a]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN10ExprParser15visitExpressionEPN4vhdl10vhdlParser17ExpressionContextE+0x3b)[0x2b65ab16018b]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN15InterfaceParser16extractVariablesEPN4vhdl10vhdlParser22Identifier_listContextEPNS1_25Subtype_indicationContextEPNS1_17ExpressionContextE+0x58)[0x2b65ab1608f8]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN19PackageHeaderParser29visitPackage_declarative_itemEPN4vhdl10vhdlParser31Package_declarative_itemContextE+0xd6)[0x2b65ab161aa6]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN19PackageHeaderParser29visitPackage_declarative_partEP13PackageHeaderPN4vhdl10vhdlParser31Package_declarative_partContextE+0x33)[0x2b65ab161e43]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN19PackageHeaderParser24visitPackage_declarationEPN4vhdl10vhdlParser26Package_declarationContextE+0x41)[0x2b65ab161ec1]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN16DesignFileParser17visitPrimary_unitEPN4vhdl10vhdlParser19Primary_unitContextE+0x188)[0x2b65ab15ecb8]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN16DesignFileParser16visitDesign_fileEPN4vhdl10vhdlParser18Design_fileContextE+0x3b)[0x2b65ab15ee5b]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN9Convertor5parseEPKc6Languebb+0x4d7)[0x2b65aaeaf727]
/home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/site-packages/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(hdlConvertor_parse+0xf2)[0x2b65aaeb0042]
python3.5(PyCFunction_Call+0x109)[0x574a19]
python3.5(PyEval_EvalFrameEx+0x834f)[0x4de4bf]
python3.5[0x4df08c]
python3.5(PyEval_EvalCode+0x4d)[0x4df1ad]
python3.5(PyRun_FileExFlags+0xcb)[0x50aa2b]
[... 76 lines cut ...]
2b65ac21f000-2b65ac220000 rw-p 00002000 00:36 108969155                  /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/grp.cpython-35m-x86_64-linux-gnu.so
2b65ac220000-2b65ac260000 rw-p 2b65ac220000 00:00 0
2b65ac260000-2b65ac263000 r-xp 00000000 00:36 107665094                  /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/_bz2.cpython-35m-x86_64-linux-gnu.so
2b65ac263000-2b65ac462000 ---p 00003000 00:36 107665094                  /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/_bz2.cpython-35m-x86_64-linux-gnu.so
2b65ac462000-2b65ac464000 rw-p 00002000 00:36 107665094                  /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/_bz2.cpython-35m-x86_64-linux-gnu.so
2b65ac464000-2b65ac46a000 r-xp 00000000 00:36 91908086                   /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/_lzma.cpython-35m-x86_64-linux-gnu.so
2b65ac46a000-2b65ac66a000 ---p 00006000 00:36 91908086                   /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/_lzma.cpython-35m-x86_64-linux-gnu.so
2b65ac66a000-2b65ac66c000 rw-p 00006000 00:36 91908086                   /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/_lzma.cpython-35m-x86_64-linux-gnu.so
2b65ac66c000-2b65ac6ac000 rw-p 2b65ac66c000 00:00 0
2b65ac6ac000-2b65ac6af000 r-xp 00000000 00:36 91908092                   /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/_posixsubprocess.cpython-35m-x86_64-linux-gnu.so
2b65ac6af000-2b65ac8ae000 ---p 00003000 00:36 91908092                   /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/_posixsubprocess.cpython-35m-x86_64-linux-gnu.so
2b65ac8ae000-2b65ac8af000 rw-p 00002000 00:36 91908092                   /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/_posixsubprocess.cpython-35m-x86_64-linux-gnu.so
2b65ac8af000-2b65ac8b4000 r-xp 00000000 00:36 67714538                   /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/select.cpython-35m-x86_64-linux-gnu.so
2b65ac8b4000-2b65acab3000 ---p 00005000 00:36 67714538                   /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/select.cpython-35m-x86_64-linux-gnu.so
2b65acab3000-2b65acab5000 rw-p 00004000 00:36 67714538                   /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/select.cpython-35m-x86_64-linux-gnu.so
2b65acab5000-2b65acb16000 rw-p 2b65acab5000 00:00 0
2b65acb16000-2b65acb1e000 r-xp 00000000 00:36 89410166                   /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/math.cpython-35m-x86_64-linux-gnu.so
2b65acb1e000-2b65acd1e000 ---p 00008000 00:36 89410166                   /home/nxf36037/3.5.1/x86_64-linux/lib/python3.5/lib-dynload/math.cpython-35m-x86_64-linux-gnu.so
[... cut ...] ux/lib/python3.5/lib-dynload/math.cpython-35m-x86_64-linux-gnu.so
2b65acd20000-2b65ace43000 rw-p 2b65acd20000 00:00 0
7fff221c3000-7fff221e3000 rw-p 7ffffffde000 00:00 0                      [stack]
7fff221fd000-7fff22200000 r-xp 7fff221fd000 00:00 0                      [vdso]
ffffffffff600000-ffffffffffe00000 ---p 00000000 00:00 0                  [vsyscall]
/home/nxf36037/.lsbatch/1509563006.673511: line 8: 16304 Aborted                 (core dumped) python3.5 /home/nxf36037/hdlbigfile.py
```

Nic30, 2017-11-02:
ae768a76603e80ab50ceee6d106740349f7301f6 try now, I forgot to increment reference on list of values.

Nic30, 2017-11-02:
e2f2390  may help, I clearly see that there is an issue but I was not able to reproduce it wit GCC what compiler do you use?

Nic30, 2017-11-03:
58fa4f4c21ed95ec29e2c8ccb88f059c071c3bca fixed it, there were wrong deallocations and _ in bitstrings were not working under python 3.5. Try if it works for you, it works on Travis.

Nic30, 2017-11-13:
Helo, can you confirm that there is no issue anymore?

closed 2017-11-20 by Nic30