Nic30/hdlConvertor #9: Segmentation fault

mohit162009, 2017-11-01:
Hello,

Thanks for all the solutions, Its working now well.

But for vhdl files having ~1000 lines , I am getting the segmentation fault. I am not really sure why is this happening. Could you please help me here.

Nic30, 2017-11-01:
Hello, can you provide example input file please?

mohit162009, 2017-11-01:
[stcu_integr_pkg.txt](https://github.com/Nic30/hdlConvertor/files/1434067/stcu_integr_pkg.txt)
Hello, Please find attached sample file.

Nic30, 2017-11-01:
Hello,
can you send how did you exactly get that segfault? I tried to parse that file you have send me and it seems to be working. I do not have python recompiled with valgrind support so I can not confirm that there is nothing wrong.

Does it happen every time or just sometimes?

mohit162009, 2017-11-01:
Hello,
I am trying a lot of things, For some files if I remove ` package body`, it works. Sometimes I have to remove `constant` arrays and sometimes few `function`
I am not able to find out the exact reason and I am not sure what factors are causing this.

mohit162009, 2017-11-01:
Do the tool does any processing of vhdl file also apart from parsing, Such as if a function is called but not defined etc, Just thinking whether this can be an issue.

mohit162009, 2017-11-01:
This is the segmentation fault message:

```
/scratch/nxf36037 1231$ bsub -I -q gui -P eda python3.5 ~/hdlbigfile.py
Job <676121> is submitted to queue <gui>.
<<Waiting for dispatch ...>>
<<Starting on lc1485-zin45>>
/home/nxf36037/.lsbatch/1509563655.676121: line 8:  4264 Segmentation fault      (core dumped) python3.5 /home/nxf36037/hdlbigfile.py

<< Job finished, querying job status ... >>

<< To interrupt, press Ctrl-C. >>
```

I switched to bsub to get more RAM, I have also tried installing hdlConvertor in python and also downloaded and built it locally, Both ways it fails.

mohit162009, 2017-11-01:
I also tried valgrind and this is the output:

valgrind python3.5 ~/hdlbigfile.py
```
==6007== Memcheck, a memory error detector
==6007== Copyright (C) 2002-2009, and GNU GPL'd, by Julian Seward et al.
==6007== Using Valgrind-3.5.0 and LibVEX; rerun with -h for copyright info
==6007== Command: python3.5 /home/nxf36037/hdlbigfile.py
==6007==
==6007== Warning: noted but unhandled ioctl 0x5451 with no size/direction hints
==6007==    This could cause spurious value errors to appear.
==6007==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
==6007== Invalid read of size 4
==6007==    at 0x41BEBC: _PyObject_Free (obmalloc.c:1420)
==6007==    by 0x4828B6: tupledealloc (tupleobject.c:249)
==6007==    by 0x5674C2: code_dealloc (codeobject.c:365)
==6007==    by 0x4FA8B7: PyImport_ImportFrozenModuleObject (import.c:1225)
==6007==    by 0x4FA947: PyImport_ImportFrozenModule (import.c:1241)
==6007==    by 0x5054D7: import_init (pylifecycle.c:241)
==6007==    by 0x506927: _Py_InitializeEx_Private (pylifecycle.c:411)
==6007==    by 0x41E343: Py_Main (main.c:669)
==6007==    by 0x41B3A1: main (python.c:69)
==6007==  Address 0x4cc3020 is 16 bytes before a block of size 808 alloc'd
==6007==    at 0x4A0610C: malloc (vg_replace_malloc.c:195)
==6007==    by 0x41C1E1: _PyObject_Alloc (obmalloc.c:293)
==6007==    by 0x521354: _PyObject_GC_Alloc (gcmodule.c:1696)
==6007==    by 0x5214B0: _PyObject_GC_NewVar (gcmodule.c:1718)
==6007==    by 0x482135: PyTuple_New (tupleobject.c:104)
==6007==    by 0x4FBB1A: r_object (marshal.c:1181)
==6007==    by 0x4FC8BB: r_object (marshal.c:1347)
==6007==    by 0x4FD5C7: PyMarshal_ReadObjectFromString (marshal.c:1566)
==6007==    by 0x4FA754: PyImport_ImportFrozenModuleObject (import.c:1192)
==6007==    by 0x4FA947: PyImport_ImportFrozenModule (import.c:1241)
==6007==    by 0x5054D7: import_init (pylifecycle.c:241)
==6007==    by 0x506927: _Py_InitializeEx_Private (pylifecycle.c:411)
==6007==
==6007== Conditional jump or move depends on uninitialised value(s)
==6007==    at 0x41BEC5: _PyObject_Free (obmalloc.c:1420)
==6007==    by 0x567438: code_dealloc (codeobject.c:364)
==6007==    by 0x56F548: frame_dealloc (frameobject.c:433)
==6007==    by 0x4DCE23: PyEval_EvalFrameEx (ceval.c:4756)
==6007==    by 0x4DCE02: PyEval_EvalFrameEx (ceval.c:4754)
==6007==    by 0x4DCE02: PyEval_EvalFrameEx (ceval.c:4754)
==6007==    by 0x4DF08B: _PyEval_EvalCodeWithName (ceval.c:3966)
==6007==    by 0x4DF157: PyEval_EvalCodeEx (ceval.c:3987)
==6007==    by 0x570D5B: function_call (funcobject.c:632)
==6007==    by 0x42D947: PyObject_Call (abstract.c:2165)
==6007==    by 0x42F73F: _PyObject_CallMethodIdObjArgs (abstract.c:2423)
==6007==    by 0x4F9A3E: PyImport_ImportModuleLevelObject (import.c:1591)
==6007==
==6007== Use of uninitialised value of size 8
==6007==    at 0x41BED8: _PyObject_Free (obmalloc.c:1420)
==6007==    by 0x567438: code_dealloc (codeobject.c:364)
==6007==    by 0x56F548: frame_dealloc (frameobject.c:433)
==6007==    by 0x4DCE23: PyEval_EvalFrameEx (ceval.c:4756)
==6007==    by 0x4DCE02: PyEval_EvalFrameEx (ceval.c:4754)
==6007==    by 0x4DCE02: PyEval_EvalFrameEx (ceval.c:4754)
==6007==    by 0x4DF08B: _PyEval_EvalCodeWithName (ceval.c:3966)
==6007==    by 0x4DF157: PyEval_EvalCodeEx (ceval.c:3987)
==6007==    by 0x570D5B: function_call (funcobject.c:632)
==6007==    by 0x42D947: PyObject_Call (abstract.c:2165)
==6007==    by 0x42F73F: _PyObject_CallMethodIdObjArgs (abstract.c:2423)
==6007==    by 0x4F9A3E: PyImport_ImportModuleLevelObject (import.c:1591)
==6007==
==6007== Conditional jump or move depends on uninitialised value(s)
==6007==    at 0x41BEC5: _PyObject_Free (obmalloc.c:1420)
==6007==    by 0x4828B6: tupledealloc (tupleobject.c:249)
==6007==    by 0x5674C2: code_dealloc (codeobject.c:365)
==6007==    by 0x56F548: frame_dealloc (frameobject.c:433)
==6007==    by 0x4DCE23: PyEval_EvalFrameEx (ceval.c:4756)
[... 483 lines cut ...]
==6007==    by 0x4F9A3E: PyImport_ImportModuleLevelObject (import.c:1591)
==6007==    by 0x4D3FA8: builtin___import__ (bltinmodule.c:213)
==6007==    by 0x574A18: PyCFunction_Call (methodobject.c:98)
==6007==    by 0x42D947: PyObject_Call (abstract.c:2165)
==6007==    by 0x4D46D5: PyEval_CallObjectWithKeywords (ceval.c:4530)
==6007==    by 0x4D8623: PyEval_EvalFrameEx (ceval.c:2750)
==6007==    by 0x4DF08B: _PyEval_EvalCodeWithName (ceval.c:3966)
==6007==    by 0x4DF1AC: PyEval_EvalCode (ceval.c:3987)
==6007==    by 0x50AA2A: PyRun_FileExFlags (pythonrun.c:970)
==6007==    by 0x50AC0B: PyRun_SimpleFileExFlags (pythonrun.c:396)
==6007==    by 0x41EA9D: Py_Main (main.c:318)
==6007==    by 0x41B3A1: main (python.c:69)

Note: see also the FAQ in the source distribution.
It contains workarounds to several common problems.
In particular, if Valgrind aborted or crashed after
identifying problems in your program, there's a good chance
that fixing those problems will prevent Valgrind aborting or
crashing, especially if it happened in m_mallocfree.c.

If that doesn't help, please report this bug to: www.valgrind.org

In the bug report, send all the above text, the valgrind
version, and what OS and version you are using.  Thanks.
```

mohit162009, 2017-11-01:
I have googled a lot about this and I have found multiple times that Segmentation Fault can happen if C code is inserting a NULL element into a PyList_Object, Is there something like this in your project?

Nic30, 2017-11-02:
Yes, there are 3 places (grep -R "PyList_New"). But now I see I forget to increment reference on array values and they can be automatically deallocated under certain conditions. 
(**now it is fixed**)

Errors you see in valgrind, are not real, valgrind does not understand python code due optimizations.
https://stackoverflow.com/questions/20112989/how-to-use-valgrind-with-python

mohit162009, 2017-11-02:
I am still getting the error, I ran the test.py 

```
python3 test.py
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
PackageHeaderParser.visitType_declaration
ExprParser ement_association.choices
**Segmentation fault**
```

mohit162009, 2017-11-02:
And when I ran with the file for which I was getting segmentation fault, Now I get this

```
python3 test.py
PackageHeaderParser.visitAttribute_declaration
*** glibc detected *** python3: double free or corruption (fasttop): 0x000000000a3f2300 ***
======= Backtrace: =========
/lib64/libc.so.6[0x3b0f67247f]
/lib64/libc.so.6(cfree+0x4b)[0x3b0f6728db]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZNSt6vectorI6SymbolSaIS0_EE19_M_emplace_back_auxIJRKS0_EEEvDpOT_+0x109)[0x2af20ca7d2a9]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN4Expr5ARRAYESt6vectorIPS_SaIS1_EE+0xdf)[0x2af20ca7d10f]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN10ExprParser14visitAggregateEPN4vhdl10vhdlParser16AggregateContextE+0x156)[0x2af20ca84706]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN10ExprParser11visitFactorEPN4vhdl10vhdlParser13FactorContextE+0x16)[0x2af20ca84876]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN10ExprParser9visitTermEPN4vhdl10vhdlParser11TermContextE+0x3b)[0x2af20ca8497b]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN10ExprParser22visitSimple_expressionEPN4vhdl10vhdlParser24Simple_expressionContextE+0x3b)[0x2af20ca84acb]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN10ExprParser21visitShift_expressionEPN4vhdl10vhdlParser23Shift_expressionContextE+0x1a)[0x2af20ca84f6a]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN10ExprParser13visitRelationEPN4vhdl10vhdlParser15RelationContextE+0x1a)[0x2af20ca8507a]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN10ExprParser15visitExpressionEPN4vhdl10vhdlParser17ExpressionContextE+0x3b)[0x2af20ca851ab]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN15InterfaceParser16extractVariablesEPN4vhdl10vhdlParser22Identifier_listContextEPNS1_25Subtype_indicationContextEPNS1_17ExpressionContextE+0x58)[0x2af20ca85918]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN19PackageHeaderParser29visitPackage_declarative_itemEPN4vhdl10vhdlParser31Package_declarative_itemContextE+0xd6)[0x2af20ca86ac6]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN19PackageHeaderParser29visitPackage_declarative_partEP13PackageHeaderPN4vhdl10vhdlParser31Package_declarative_partContextE+0x33)[0x2af20ca86e63]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN19PackageHeaderParser24visitPackage_declarationEPN4vhdl10vhdlParser26Package_declarationContextE+0x41)[0x2af20ca86ee1]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN16DesignFileParser17visitPrimary_unitEPN4vhdl10vhdlParser19Primary_unitContextE+0x188)[0x2af20ca83cd8]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN16DesignFileParser16visitDesign_fileEPN4vhdl10vhdlParser18Design_fileContextE+0x3b)[0x2af20ca83e7b]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(_ZN9Convertor5parseEPKc6Languebb+0x4d7)[0x2af20c7d4727]
../build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so(hdlConvertor_parse+0x105)[0x2af20c7d5055]
python3(PyCFunction_Call+0x109)[0x574a19]
python3(PyEval_EvalFrameEx+0x818d)[0x4de2fd]
python3(PyEval_EvalFrameEx+0x6c93)[0x4dce03]
python3[0x4df08c]
python3(PyEval_EvalCode+0x4d)[0x4df1ad]
python3(PyRun_FileExFlags+0xcb)[0x50aa2b]
python3(PyRun_SimpleFileExFlags+0xec)[0x50ac0c]
python3(Py_Main+0xaae)[0x41ea9e]
python3(main+0x152)[0x41b3a2]
/lib64/libc.so.6(__libc_start_main+0xf4)[0x3b0f61d994]
python3[0x41b199]
======= Memory map: ========
00400000-0060c000 r-xp 00000000 00:2c 95392104                           /home/nxf36037/3.5.1/x86_64-linux/bin/python3
0080c000-0086f000 rw-p 0020c000 00:2c 95392104                           /home/nxf36037/3.5.1/x86_64-linux/bin/python3
0086f000-0089f000 rw-p 0086f000 00:00 0
095a0000-0a40d000 rw-p 095a0000 00:00 0                                  [heap]
3b0ee00000-3b0ee1c000 r-xp 00000000 08:02 1009082                        /lib64/ld-2.5.so
3b0f01c000-3b0f01d000 r--p 0001c000 08:02 1009082                        /lib64/ld-2.5.so
3b0f01d000-3b0f01e000 rw-p 0001d000 08:02 1009082                        /lib64/ld-2.5.so
3b0f600000-3b0f74f000 r-xp 00000000 08:02 1138615                        /lib64/libc-2.5.so
3b0f74f000-3b0f94e000 ---p 0014f000 08:02 1138615                        /lib64/libc-2.5.so
3b0f94e000-3b0f952000 r--p 0014e000 08:02 1138615                        /lib64/libc-2.5.so
3b0f952000-3b0f953000 rw-p 00152000 08:02 1138615                        /lib64/libc-2.5.so
3b0f953000-3b0f958000 rw-p 3b0f953000 00:00 0
3b0fa00000-3b0fa02000 r-xp 00000000 08:02 1138619                        /lib64/libdl-2.5.so
3b0fa02000-3b0fc02000 ---p 00002000 08:02 1138619                        /lib64/libdl-2.5.so
3b0fc02000-3b0fc03000 r--p 00002000 08:02 1138619                        /lib64/libdl-2.5.so
3b0fc03000-3b0fc04000 rw-p 00003000 08:02 1138619                        /lib64/libdl-2.5.so
3b0fe00000-3b0fe82000 r-xp 00000000 08:02 1138621                        /lib64/libm-2.5.so
3b0fe82000-3b10081000 ---p 00082000 08:02 1138621                        /lib64/libm-2.5.so
3b10081000-3b10082000 r--p 00081000 08:02 1138621                        /lib64/libm-2.5.so
[... cut ...]
2af20ce10000-2af20ce6b000 rw-p 00795000 00:2f 79462443                   /scratch/nxf36037/newhdltry/hdlConvertor/build/lib.linux-x86_64-3.5/hdlConvertor.cpython-35m-x86_64-linux-gnu.so
2af20ce6b000-2af20ce6e000 rw-p 2af20ce6b000 00:00 0
2af20ce6e000-2af20cfd6000 r-xp 00000000 00:19 91280931                   /pkg/.site/pkgs04/OSS-gcc-/5.4.0/x86_64-linux2.6-glibc2.5/lib64/libstdc++.so.6.0.21
2af20cfd6000-2af20d1d6000 ---p 00168000 00:19 91280931                   /pkg/.site/pkgs04/OSS-gcc-/5.4.0/x86_64-linux2.6-glibc2.5/lib64/libstdc++.so.6.0.21
2af20d1d6000-2af20d1e0000 r--p 00168000 00Abort
```

Nic30, 2017-11-02:
https://travis-ci.org/Nic30/hdlConvertor/builds/296413410

Problem is combination of python3.5 and gcc 5, at leas I know where to search

Nic30, 2017-11-03:
58fa4f4c21ed95ec29e2c8ccb88f059c071c3bca fixed it, there were wrong deallocations and _ in bitstrings were not working under python 3.5.

Nic30, 2017-11-13:
Helo, can you confirm that there is no issue anymore?

closed 2017-11-20 by Nic30