Nic30/hdlConvertor #12: Installation issue, 'cl.exe' failed with exit status 2

whitew1994WW, 2017-11-24:
Hi,

Thanks for uploading the package however I have been unable to install. I have downloaded the entire package and am trying to install using the command "python setup.py install" however I return the following error:

ime-cpp\BufferedTokenStream.cpp(387): warning C4273: 'antlr4::BufferedTokenStrea
m::getText': inconsistent dll linkage
c:\users\william\desktop\hdlconvertor-master\hdlconvertor-master\src\antlr4-runt
ime-cpp\BufferedTokenStream.h(91): note: see previous definition of 'getText'
C:\Users\william\Desktop\hdlConvertor-master\hdlConvertor-master\src/antlr4-runt
ime-cpp\BufferedTokenStream.cpp(391): warning C4273: 'antlr4::BufferedTokenStrea
m::getText': inconsistent dll linkage
c:\users\william\desktop\hdlconvertor-master\hdlconvertor-master\src\antlr4-runt
ime-cpp\BufferedTokenStream.h(92): note: see previous definition of 'getText'
C:\Users\william\Desktop\hdlConvertor-master\hdlConvertor-master\src/antlr4-runt
ime-cpp\BufferedTokenStream.cpp(399): warning C4273: 'antlr4::BufferedTokenStrea
m::fill': inconsistent dll linkage
c:\users\william\desktop\hdlconvertor-master\hdlconvertor-master\src\antlr4-runt
ime-cpp\BufferedTokenStream.h(95): note: see previous definition of 'fill'
C:\Users\william\Desktop\hdlConvertor-master\hdlConvertor-master\src/antlr4-runt
ime-cpp\BufferedTokenStream.cpp(410): warning C4273: 'antlr4::BufferedTokenStrea
m::InitializeInstanceFields': inconsistent dll linkage
c:\users\william\desktop\hdlconvertor-master\hdlconvertor-master\src\antlr4-runt
ime-cpp\BufferedTokenStream.h(197): note: see previous definition of 'Initialize
InstanceFields'
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64\cl.exe /c /nolo
go /Ox /W3 /GL /DNDEBUG /MD -IC:\Users\william\Desktop\hdlConvertor-master\hdlCo
nvertor-master\src/antlr4-runtime-cpp/ -IC:\Anaconda3\include -IC:\Anaconda3\inc
lude "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Pr
ogram Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program
Files (x86)\Windows Kits\10\include\10.0.10150.0\ucrt" "-IC:\Program Files (x86)
\Windows Kits\NETFXSDK\4.6\include\um" "-IC:\Program Files (x86)\Windows Kits\8.
1\include\shared" "-IC:\Program Files (x86)\Windows Kits\8.1\include\um" "-IC:\P
rogram Files (x86)\Windows Kits\8.1\include\winrt" "-IC:\Program Files (x86)\Mic
rosoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual
 Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\includ
e\10.0.10150.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6\include
\um" "-IC:\Program Files (x86)\Windows Kits\8.1\include\shared" "-IC:\Program Fi
les (x86)\Windows Kits\8.1\include\um" "-IC:\Program Files (x86)\Windows Kits\8.
1\include\winrt" /EHsc /TpC:\Users\william\Desktop\hdlConvertor-master\hdlConver
tor-master\src/antlr4-runtime-cpp\CharStream.cpp /Fobuild\temp.win-amd64-3.5\Rel
ease\Users\william\Desktop\hdlConvertor-master\hdlConvertor-master\src/antlr4-ru
ntime-cpp\CharStream.obj -std=c++11
cl : Command line warning D9002 : ignoring unknown option '-std=c++11'
CharStream.cpp
c:\users\william\desktop\hdlconvertor-master\hdlconvertor-master\src\antlr4-runt
ime-cpp\IntStream.h(41): warning C4251: 'antlr4::IntStream::UNKNOWN_SOURCE_NAME'
: class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' ne
eds to have dll-interface to be used by clients of class 'antlr4::IntStream'
C:\Users\william\Desktop\hdlConvertor-master\hdlConvertor-master\src/antlr4-runt
ime-cpp\CharStream.cpp(10): warning C4273: 'antlr4::CharStream::~CharStream': in
consistent dll linkage
c:\users\william\desktop\hdlconvertor-master\hdlconvertor-master\src\antlr4-runt
ime-cpp\CharStream.h(16): note: see previous definition of '{dtor}'
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64\cl.exe /c /nolo
go /Ox /W3 /GL /DNDEBUG /MD -IC:\Users\william\Desktop\hdlConvertor-master\hdlCo
nvertor-master\src/antlr4-runtime-cpp/ -IC:\Anaconda3\include -IC:\Anaconda3\inc
lude "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Pr
ogram Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program
Files (x86)\Windows Kits\10\include\10.0.10150.0\ucrt" "-IC:\Program Files (x86)
\Windows Kits\NETFXSDK\4.6\include\um" "-IC:\Program Files (x86)\Windows Kits\8.
1\include\shared" "-IC:\Program Files (x86)\Windows Kits\8.1\include\um" "-IC:\P
rogram Files (x86)\Windows Kits\8.1\include\winrt" "-IC:\Program Files (x86)\Mic
rosoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual
 Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\includ
e\10.0.10150.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6\include
\um" "-IC:\Program Files (x86)\Windows Kits\8.1\include\shared" "-IC:\Program Fi
les (x86)\Windows Kits\8.1\include\um" "-IC:\Program Files (x86)\Windows Kits\8.
1\include\winrt" /EHsc /TpC:\Users\william\Desktop\hdlConvertor-master\hdlConver
tor-master\src/antlr4-runtime-cpp\CommonToken.cpp /Fobuild\temp.win-amd64-3.5\Re
lease\Users\william\Desktop\hdlConvertor-master\hdlConvertor-master\src/antlr4-r
[... 213 lines cut ...]
ime-cpp\CommonToken.h(149): note: see previous definition of 'getInputStream'
C:\Users\william\Desktop\hdlConvertor-master\hdlConvertor-master\src/antlr4-runt
ime-cpp\CommonToken.cpp(155): warning C4273: 'antlr4::CommonToken::toString': in
consistent dll linkage
c:\users\william\desktop\hdlconvertor-master\hdlconvertor-master\src\antlr4-runt
ime-cpp\CommonToken.h(151): note: see previous definition of 'toString'
C:\Users\william\Desktop\hdlConvertor-master\hdlConvertor-master\src/antlr4-runt
ime-cpp\CommonToken.cpp(159): warning C4273: 'antlr4::CommonToken::toString': in
consistent dll linkage
c:\users\william\desktop\hdlconvertor-master\hdlconvertor-master\src\antlr4-runt
ime-cpp\CommonToken.h(153): note: see previous definition of 'toString'
C:\Users\william\Desktop\hdlConvertor-master\hdlConvertor-master\src/antlr4-runt
ime-cpp\CommonToken.cpp(186): warning C4273: 'antlr4::CommonToken::InitializeIns
tanceFields': inconsistent dll linkage
c:\users\william\desktop\hdlconvertor-master\hdlconvertor-master\src\antlr4-runt
ime-cpp\CommonToken.h(155): note: see previous definition of 'InitializeInstance
Fields'
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\
amd64\\cl.exe' failed with exit status 2

I have visual studio 14 installed and it is finding the c++ compiler it just hits an error, any tips?

Thanks,

Will

Nic30, 2017-11-25:
Hello, can you try to build ANTLR alone?
www.antlr.org/download/antlr4-cpp-runtime-4.7-source.zip

* Maybe there is an issue in setup.py 
 extra_compile_args=['-std=c++11'],
 is not required and for windows '-std=c++11' should be replaced with  '/std:c++11' or '/std:c++latest'

I do not have time to fix it until 15.12. But I try to help you, if you already found out how to fix this issue please make pull request or tell me to fix it for others.

Nic30, 2017-11-25:
I have to admit that I do not use windows compiler often.
Now I understand the issue. The problem probably is that the compiler tries to export all functions from in DLL. This probably happends because there is no function specified with _declspec(dllexport).
At least hdlConvertor/src/hdlConvertorModule.cpp/PyInit_hdlConvertor should have it.

* But there already windows users and library used to work for them. What is your python version?

whitew1994WW, 2017-11-27:
I will try to build ANTLR alone, how would I then build the rest?

My python version is 3.5.1. I have tried swapping out the command as you suggested however it doesn't recognize the argument'/std:c++11' or '/std:c++latest', however I can remove the argument completely.

whitew1994WW, 2017-11-27:
Building ANTLR on its own works just fine. How would I now go about compiling your package?

Nic30, 2017-11-27:
Can you try newer python for example python3.6.3 https://www.python.org/downloads/release/python-363/ ?

Nic30, 2017-11-27:
Can you post full report from compiler? I do not have windows machine at the moment.

whitew1994WW, 2017-11-30:
Sorry for the delay in getting back to you, thanks for your help. I will try with the latest python, and will then post a full report of the compiler.

Nic30, 2017-11-30:
OK, I just reminding that I do not have time to install windows and fix it until 15.12. because of other projects.

whitew1994WW, 2017-11-30:
That's ok, I understand. Im just happy for any support you can give!

whitew1994WW, 2017-11-30:
eam::getText': inconsistent dll linkage
c:\users\william\downloads\hdlconvertor-master\hdlconvertor-master\src\antlr4-ru
ntime-cpp\BufferedTokenStream.h(90): note: see previous definition of 'getText'
C:\Users\william\Downloads\hdlConvertor-master\hdlConvertor-master\src/antlr4-ru
ntime-cpp\BufferedTokenStream.cpp(387): warning C4273: 'antlr4::BufferedTokenStr
eam::getText': inconsistent dll linkage
c:\users\william\downloads\hdlconvertor-master\hdlconvertor-master\src\antlr4-ru
ntime-cpp\BufferedTokenStream.h(91): note: see previous definition of 'getText'
C:\Users\william\Downloads\hdlConvertor-master\hdlConvertor-master\src/antlr4-ru
ntime-cpp\BufferedTokenStream.cpp(391): warning C4273: 'antlr4::BufferedTokenStr
eam::getText': inconsistent dll linkage
c:\users\william\downloads\hdlconvertor-master\hdlconvertor-master\src\antlr4-ru
ntime-cpp\BufferedTokenStream.h(92): note: see previous definition of 'getText'
C:\Users\william\Downloads\hdlConvertor-master\hdlConvertor-master\src/antlr4-ru
ntime-cpp\BufferedTokenStream.cpp(399): warning C4273: 'antlr4::BufferedTokenStr
eam::fill': inconsistent dll linkage
c:\users\william\downloads\hdlconvertor-master\hdlconvertor-master\src\antlr4-ru
ntime-cpp\BufferedTokenStream.h(95): note: see previous definition of 'fill'
C:\Users\william\Downloads\hdlConvertor-master\hdlConvertor-master\src/antlr4-ru
ntime-cpp\BufferedTokenStream.cpp(410): warning C4273: 'antlr4::BufferedTokenStr
eam::InitializeInstanceFields': inconsistent dll linkage
c:\users\william\downloads\hdlconvertor-master\hdlconvertor-master\src\antlr4-ru
ntime-cpp\BufferedTokenStream.h(197): note: see previous definition of 'Initiali
zeInstanceFields'
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /
nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Users\william\Downloads\hdlConvertor-master
\hdlConvertor-master\src/antlr4-runtime-cpp/ -IC:\Users\william\AppData\Local\Pr
ograms\Python\Python36\include -IC:\Users\william\AppData\Local\Programs\Python\
Python36\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLU
DE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-
IC:\Program Files (x86)\Windows Kits\10\include\10.0.10150.0\ucrt" "-IC:\Program
 Files (x86)\Windows Kits\NETFXSDK\4.6\include\um" "-IC:\Program Files (x86)\Win
dows Kits\8.1\include\shared" "-IC:\Program Files (x86)\Windows Kits\8.1\include
\um" "-IC:\Program Files (x86)\Windows Kits\8.1\include\winrt" /EHsc /TpC:\Users
\william\Downloads\hdlConvertor-master\hdlConvertor-master\src/antlr4-runtime-cp
p\CharStream.cpp /Fobuild\temp.win-amd64-3.6\Release\Users\william\Downloads\hdl
Convertor-master\hdlConvertor-master\src/antlr4-runtime-cpp\CharStream.obj \std:
c++11
cl : Command line warning D9024 : unrecognized source file type '\std:c++11', ob
ject file assumed
cl : Command line warning D9027 : source file '\std:c++11' ignored
CharStream.cpp
c:\users\william\downloads\hdlconvertor-master\hdlconvertor-master\src\antlr4-ru
ntime-cpp\IntStream.h(41): warning C4251: 'antlr4::IntStream::UNKNOWN_SOURCE_NAM
E': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
needs to have dll-interface to be used by clients of class 'antlr4::IntStream'
C:\Users\william\Downloads\hdlConvertor-master\hdlConvertor-master\src/antlr4-ru
ntime-cpp\CharStream.cpp(10): warning C4273: 'antlr4::CharStream::~CharStream':
inconsistent dll linkage
c:\users\william\downloads\hdlconvertor-master\hdlconvertor-master\src\antlr4-ru
ntime-cpp\CharStream.h(16): note: see previous definition of '{dtor}'
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /
nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Users\william\Downloads\hdlConvertor-master
\hdlConvertor-master\src/antlr4-runtime-cpp/ -IC:\Users\william\AppData\Local\Pr
ograms\Python\Python36\include -IC:\Users\william\AppData\Local\Programs\Python\
Python36\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLU
DE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-
IC:\Program Files (x86)\Windows Kits\10\include\10.0.10150.0\ucrt" "-IC:\Program
 Files (x86)\Windows Kits\NETFXSDK\4.6\include\um" "-IC:\Program Files (x86)\Win
dows Kits\8.1\include\shared" "-IC:\Program Files (x86)\Windows Kits\8.1\include
\um" "-IC:\Program Files (x86)\Windows Kits\8.1\include\winrt" /EHsc /TpC:\Users
\william\Downloads\hdlConvertor-master\hdlConvertor-master\src/antlr4-runtime-cp
p\CommonToken.cpp /Fobuild\temp.win-amd64-3.6\Release\Users\william\Downloads\hd
lConvertor-master\hdlConvertor-master\src/antlr4-runtime-cpp\CommonToken.obj \st
d:c++11
cl : Command line warning D9024 : unrecognized source file type '\std:c++11', ob
ject file assumed
cl : Command line warning D9027 : source file '\std:c++11' ignored
CommonToken.cpp
c:\users\william\downloads\hdlconvertor-master\hdlconvertor-master\src\antlr4-ru
[... 207 lines cut ...]
eam': inconsistent dll linkage
c:\users\william\downloads\hdlconvertor-master\hdlconvertor-master\src\antlr4-ru
ntime-cpp\CommonToken.h(149): note: see previous definition of 'getInputStream'
C:\Users\william\Downloads\hdlConvertor-master\hdlConvertor-master\src/antlr4-ru
ntime-cpp\CommonToken.cpp(155): warning C4273: 'antlr4::CommonToken::toString':
inconsistent dll linkage
c:\users\william\downloads\hdlconvertor-master\hdlconvertor-master\src\antlr4-ru
ntime-cpp\CommonToken.h(151): note: see previous definition of 'toString'
[... cut ...]
ntime-cpp\CommonToken.cpp(186): warning C4273: 'antlr4::CommonToken::InitializeI
nstanceFields': inconsistent dll linkage
c:\users\william\downloads\hdlconvertor-master\hdlconvertor-master\src\antlr4-ru
ntime-cpp\CommonToken.h(155): note: see previous definition of 'InitializeInstan
ceFields'
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\
x86_amd64\\cl.exe' failed with exit status 2

C:\Users\william\Downloads\hdlConvertor-master\hdlConvertor-master>

This is the current error with the latest python.

Nic30, 2017-12-01:
ok, I found out that ANTLR probably needs some defines to work on windows,  bb95b3d, latest commit should have them

whitew1994WW, 2017-12-01:
It has now compiled! Thanks you, will confirm after I have tested in python.

Nic30, 2017-12-01:
ok, there is some simple test script in repo.