Nic30/hdlConvertor #13: A humble request

mohit162009, 2018-03-01:
Hello Nic,

I have a humble request, Could you please remove the examples you have added in the tests, I am not supposed to share such information, I can get in serious trouble. I am very sorry for the inconvenience and thanks a lot for your fixes and help.

Nic30, 2018-03-01:
Hello mohit162009,
I will remove it, but are you sure? There is nothing in package example just an array of 0 constants?
Do you mean package example or different file?

mohit162009, 2018-03-01:
Hello , Thanks a lot Nic! Yes these two files :+1: 
[image]

These files contains confidential information.

Nic30, 2018-03-01:
It is removed.

closed 2018-03-01 by Nic30