Nic30/hdlConvertor #15: Export comments from HDL

Nic30, 2018-09-27:
Add __doc__ for Entity, Architecture Module, functions, statements etc.

Comments are staged on hidden channel. It would be useful to include original HDL comments (maybe optionally) for doc generators etc. 

https://stackoverflow.com/questions/12485132/catching-and-keeping-all-comments-with-antlr
(first answer which was accepted is not actual ...)

Thomasb81, 2018-09-27:
Hi Nic,

In HDL languages there is no way to know that a comment is related to a specific statement. 
What will be your assumption ?

What I have already seen in some tool is the capture documentation when it's on the definition line.
Example :+1: 

```
/* 
 * Copyright to John Doe 2001- to 2018.
 * myModule is an illustration but this comment is not captured
 */

module myModule( a,b,c);
  input a; // Captured comment to document a as input  
  input [1:0] b; /* Caputred comment 
                          to  document b as an input of 2 bits */
  output c; // capture comment for documentation of c as output

  parameter d =8; // captured comment to document d as a parameter with default value of 8
  
  // prototype stuff is done
  // clever thing are written now but are secret and not disclosed.
endmodule
```

Nic30, 2018-09-28:
Hello Thomas,

1. 100% true, it is exactly like you are saying.
2. if we choose the most simple way and assign all previous comments to current objects
(Copyright - myModule, myModule.a - "", input [1:0] b - "Captured comment to document a as input"., document - "prototype stuff ...")
   and add start,end line no., char no. to objects and  docs users will be able to create the libraries which can resolve correct comment owner. 

This is not to meant to work in general, instead it should allow to do comment parsing for users of this library and it will work out-of-the box for well formatted code.

Some users like [GHDL](https://github.com/ghdl/ghdl) and [pyVHDLParser](https://github.com/Paebbels/pyVHDLParser) need this feature for document generators.

I am not the big fan of GHDL or reinventing the wheel, but true is that this feature is missing.
My real question is if I should implement rest of the process parsing or if I should implement this first.

Thomasb81, 2019-01-23:
Hello
I have now an idea on how this feature can be implemented. But this come with  limitation at least for verilog:

Line comment are not preserved in macro. If they are such code would not work:
```
`define DATA_PORTA input [63:0] a; // my data input port
`define DATA_PORTB ouput [63:0] b;

my module (a, b ... );
`DATA_PORTA `DATA_PORTB
...
```
Only comment of style `/* ...  */ `could be fully supported

Nic30, 2019-10-12:
Comments exported for definition of objects (module, component, port, variable, ...) and statements.
Comment has to be before the code of object an may be divided to multiple comments with any white-space separation.

Not that for ports are written like this he doc is not parsed correctly as the comment is actually before a different object.
```
module x(
input clk, //clk
input en, // en
)
```

But the doc in following code is parsed correctly:
```
module x(
//clk
input clk,
// en
input en,
)
```

It is not possible to resolve the owner of the comment automatically. But user of this library can write a script to move the comments to a correct object to fit any code-style.

closed 2019-10-12 by Nic30