Nic30/hdlConvertor #18: Failure to Install

ptracton, 2018-12-11:
I am attempting to install via pip using python3.7 from Anaconda on Red Hat Linux.

uname -a
Linux 2.6.32-696.el6.x86_64 #1 SMP Tue Feb 21 00:53:17 EST 2017 x86_64 x86_64 x86_64 GNU/Linux

gcc --version
gcc 4.8.3

    In file included from /tmp/pip-install-qkuw3rae/hdlConvertor/src/antlr4-runtime-cpp/antlr4-runtime.h:10:0,
                     from /tmp/pip-install-qkuw3rae/hdlConvertor/src/vhdlConvertor/../VhdlParser/vhdlParser.h:7,
                     from /tmp/pip-install-qkuw3rae/hdlConvertor/src/vhdlConvertor/packageHeaderParser.h:4,
                     from /tmp/pip-install-qkuw3rae/hdlConvertor/src/vhdlConvertor/packageHeaderParser.cpp:1:
    /tmp/pip-install-qkuw3rae/hdlConvertor/src/antlr4-runtime-cpp/antlr4-common.h:10:19: fatal error: codecvt: No such file or directory
     #include <codecvt>
                       ^
    compilation terminated.

I am missing this codecvt include.  Where did it come from and should getting it be part of the install process?

Thomasb81, 2018-12-11:
HdlConvertor use Antlr4 C++ runtime and used the stdlib c++. 

You should check that you have the standard c++ library development package install on your system. Probably libstdc++-devel ...
Thomas

Nic30, 2018-12-11:
I would advice to update GCC to 7+ (5.4 should be enough) as a first step.
The missing include  codecvt is part of standard library and probably was not present is 4.8.3.
But maybe it requires just some additional packages https://github.com/bakwc/JamSpell/issues/12.

I know some users are using this library on CentOS which is basically beta of RHEL so it should work.

Thomasb81, 2018-12-12:
In CMakefile.txt of antlr4 there is following comment:

> g++-5.0 and greater contain <codecvt> header. (test in ubuntu)
https://raw.githubusercontent.com/antlr/antlr4/master/runtime/Cpp/CMakeLists.txt

ptracton, 2018-12-12:
I'm a user, not administrator, of this machine.  I am unable to change the  version of gcc.  
Looks like this won't work on an older platform.

Nic30, 2018-12-12:
You can download the GCC yourself, compile it etc. but is is not comfort to setup everything by your self. Does your machine has docker or something similar installed?
What kind of machine it is? I mean you have legacy kernel and GCC, maybe update is right choice.

Nic30, 2018-12-12:
Also what exactly do you need there are another projects I can refer you to some if the GCC is problem for you?

ptracton, 2018-12-14:
This is just some exploratory work.  I am looking for an easy to use HDL parser in python.

Nic30, 2018-12-15:
Look at readme of this project. It has similar project section.
If I were you I would ask administrator for compiler update because your GCC is like 5 years old and it does not even support c++11 entirely.

Are you really sure that that newer compiler is not installed is there any gcc-5.4, newer compiler does not have to be default gcc.

Thomasb81, 2018-12-15:
Red-hat linux distribution still ship gcc 4.8.x has the default compiler: https://access.redhat.com/solutions/19458 even for the lastest RHEL7
To access to update compiler toolsuite you have to pay for "Red Hat Developer Toolset" in order to be supported by Red-Hat, I think.
Otherwise you need to download and install your self the toolset and it's dependency.

Industry like very stable software :)

Thomasb81, 2018-12-16:
By the way, the "out of eclipse build" does not work anymore at my end:
`$python setup.py build --verbose
running build_ext
building 'hdlConvertor' extension
creating build
creating build/temp.linux-x86_64-3.6
creating build/temp.linux-x86_64-3.6/home
creating build/temp.linux-x86_64-3.6/home/tom
creating build/temp.linux-x86_64-3.6/home/tom/prog
creating build/temp.linux-x86_64-3.6/home/tom/prog/git
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/svConverter
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/hdlObjects
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/verilogConvertor
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/vhdlConvertor
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/VhdlParser
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/SVParser
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/antlr4-runtime-cpp
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/antlr4-runtime-cpp/atn
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/antlr4-runtime-cpp/tree
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/antlr4-runtime-cpp/tree/pattern
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/antlr4-runtime-cpp/tree/xpath
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/antlr4-runtime-cpp/misc
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/antlr4-runtime-cpp/support
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/antlr4-runtime-cpp/dfa
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/baseHdlParser
creating build/temp.linux-x86_64-3.6/home/tom/prog/git/hdlConvertor/src/VerilogParser
`
Then it hang for ever, no compile process in the list of process running. 

I have to comment the following line in setup.py:
`distutils.ccompiler.CCompiler.compile = parallelCCompile`

It is unclear why the build work out of the box in eclipse... But I don't know the kind of magic is occurring in eclipse neither.

Nic30, 2018-12-17:
It seems that setuptools supports -j to specify number of build processes. It means that `parallelCCompile` in `setup.py` of this project is deprecated and will be removed.

Nic30, 2018-12-20:
No, it does not it is implemented in numpy private code https://github.com/numpy/numpy/blob/master/numpy/distutils/command/build_clib.py#L34

Nic30, 2018-12-20:
Anyway I am closing this issue as @ptracton has response.

closed 2018-12-20 by Nic30