Nic30/hdlConvertor #24: appveyor ci tests for windows

Nic30, 2019-01-03:


Nic30, 2019-01-14:
ci tests set, but not passing due ANTLR dll exports

Thomasb81, 2019-01-14:
It seems an additional optional `/bigobj` option is required for MSVS2017. Extract from the logof the build system:
`[00:03:34] C:\projects\hdlconvertor\src\svConverter\sv2012Parser\sv2012Parser.cpp : fatal error C1128: number of sections exceeded object file format limit: compile with /bigobj`

Thomasb81, 2019-01-14:
Or may be a define need to be setup to control the compilation of the sv part. Since it is today not usable.

Nic30, 2019-04-08:
Now compiles and fails while linking. https://ci.appveyor.com/project/nic30/hdlconvertor/builds/23688911
This errors are also related to vpp, as a msvc is what it is I will rather wait on vpp and then solve the rest of errors so I do not have to go trough same hell twice.

Nic30, 2019-05-01:
c1f8171f8049550824707a752a1ed68c30b7721c compiles and tests are passing

closed 2019-05-01 by Nic30