Nic30/hdlConvertor #27: [verilog] Cannot capture parameter in module_item

Thomasb81, 2019-01-07:
The verilog grammar allow to specify grammar in different way. This one is not yet implemented in hdlConvertor:
```
module paramM ();

  parameter integer DWIDTH = 9;
  parameter NOTYPE = 8 ;

endmodule
```

For reference : https://github.com/antlr/grammars-v4/issues/1343

Nic30, 2019-01-07:
Ok, I will fix it but it will be after 20.1. because I do have quite lot of 99% finished projects which should be done already.

Nic30, 2019-01-07:
There is just not implemented call in visitor of body items.

Nic30, 2019-04-24:
After seeing this I really feel bad...

Thomasb81, 2019-04-25:
Sorry I did not see your last comment and close the issue.
Should I send revert patch ?
Thomas

Nic30, 2019-04-25:
Me  on 7 Jan:

> Ok, I will fix it but it will be after 20.1.

and then I forgot...

closed 2019-04-25 by Nic30