Nic30/hdlConvertor #32: Visibility of preprocessor defines between the languages

Nic30, 2019-01-14:
Main problem is that verilog uses includes and vhdl imports.
The functionality is different and how to `define something in verilog for vhdl or reversed, is not known to me. 
There are tools which supports vhdl+verilog in single design, how does the preprocessor fork there?
For example how it works in Xilinx Vivado or Intel Quartus?

Nic30, 2019-01-15:
Yes, that is completely true. There is not a preprocessor in VHDL.
But is it possible to share for example defined constants between VHDL/Verilog?
If so, how?

Thomasb81, 2019-01-16:
Source code VHDL and verilog are incompatible.
However tool allow mix-language model interconnection : for example a VHDL model can instantiate model written in verilog and verilog model can instanciate VHDL model. Parameters can be transmit in a top - bottom approach. 

By the way the verilog Preprocessor need to be improve / rework. I just figure out that it does not support systemVerilog. For example it is not possible to include uvm_macros.svh file of a systemVerilog/uvm project. This is show stopper to validate my change in sv2012.g4.

I use systemVerilog/uvm to have quickly access to a variety of systemVerilog construction.

I am taking any advise to do that : 
- new grammar
- extension of existing grammar with a Boolean to allow specific systemVerilog token and grammar rules

Nic30, 2019-03-30:
OK, closing this issue as the comunicaiont between verilog preprocessor and others is not required

closed 2019-03-30 by Nic30