Nic30/hdlConvertor #34: Build system regression

Thomasb81, 2019-01-16:
Any change in the hdlConvertor trig the recompilation of libparsesv.a, libparseVerilog.a, libparseVhdl.a
even if it is not necessary.

Because libparsesv.a, libparseVerilog.a, libparseVhdl.a containt the antlr C grammar details, the compilation take time and is useless if grammar does not change.

This is a pain in the development short loop : edit-compile-test.

Thomasb81, 2019-01-16:
VerilogPreproc and Verilog parser source code are compile in different library but that have the same filename:
```
tom@venusia:~/prog/git/hdlConvertor$ find . -name '*.a'
./_skbuild/linux-x86_64-2.7/cmake-build/src/svConverter/libparsesv.a
./_skbuild/linux-x86_64-2.7/cmake-build/src/verilogPreproc/libparseVerilog.a
./_skbuild/linux-x86_64-2.7/cmake-build/src/verilogConvertor/libparseVerilog.a
./_skbuild/linux-x86_64-2.7/cmake-build/src/vhdlConvertor/libparseVhdl.a
./_skbuild/linux-x86_64-2.7/cmake-install/lib/libantlr4-runtime.a
```
It is a possible clue ...

Nic30, 2019-01-16:
Maybe the build is triggered because the parser for each language is generated again?
(Just a clue, I check it tomorrow.)

Thomasb81, 2019-01-18:
Issue solved

closed 2019-01-18 by Thomasb81