Nic30/hdlConvertor #36: [Verilog preprocessor] crash / incorrect support of nested macro

Thomasb81, 2019-01-21:
Following code crash the tool :

```
`define port_gen(inN,ouN) /* comment */ \
  input inN_clk; /* comment */ \
  input inN_rstn; /* comment */ \
  input inN_en; /* comment */ \
  input [7:0] inN_data;/* comment */ \
  /* comment */ \
  output [7:0] ouN_data; 

`define out_gen down_data

`define	sqr( x ) (x * x) // comment

`define	sum( a /* comment */ , b /* comment */ ) /* comment */ \
  (a + b)

`define sumsqr(
    a // comment
  ,
    b // comment
  ) \
  `sum ( \
    `sqr(a) \
   , \
    `sqr(b) \
   )

module macro();
 `port_gen(up,down)
 
  assign `out_gen = `sumsqr(
            up_data /* input data */
           ,
            (4 + 8) /* 12 */
           );

endmodule
```

Thomasb81, 2019-01-21:
I currently work on this issue.

Nic30, 2019-01-21:
Also add this code to tests so it will stay working in the future.

Thomasb81, 2019-01-21:
That is what I have in mind.
However the provided testcase is not valid. I will provide an other one.

closed 2019-01-25 by Nic30