Nic30/hdlConvertor #38: VHDL architecture is not parsed

Drolla, 2019-03-25:
Nice library - thanks for sharing it!

I have installed it on Lubuntu 18.10 x64. The parser seems generally to work, however it does not handle any VHDL architecture content. For example, the file 'mux.vhd' that is provided as test example is turned into the structure stored in the attached file 'mux_ast.txt':

[mux_ast.txt](https://github.com/Nic30/hdlConvertor/files/3005675/mux_ast.txt)

The entire following section of 'mux.vhd' is not parsed at all:

' process(i0,i1,i2,i3,sel)
' begin
' case sel is
'   when "00" => bitout <= i0;
'   when "01" => bitout <= i1;
'   when "10" => bitout <= i2;
'   when others => bitout <= i3;
' end case;
' end process;

My understanding was that all the VHDL 2002 constructs should be understood and parsed, shouldn't it?

Thomasb81, 2019-03-25:
Hi

The tool use a parser based on antlr4 library, this part is compatible with VHDL 2002. 
But it remain to the application developer to do something useful with the recognized data.

Currently hdlConvertor only deal with "prototype" of component that is convert to a structure for later use (example : documentation, netlisting ...) 
The vhdl process you just mention is part of the architecture section that is simply ignored by the code.

Thomas

Nic30, 2019-03-26:
@Drolla  as the @Thomasb81 said, the AST of VHDL is parsed however the data is not translated to python objects.

I understand that this feature is must-have. The implementation is scheduled on this month https://drive.google.com/file/d/1zyegLIf7VaBRyb-ED5vgOMmHzW4SRZLp/view

The development of anything related to VHDL is extremely time consuming and we are just two. The implementation will be straightforward. Any help would be appreciated or you can just wait or use any other lib from the list provided in readme however I just remind you that we are developing this library because the existing libraries have issues.

Drolla, 2019-03-26:
Hi @Thomasb81 and @Nic30,
Thanks for your kind reply. It was not clear to me that this tool is still in the middle of the development.
I have checked already the alternative solutions, but as you say they have also weaknesses. There is no problem to wait a bit until you progressed further.
You can of course close this case.
Andreas

Drolla, 2019-03-26:
Closing this issue since the finding is not a bug but a natural limitation due to the early development stage of the project.

closed 2019-03-26 by Drolla

Nic30, 2019-03-26:
@Drolla the main reason why this is not implemented is because everyone required the hierarchy parsing and not the parsing of the body. But this changed.

Drolla, 2019-03-26:
@Nic30, what is the reason this has changed? I guess it is not just my case I have opened, isn't it?

Nic30, 2019-03-26:
https://github.com/HardwareIR/netlistDB/issues/11

Nic30, 2019-04-01:
branch  process_parser: verilog process parsing works for all examples
vhdl planed on this week

Nic30, 2019-04-03:
simple vhdl processes  are now parsed and present in output, some of the things like generate constructs are still not implemented but it will come