Nic30/hdlConvertor #46: verilog preprocessor: tests of include in complex scenarios

Nic30, 2019-05-01:


Nic30, 2019-06-15:
https://ncrmnt.org/2017/06/17/the-painfull-verilog-preprocessor-pitfall/

The visibility of the preprocessor symbols in project with multiple files works differently under different tools.

* Specify the visibility mode per file.
   * for each top file new macro database
   * for all top files same macrodatabase

* [TODO]
   * warning on macro redefinition

* Our default implementation
   * each call of `hdlConvertor.parse` will use new clean macro database
       * the database will be accessible from python after method returns
       * the database could be initialized from dictionary or previous one could be reused based on flag
   * for each file in file list in `hdlConvertor.parse` call the macro database is shared

Thomasb81, 2019-06-15:
verilog2001  / 2.7.5 "Compiler directives" :
"A compiler directive in one description file can therefore control compilation behavior in multiple description files."

In SystemVerilog the notion of compilation unit, is better defined.

I agree, currently the preprocessor does not handle the notion of compilation unit.
I consider this as very advance verilog/systemVerilog feature. 
Most of the user will just add the `include statement in all file that require the symbol to be define. So compiling file one by one give the same result that compiling them in a row.

Nic30, 2019-06-15:
I agree,

my primary goal is to allow python to add defines and the rest is like very easy thing to do, which will allow users to do anything if they want to.

Thomasb81, 2019-06-15:
Well,
We just need to prefill the following map with user macro:
`   verilog_pp::macroSymbol defineDB;`

There is 3 usecase.
1) the user specify a command line option to define a symbol without any definition. ie `-define MYMACRO`
2) the user specify a command line option to define symbol with a definition` -define MYMACRO=6`
3) the last one is the case where the symbol has argument. This become tricky use case. I never had the need to support this with a commercial tool. I guess we can drop this requirement for the moment.

Nic30, 2019-06-15:
I was using SV every day, but now I did not used it for 0.5y and I do not have and compiler.
```
`define A
`define A(x) x
```
Is the redefinition of A or the preproc. does support "polymorphism" and there are two def. of A with a different type?

It seems to me that each macro def. is actually a function with arbitrary number of arguments which returns the string.
The correct number of args is only checked if the symbol is used.

This means that from python the `verilog_pp::macroSymbol` is a 
```
Dict[str,
       Union[
            Callable[[List[str], ], str],
            str,
            None,
        ]
]
```
I am right?
(I know that `verilog_pp::macroSymbol` now is not directly compatible with this, but my question is if this is correct.)

Thomasb81, 2019-06-15:
In case you really need to fill the lack of SV tool, there is https://www.edaplayground.com/ :)

No polymorphism. 
A is redefined.

Actually that's not a function because the behavior is string replacement.

The equivalent python structure would be more like for systemverilog
Because  systemVerilog support default argument.
:
Dict[
       str : Union[
               str,     <---- replacement template
               OrdDict[
                 str: str, <---- argument : default value of the argument if any
               ]  
            ],   
]

For verilog the data of the OrdDict is always "" (empty string)

As an user, if you think to usecase 3 ... You would have better to change your code than to write the equivalent on a command line ... because anyway the macro is so complex that you will need to check your source code to have the correct macro-prototype... 
Additionally you will need to know how your macro is used, in order to provide or not default argument.

Nic30, 2019-06-17:
@Thomasb81 today I found and fix:
* the content of ifdef/ifndef was always deleted
* the local directory not added correctly as an include dir

Do you know about any other bug like this two?

It still seems to me that newlines are not processed correctly but it is a detail.

Thomasb81, 2019-06-17:
I have doubt or don't remember how I handle to keep filename sync with the source code.
So in case of syntax error in a code imported in a `include statement, the tool will report the correct file and line.

It probably involve the `line that is probably not handle in the verilog grammar.

Nic30, 2019-06-18:
Now, which include directory is supposed to be searched first? Now the last one has highest priority, but is it correct? I would expect the first one to be searched first.

Nic30, 2019-06-18:
316e8ba5432ed781a946096ff480875ef93222b1 e5d3bca88474280d4a1832d08b16ea9c2286c55c

Nic30, 2019-06-18:
@Thomasb81 I have rewritten the macro handling, so filename is easy now.
However I do not know how to find out line no in output file without conversion of everything to string.
I would like to just walk some buffer of strings or so, but I do not see anything suitable in `antlr4::TokenStreamRewriter`.

Thomasb81, 2019-06-18:
```
// [TODO] MacroDef__LINE__, MacroDef__FILE__ are showing the position in current file
//        ant they should return the value for the final source file
```

Not exactly. It is a little bit more complicate than that. 
I think final source file terminology is incorrect, because, here the final source file is the result of the preprocessed code. This code is an "intermediate result" that is not supposed to be exposed to user.
 
See chapter  **22.13 `__FILE__ and `__LINE__** of system verilog 2012 spec.

It is the current file and current line position. Except after an ``include` where `__FILE__ and `__LINE__ must report the position in the include file being included. 

So __LINE__ always refer to the token line number in the original source code. Never in the "output" file.

Those "macro" will tell to user where he can retrieve  easily the position of the printed message in this source code. It make non-sense to point in the intermediate - preprocessing result...

Then I think we should be able to get a tokenStream from the TokenStreamRewriter object. 
This tokenStream probably need to be re-tockenize with the second step lexer. As a result the tokenStream is a patch work of token : according to the token origin, token can have different filename and line attribute that would be use by the error handler of the second step parser. This will allow to produce an error message with the correct source location...

At least that's how I think proper source code details can be share with the second step parser...

Nic30, 2019-06-21:
``` 
An `include directive changes the expansions of `__FILE__ and `__LINE__ to correspond to the
included file. At the end of that file, when processing resumes on the input file that contained the `include directive, the expansions of `__FILE__ and `__LINE__ revert to the values they had before the `include (but `__LINE__ is then incremented by one as processing moves to the line after the `include ).
```

Does this mean that after include the __FILE__ is set to file which was including (= top file) and the __LINE__ is line in original line + offset of the lines expanded by macros?

Thomasb81, 2019-06-21:
my_include.sv:
```

$display(`__FILE__,`__LINE__)
```
my_file.sv:
```
`include "my_include.sv"`
```
should display (according to me):
```
my_include.sv 2
```
the `$display` is on the second line of the my_include file.

You can have a look to fileline2.v in icarus testsuite: https://github.com/steveicarus/ivtest/
(Be careful, the testsuite contain a mix of verilog200, 2005 and systemVerilog)

fileline2.v can be seen as an intermediate file once the `include` directive has been process and code has been instruct with `line`directive.

https://github.com/steveicarus/ivtest/blob/master/ivltests/fileline2.v
https://github.com/steveicarus/ivtest/blob/master/gold/fileline2.gold

Thomasb81, 2019-06-22:
> Then I think we should be able to get a tokenStream from the TokenStreamRewriter object.
> This tokenStream probably need to be re-tockenize with the second step lexer. As a result the tokenStream is [...]

I change my mind. We should pass the file/line information to the second step parser using the `line` directive.
Which is probably the way language developer have though to propagate the file and line information from preprocessor to processor in order to build decent error message.

Nic30, 2019-09-17:
Tests already present.

closed 2019-09-17 by Nic30