Nic30/hdlConvertor #48: VDHL-2008, AST as Python objects

Nic30, 2019-06-02:
In few days, the radical change will come.
  * VHDL will get new full 2008 grammar
  * all AST object will have own class defined in Python and instantiated by C++ (JSON conversion will disappear from C++)
  * AST and HDL objects will be simplified to reduce codebase and to simplify processing
  * full reverse conversion to Verilog/VHDL to assert the functionality

I am still having problems with the grammar and a time.

After this step, probably next month it would be great if we can split the header files and make a C++ target.

imec-myhdl, 2019-07-10:
Hi, I have used a previous version to parse our vhdl source, with the intent to convert them (partially) to myhdl (see myhdl.org), to be able to co-simulate with the rest of our system model. I am excited to see that you are working on a simple universal AST, because that will greatly ease the addition of translation to myhdl. 

My question: Do you have a timeline for this  simple universal AST? (weeks/months/years).

Kind regards, Paul

P.S. MyHDL already has a conversion to Verilog/Vhdl, so I would 'only' need to work out the conversion from this simplified AST to myhdl

Nic30, 2019-07-10:
> My question: Do you have a timeline for this simple universal AST? (weeks/months/years).

Month. Can be done much sooner, but I have to do some things first in order to advance in my PhD.

> MyHDL already has a conversion to Verilog/Vhdl

* i know, without support for user defined types etc.
* it is the most easy part and I am not worry about it now

Also share link on your project.
It would be great if I can also use some tests from your project to test this library, as writing tests consumes a lot of time.

imec-myhdl, 2019-07-11:
Hi Nic30,
I am pretty fluent in python, do occasionally write verilog, but the vhdl is rusty. I am interested in hdlConverter because we need to revive a legacy project with a lot of vhdl, while all our models and testbenches are in myhdl. 

I currently work on Spyce (https://github.com/imec-myhdl/Spyce) a graphical drawing tool that generates myhdl code. It is work in progress: it does work, but there are quite some outdated blocks, and I still need to add a clear example. Do not use our myhdl branch (yet) if you want to test, because we are currently updating the fixbv implementation to be able to convert to verilog and the modifications are not backward compatible.

Unfortunately I cannot share even a line of vhdl, but of course I can test the hdlConvertor on our code base, and report the results/bugs. 

Kind regards, Paul

Nic30, 2019-07-11:
> a graphical drawing tool that generates myhdl code

(May not be necessary useful to you but maybe interesting.)

3 years ago I was thinking about exactly the same thing. But I find out that it  is actually much more developer friendly to write myhdl (or any other hdl) and see what the circuit looks like.
(Which leads to very hard computational problem how to draw a nice graph from the netlist... etc.)

I do have library which [generates](https://github.com/Nic30/hwtGraph) and [draws](https://github.com/Nic30/d3-hwschematic) such a diagrams I am using it in doc generator ([example](https://hwtlib.readthedocs.io/en/latest/_static/schematic_viewer/schematic_viewer.html?schematic=../../_static/hwt_schematics/hwtLib.clocking.clkDivider.ClkDiv3-0.json))

> I can test the hdlConvertor on our code base, and report the results/bugs.

Nice, exactly what I need. (Sharing VHDL code is not required, it is always about just some construct in file and not about code as a whole.)

imec-myhdl, 2019-07-11:
> I do have library which generates and draws such a diagrams I am using it in doc generator (example)

Nice tools! You are right about developer friendly, but customers and occasional users prefer point and click (the schematic is a communication tool that happens to output executable code...) And the barrier for non language-experts is too high. 

Perfect example is simulink; horrible tool, and way too expensive, yet we found that projects in matlab are never explored by non-experts, whereas simulink is used all the time. The tool is only to make the barrier lower. You would never draw a fsm for example, but simply use a myhdl view under a symbol, it is mainly useful to visualize signal-flow

Kind regards, Paul

closed 2019-10-11 by Nic30