Nic30/hdlConvertor #50: check parse errors before conversions

Nic30, 2019-06-03:
If the source file is in invalid format the ANTL4 will recover, however some of the parse tree nodes may become NULL. It would be very obscure to check every node access for null.

From this reason the ANTL4 parse errors has to be checked before conversion and  by default conversion can not be performed even if the parser recovered. And proper exception should be raised instead

Thomasb81, 2019-06-03:
Hello

Could you attach a source code with an issue that do not trig an exception ?
I have the felling that exception are raise... But I may miss something.
Thanks

Nic30, 2019-06-03:
Well today I just tried malformed VHDL and it just works as expected, but when I had broken grammar it never raised an exception, I had to uncomment https://github.com/Nic30/hdlConvertor/blob/master/src/syntaxErrorLogger.cpp#L41 to actually see the errors.
I try to replicate issue. But this is odd.

Thomasb81, 2019-06-03:
The verilog preprocessor has not the antlr4ErrorListener installed. ... So if there is a preprocessor syntax issue, the output message is produce in an uncontrolled way. 

At some point the SyntaxErrorLogger::CheckErrors() methdod should be call and the error must be reported through exception mechanism.

Nic30, 2019-06-19:
done in #57

closed 2019-06-19 by Nic30