Nic30/hdlConvertor #51: proper brace resolution alg.

Nic30, 2019-06-08:
There is an algorithm which prints braces in expression only if they are truly required, based on operator precedence.
https://github.com/HardwareIR/netlistDB/blob/master/src/serializer/serializer.cpp#L56
Need an equivalent for toVhdl/toVerilog for output to look nice.

Nic30, 2019-11-13:
implemented in https://github.com/Nic30/hwt/blob/master/hwt/serializer/generic/serializer.py#L272
Current plan is to extract everything related to VHDL/Verilog generating to independent library (from hdlConvertor and HWT).

This will brings:
* reduction of code duplication
* high quality VHDL/Verilog/SystemC/... code generators to hdlConvertor
* analysis and visualization tools for HWT to VHDL/Verilog ecosystem
* unified API for imports from HDL to HWT

rhinton, 2020-02-21:
Has this extraction to a separate library progressed?

Nic30, 2020-02-21:
Yes, I did finish it, however I did not pushed it yet.
The reason I did not pushed it yet is that it is dependent on new SV preprocessor code line-error handling logic which is not complete yet.

closed 2020-03-31 by Nic30