Nic30/hdlConvertor #52: Verilog Assign to Concatenation Error

GCHQDeveloper560, 2019-06-14:
Verilog allows concatenated values on the left hand side of an assign statement. From [this code](https://opencores.org/websvn/filedetails?repname=aes_core&path=%2Faes_core%2Ftrunk%2Frtl%2Fverilog%2Faes_cipher_top.v):

`assign {sa00_mc, sa10_mc, sa20_mc, sa30_mc}  = mix_col(sa00_sr,sa10_sr,sa20_sr,sa30_sr);`

However an attempt to parse this file results in the following error:

`hdlConvertor._hdlConvertor.ParseException: wrong assign initialisation`

The grammar seems to support this assignment style. I see `net_concatenation` included in `net_lvalue` which is used by `net_assignment`, so it's not clear to me what might be happening.

Nic30, 2019-06-14:
Fixed.

There was a bug in Verilog AST -> simplified HDL AST conversion.
https://github.com/Nic30/hdlConvertor/commit/4bf61250c59ccf7b8501e85f50047b95f7574178#diff-8ced4ecfb29c59ecc820193fa1714789L226

The functionality was not tested properly as the complete testsuite is under construction 
(
All examples from Verilog STD will be extracted, parsed, converted back to Verilog using hdlConvertor.toVerilog and then checked for equivalent meaning

Currently I extracted only first 87 (~10%) pages from VHDL std. It is simple, but it takes a time.
Any help appreciated.
)

Thanks for reporting.

Anyway your example contains definitions of the verilog functions. This objects are parsed but they are not converted to Python same as delay specifications which are also present.

The functionality will be implemented this weekend in order described in this issue.
https://github.com/Nic30/hdlConvertor/issues/48

Nic30, 2019-06-15:
@GCHQDeveloper560 can you please confirm that it did work so we can close this issue?

GCHQDeveloper560, 2019-06-17:
I wasn't able to build the module from master. I'm using GCC 7.4 (Ubuntu Bionic), and I get multiple errors about use of the new C++ filesystem library. [According to Stack Overflow](https://stackoverflow.com/questions/39231363/fatal-error-filesystem-no-such-file-or-directory) this is present in GCC 8 but is experimental in earlier releases. Do you plan to support older GCC releases?

Nic30, 2019-06-17:
Yes the support will probably come later today. Now it works only under gcc 8+ msvc 2017+.

Nic30, 2019-06-17:
@GCHQDeveloper560  theoretically it is possible to have support for 5.4 and later. However I do not have a motivation to check the support if no one is using this lib. with this compilers. Also adding a new compiler to the build consumes resources on travis-ci.

GCHQDeveloper560, 2019-06-17:
It looks like things build fine now, and a quick check indicates I'm not having problems with the file I referenced. However, I get a segfault if I try to parse [this code](https://opencores.org/websvn/filedetails?repname=aes_core&path=%2Faes_core%2Ftrunk%2Frtl%2Fverilog%2Faes_inv_cipher_top.v) in the same repo. Should I open that as a separate issue?

Nic30, 2019-06-17:
Ah, I add whole repo to the tests, it is another "small error", I really need to extract the examples from Verilog standard to have properly tested everything. But I need to write some paper first.

GCHQDeveloper560, 2019-06-24:
Thank you for the fixes! Things seem to be working on the examples I've tried so far, so I'll close the issue.

closed 2019-06-24 by GCHQDeveloper560