Nic30/hdlConvertor #53: A fast System Verilog grammar

Nic30, 2019-06-15:
https://github.com/antlr/grammars-v4/issues/1438
https://github.com/Nic30/hdlConvertor/commit/736ae3bd95a02e88ba0018f6cc1e6e797f039066#commitcomment-33951857

The current sv2012 grammar is so slow that it is "not possible" to use it for real libraries.
The reason for this is simply the code of the grammar.
In order to improve it we have to remove simple rules and deal with ambiguity. Because the main performance issues comes from ambiguity and full context analysis caused by it.

It may be wise to first update to sv2017 and optimize after.

Nic30, 2019-06-15:
As I mentioned in https://github.com/antlr/grammars-v4/issues/1438 there are many entirely different System Verilog grammars on the internet. It would be good to first test their performance before creating an another grammar.

Thomasb81, 2019-07-10:
Hello

My current idea is:
1) Build a symbol table, that would record element in their hierarchy and skip statement in a first pass.
Such code of 20 lines  is already generate a big number of ambiguity slowing down the parsing analyze.
```
package ComplexPkg;

  typedef struct {
    shortreal i, r;
  } Complex;

  
  function Complex add(Complex a, b);
    add.r = a.r + b.r;
    add.i = a.i + b.i;    
  endfunction

  function Complex mul(Complex a, b);
    mul.r = (a.r * b.r) - (a.i * b.i);
    mul.i = (a.r * b.i) + (a.i * b.r);
  endfunction

endpackage : ComplexPkg
```
In current code example. A prerequisite can forbid the parsing of the function statement during the first pass parsing. 

2) Once this symbol table is fill a phase parsing  will parse the all language. Ambiguity will be solve by quering the symbol table structure.

The symbol table can be leverage for verilog and VHDL parsing has well.

One inspiration for the symbol table could be :  [antlr symtab](https://github.com/antlr/symtab) But that's java code. Also for systemVerilog there is far more item to capture (package,program, class, interface, module, function, task, type  covergroup ...)

Nic30, 2019-07-11:
Maybe we do not need an extra pass as we can build symbol table while parsing the AST (I mean in antlr parser) as the symbol has to be first defined before it can be used.

The problem is that in order to perform resolution of symbol type we need to have whole code.
I do not know if it is a problem. But it would be better if we can just move priority of rules in grammar and remove duplicity from rules.

However I need to find time to learn what are the grammar optimisations which antlr performs in order to think about this.

Thomasb81, 2019-07-11:
May be I don't really understand what you mean by AST ... 

At some point you need to have in data structure that:
You have define a package named `ComplexPkg`. 
This package define a type `Complex`.  
This package define a function `add` that return a `Complex` type, take an argument named `a` of kind `Complex ` and take a argument call `b` of kind `Complex`
This package define a function `mul` that return a `Complex` type, take an argument named `a` of kind Complex  and `b`  of kind `Complex`

On big source of ambiguity that I have found is on this part of the formal syntax:
```
primary ::=
primary_literal
| [ class_qualifier | package_scope ] hierarchical_identifier select
| empty_unpacked_array_concatenation
| concatenation [ [ range_expression ] ]
| multiple_concatenation [ [ range_expression ] ]
| function_subroutine_call
| let_expression
| ( mintypmax_expression )
| cast
| assignment_pattern_expression
| streaming_concatenation
| sequence_method_call
| this
| $
| null
```
Basically most of the time ambiguity between 2, 6 and 7 alternative kill the performance.
Changing the order of alternative, will only change the alternative the tool end up to chose. It may chose a wrong one.
To solve this issue, I have no other way than to be able to pickup the declaration of identifier and assign them a scope. So we are able to add predicate in the grammar of the kind `{isHierarchicalId()}?`,`{isLetId()}?` or `{isfunction()}?`

This way the antlr will choose the correct alternative in function of the code it has already seen. Which make sense most of time for example in C language: Usually you cannot use a function if you did not define the function or provide it's prototype.

Maybe you are right a second pass is may be not required... I have in mind that predicate call can occurred a big number of time. So no side effect in this code, the code should be short to avoid other performance issue. It is a bad idea to browse the AST in a predicate function call.
Also I don't think I have need to parse the function definition for my usage. I conclude quickly (but that's probably wrong) that a 2 step parsing is a solution.

Nic30, 2019-07-11:
> May be I don't really understand what you mean by AST ...

result of antlr parser in this case (parse tree)

> Basically most of the time ambiguity between 2 (hierarchical_identifier), 6 (function_subroutine_call) and 7 (let expression) alternative kill the performance.

Maybe we can extract the shared part (identifier) from this rules.

Nic30, 2019-07-29:
OK, as I mentioned on gitter the grammar is prepared and seems to be working and working fast enough. The problem now is my time and the push is expected on Friday. At the beginning there will be some testing branch but the goal is to replace all verilog grammars to reduce codebase.

Nic30, 2019-08-03:
* TIME_LITERAL missing whitespace

Nic30, 2019-08-08:
All known issues solved, now it is required:

* few manual tests in antlr gui to see if something is not seriously wrong.
* keyword banning based on language version
* integration
* extension of current test suite to cover commonly used SV features
*  performance tuning (There are plenty of opportunities, however some improvements are quite invasive and we need to have test suite first. )

martinda, 2019-08-09:
Regarding a test suite, do you have some measure of code coverage?

Nic30, 2019-08-10:
No, there is not a code coverage report available. The reason is that I currently do not know how to setup gcov or something else and submit outputs to coveralls.io.
It is must have, but somehow it felt trough priority list.

Thomasb81, 2019-08-10:
Sorry but the problem I was pointing is not solve with the reshuffle of the grammar.
[test.sv.tar.gz](https://github.com/Nic30/hdlConvertor/files/3488667/test.sv.tar.gz)

```
$ grun sv2017 source_text -diagnostics test.sv 
line 5:11 reportAttemptingFullContext d=356 (variable_decl_assignment), input='field1;'
line 5:11 reportAmbiguity d=356 (variable_decl_assignment): ambigAlts={1, 3}, input='field1;'
line 6:10 reportAttemptingFullContext d=356 (variable_decl_assignment), input='field2;'
line 6:10 reportAmbiguity d=356 (variable_decl_assignment): ambigAlts={1, 3}, input='field2;'

```
Ambiguity leading to full context parsing remain... 

[image]

Nic30, 2019-08-16:
note that current development is happening in [sv2017 branch](https://github.com/Nic30/hdlConvertor/tree/sv2017)

Nic30, 2019-09-09:
25d96059ee9d2744b2d3310cac196a33304c5915

Nic30, 2019-09-19:
After dad2dc791076f41f65fdeaa958ccc26c4675da9f I do not see any major issues with the performance of new SV grammar.

closed 2019-09-19 by Nic30