Nic30/hdlConvertor #55: Toplevel assignments ignored in VHDL

zegervdv, 2019-06-17:
Parsing a VHDL file, the toplevel assignments in a process are not parsed.

Example VHDL input:
``` vhdl
architecture arch of module is
begin
   data <= i_data and i_mask;
end architecture arch;
```
I get a module:
```
(Pdb) pp d.objs
[<hdlConvertor.hdlAst.HdlModuleDef object at 0x7f6fc6d75120>]
```
but it has no content:
```
(Pdb) pp d.objs[0].objs
[]
```

Nic30, 2019-06-18:
Should be working now. 

The objects were not converted as they are not regular assignments but so called concurrent statements which are entirely separated from from regular statements in now VHDL-2008 grammar.

Nic30, 2019-06-18:
Also everything which is not implemented is notified if you call `HdlConverotr.parse` method with `debug=True`.

Thanks for reporting.

zegervdv, 2019-06-18:
I can confirm it works now, thanks!

closed 2019-06-18 by Nic30