Nic30/hdlConvertor #56: Parse crashes on VHDL 'range attribute

zegervdv, 2019-06-19:
Example file:
``` vhdl
architecture arch of module is
begin
    data <= i_data(i_data'range) and i_mask;
end architecture arch;
```

Parsing is successful when leaving out the `'range`.
There are no visible errors, even with `debug=True`.

Nic30, 2019-06-19:
There are 2 problems. 

1. #57 
2. the grammar fails to detect call with attr. access inside https://github.com/Nic30/hdlConvertor/blob/master/grammars/vhdl.g4#L1032
but it should

I am planing to fix it tonight.

zegervdv, 2019-06-19:
Found a similar issue with casting: `string'("text")`.
Should I make a new issue for this?

Also it seems to crash when there are syntax issues in the file, but I guess that fall more under the scope of #57?

Nic30, 2019-06-19:
#57 is serious.

About this issue and `string'("text")` 

I think that https://github.com/Nic30/hdlConvertor/blob/master/grammars/vhdl.g4#L217 and maybe something around is not correct, 
(I belive that everything related to `<expression>'(<expression>)` is the same issue. I had to modify this part of grammar in order to make it suitable for ANTL4 but I probably made some mistake. You can compare this (`name`) section with the vhdl standard if you have a time.)

 I will fix it as soon as I finish writing something else.

Nic30, 2019-06-19:
2f2cbb3203b18f5402c6f7b28a3f92c65b4eff75

Nic30, 2019-06-20:
Both issues were fixed. Verify if it works for you. After fixing #57 the segfaults without previous error msg. should be very very rare. There are still some functions which are returning nullptr, because they are not implemented but all of them should log it at least.

zegervdv, 2019-06-20:
Great, it works for me now.
Thanks for the quick fix.

closed 2019-06-20 by zegervdv