Nic30/hdlConvertor #60: [Feature] Sourcetrail: a perfect use case for hdlConvertor?

pidgeon777, 2019-07-02:
Today I'm here to discuss an amazing tool I recently discovered. It is called Sourcetrail:

https://www.sourcetrail.com/

It is an actively maintained and free for non-commercial use project.

Its super cool feature is related to the ability of indexing source codes of different types, at the same time, and navigating through them by making use of one of the most advanced code-browsing systems I've ever seen.

The real game-changer is, however, the Graph diagram, which really makes you understand how the code is structured.

At some point, I thought: wouldn't it be great for this tool to support Verilog/VHDL?

And then I did a great discovery, the Sourcetrails authors told me that it is indeed possible to add any custom parse-able language/data to the indexing database, by making use of this library:

https://github.com/CoatiSoftware/SourcetrailDB

From the project page:

> The SourcetrailDB project provides write access to Sourcetrail database files. You can use the SourcetrailDB project to write an indexer for an arbitrary programming language (or other kinds of data, e.g. see poetry indexer example) and export a Sourcetrail database file that can be viewed and navigated within Sourcetrail.

There are even some nice examples which explain how to create an indexer for some poetry texts (!):

https://github.com/CoatiSoftware/SourcetrailDB/tree/master/examples/cpp_poetry_indexer

I asked Sourcetrails authors about adding VHDL support, and they told me that a robust VHDL parser providing the necessary elements to the SourcetrailDB would have done the trick.

So, hdlConvertor (and other tools) came into my mind.

By adding the parsed results to the SourcetrailDB, then something like the following could be obtained:

[image]

You could, for example, click on the *SUBCOMPONENT* entity of *COMPONENT2*, and this would expand it. Or, you could click on the *SIGNALS* section of *COMPONENT1*, and a box with all of the internally declared signals would appear. Or, even more, when clicking on a certain element of *PORT OUT* of *COMPONENT 1*, then Sourcetrail would show all of its references, for example, its connections with other components. Graphically **and** in the source code section.

The results would be extremely cool, in my opinion: a different way to visually navigate VHDL code, without having to go down to the RTL level. This has the potential to be a real time-saver, especially when navigating through complex designs. As a comparison, just consider each Verilog/VHDL component as a C++ class with internal variables, methods, instantiations and so on, and the interconnections between these classes.

Thus, I would like to discuss with you about the do-ability of my idea and the possible benefits arising from it.

Nic30, 2019-07-02:
Hello @pidgeon777, free for non commercial use can be tricky.
However the Sourcetrail looks nice, I am do have project which does something similar.
You can see the result for example [there](https://hwtlib.readthedocs.io/en/latest/_static/schematic_viewer/schematic_viewer.html?schematic=../../_static/hwt_schematics/hwtLib.clocking.clkDivider.ClkDiv3-0.json).
 
Also the format, I am using now, seems to be directly conversible.
Actually my only problem is that most precious element of them all, the time.
Can you manage the communication with Sourcetrail team and check their's plane or help with programming/testing?
How can we present the thing to potential users?

Nic30, 2019-09-07:
Closing this issue due lack of feedback.

closed 2019-09-07 by Nic30