Nic30/hdlConvertor #68: [SV parser] randomize_call

Thomasb81, 2019-08-10:
https://github.com/Nic30/hdlConvertor/blob/845edd15ac73cd5026fbaaafd8f04083b30d94c7/grammars/sv2017Parser.g4#L1474

Parsing of https://github.com/Nic30/hdlConvertor/blob/845edd15ac73cd5026fbaaafd8f04083b30d94c7/tests/sv_test/std2017/p137.sv#L10 failed

LRM state for following definition
randomize_call ::=
randomize { attribute_instance }
[ ( [ variable_identifier_list | null ] ) ]
[ with [ ( [ identifier_list ] ) ] constraint_block ]

However

the 'with' part should use curly bracket instead of bracket.

and 

variable_identifier_list should be a hierarchical_variable_identifier_list that does not exist in order to reference hierarchical variable.

Nic30, 2019-08-12:
> the 'with' part should use curly bracket instead of bracket.

The `{}` block is  `constraint_block`.

Thomasb81, 2019-08-12:
Correct I should have mix something.

commit d5e827c2 (2019-10-10): sv2017: fix randomize_call, dpi_import_export, close #68, close #72
closed 2019-08-16 by Nic30