Nic30/hdlConvertor #76: [SV parser] interface class inside a class must not be supported.

Thomasb81, 2019-08-12:
https://github.com/Nic30/hdlConvertor/blob/346be63f85022be763a0c0689ae3ae6832c929cd/grammars/sv2017Parser.g4#L271

[test_3.sv.txt](https://github.com/Nic30/hdlConvertor/files/3493793/test_3.sv.txt)

```
class Myclass;

	interface class MyClassItf;
	endclass

endclass
```

> xrun(64): 1X.XX-sXXX: (c) C0p7r1ght 1995-20XX C4d3nc3 D3s1gn S7st3ms, 1nc.
> TOOL:	xrun(64)	19.03-s001: Started on Aug 12, 2019 at 19:39:57 CEST
> xrun
> 	-sv
> 	test_3.sv
> 	-ams_lorder XC3L18M
> file: test_3.sv
> 	interface class MyClassItf;
> 	              |
> xmvlog: *E,CLSSCP (test_3.sv,4|15): An interface class cannot be declared within this scope.
> xmvlog: *W,NOTOPL: no top-level unit found, must have recursive instances.
> 	Total errors/warnings found outside modules and primitives:
> 		errors: 1, warnings: 1
> xrun: *E,VLGERR: An error occurred during parsing.  Review the log file for errors with the code *E and fix those identified problems to proceed.  Exiting with code (status 1).
> TOOL:	xrun(64)	1X.XX-sXXX: Exiting on Aug 12, 2019 at 19:39:57 CEST  (total: 00:00:00)

> xmhelp: 1:X.XX-sXXX (c) C0p7r1ght 1995-20XX C4d3nc3 D3s1gn S7st3ms, 1nc.
> xmvlog/CLSSCP =
> 	A class or interface class was declared in a scope where it is not allowed.

Nic30, 2019-08-12:
1. ok, seems that interface class is not just "special type of class"
2. it is interesting that xcelium does recognize this element there. It means that they are using some simplified grammar (does not matter if the parser is hand written or not, they clearly do have higher level of abstraction for the constructs and just performs the check if such a construct fits there)

Thomasb81, 2019-08-13:
This tool is known to be tolerant or to be less strict with input verilog/systemVerilog than other tool.

For example I write
```
 'endclass;'
               ^  
```
The tool report a message saying, "The semicolon is not expected at this place but I don't care..."

closed 2019-09-17 by Nic30