Nic30/hdlConvertor #78: [SV parser] module_or_generate_item rule performance

Nic30, 2019-08-21:
The `module_or_generate_item` rule is performance disaster, it can overlap with many others and usually check for exit is performed on many places (deep lookahead) in sv code corresponding to this rule.

[image]

Thomasb81, 2019-08-21:
My analyze is the following, I use a simple testcase:
```
module dimm(i,o);
input i;
output o;
genvar i;
for (i=0; i<4; i=i+1) begin:word16
        sms_08b216t0 p();
      end

endmodule
```
On rule:
```
module_or_generate_item:
 ( attribute_instance )* ( parameter_override 
                              | gate_instantiation
                              | udp_instantiation
                              | module_instantiation 
                              | module_common_item 
                              );
```
The ambiguity is between `module_instantiation` and `module_common_item` . But if we get a deeper look at `module_common_item` :
```
module_common_item:
 module_or_generate_item_declaration 
      | interface_instantiation 
      | program_instantiation 
      | assertion_item 
      | bind_directive 
      | continuous_assign 
      | net_alias 
      | initial_construct 
      | final_construct 
      | always_construct 
      | loop_generate_construct 
      | conditional_generate_construct 
      | elaboration_system_task 
     ;
```
You can figure out that _hard_ ambiguity should be between `module_instantiation`, `interface_instantiation` and `program_instantiation`.  
The syntax is the same to instantiate a module, an interface or a program: it is context dependent.

However it seems that `module_or_generate_item_declaration`  is also part of the ambiguity, because commenting `interface_instantiation` and `program_instantiation` of `module_common_item` is not enough to completely remove the ambiguity on `module_or_generate_item` rule.

Nic30, 2019-08-21:
It seems to me that all `*_item` and `*_item_declaration` are just duplicates of each other and rules often using them all. I mean: We need to do some expansion of `*_item*`, make items unique and then extract the items again for better readability.

Thomasb81, 2019-08-21:
Sorry I do not understand. `*_item `and `*_item_declaration` serve different purpose...

Agree, some structure are recursive, like the possibility to declare module inside a module...

Nic30, 2019-08-21:
I meant `*_item `and `*_item_declaration` suffers from similar issue.

Nic30, 2019-08-24:
There has to be something tricky. Many rules in this section are starting with identifier, that is true, but the ambiguity is quite big, it has to mean that some rules are subset of other rule, but I do not see it.

Nic30, 2019-09-19:
The grammar was entirely reworked, this issue is not relevant any more.

closed 2019-09-19 by Nic30