Nic30/hdlConvertor #85: verilog preproc: Full expr analysis for macro arguments and default values

Nic30, 2019-09-19:
```
`define xxerror(logfile, msg) $blah(logfile,msg)
`xxerror("ab,cd","e,f");
`xxerror(this.logfile, vec);
`xxerror(this.logfile, vec[1,2,3]);
`xxerror(this.logfile, {blah.name(), " is not foo"});
```

Currently only string and () expressions are processed correctly, but there are also [] and {}.

Thomasb81, 2019-09-19:
Have you an example with [] to share ? I don't have [ ... , ...] use case in mind.

Nic30, 2019-09-19:
I found this issues thanks to this file https://github.com/verilator/verilator/blob/master/test_regress/t/t_preproc.v

Nic30, 2019-09-19:
Currently I am rewriting the expression part of the preproc so it it is faster and buf-free.

Nic30, 2019-09-20:
fixed in ddf7f896c857fe81cee378f619adc2a8ccea1c3c

closed 2019-09-20 by Nic30