Nic30/hdlConvertor #88: Missing positional information for parsed objects

hs-apotell, 2019-11-09:
Positional information is lost for variables declared on the same line. For instance, for the following expression

```
reg Ctrl1,Ctrl2,Ctrl3,Ctrl4,Ctrl5,Ctrl6,Ctrl7;
```

positional information is correct only for Ctrl1 and missing for all the other variables. Also, precise column information is missing for all parsed data.

Nic30, 2019-11-09:
The problem is that the position is updated there https://github.com/Nic30/hdlConvertor/blob/master/src/svConvertor/moduleParser.cpp#L527
and not there 
https://github.com/Nic30/hdlConvertor/blob/master/src/svConvertor/moduleParser.cpp#L701

hs-apotell, 2019-11-09:
> 
> 
> The problem is that the position is updated there https://github.com/Nic30/hdlConvertor/blob/master/src/svConvertor/moduleParser.cpp#L527
> and not there
> https://github.com/Nic30/hdlConvertor/blob/master/src/svConvertor/moduleParser.cpp#L701

I will give that a shot and post a PR if it works as expected.

Nic30, 2019-11-09:
Thanks, I have to push work on some papers and other projects, I hope I will get back soon, if you can fix this, that would be nice.

hs-apotell, 2019-11-10:
Positional information is missing for many different object types. It might be worth actually redirecting calls to ```make_unique``` such that every call goes thru' some global function which sets up core parameters like position and source file.

Any thoughts on the right approach to ensure this information is "always" filled?

Nic30, 2019-11-10:
make_unique call the constructor of the class. You mean to initialize position in constructor, right?
One problem I see with this is that it is sometimes required to create `hdlConvertor::hdlObjects` without the antlr context with the position etc.
If I search `make_unique<Hdl` in project I see 120 appearances, majority of them are just copy. I would rather add the position update on this place.

And let's add filename to `update_position`.
https://github.com/Nic30/hdlConvertor/blob/master/include/hdlConvertor/hdlObjects/position.h#L25
like this https://github.com/Nic30/hdlConvertor/blob/master/src/notImplementedLogger.cpp#L13

But we have to be careful with the granularity, if we add the filename for every lets say operator, this could result in enormous number of strings, if this is a case we would rather want to set this filename string only on parent object and then add some functions so we can easily access this property also on children.

hs-apotell, 2019-11-11:
> make_unique call the constructor of the class. You mean to initialize position in constructor, right?

I was proposing a virtual constructor - a function that creates the required object and conditionally initializes the position and any other core members. 

> One problem I see with this is that it is sometimes required to create hdlConvertor::hdlObjects without the antlr context with the position etc.

We can do this conditionally. As long as these "special" cases are exceptions and not the norm, it is still worthwhile having a single point in code where "core" stuff is initialized for every constructed object.

> If I search ```make_unique<Hdl``` in project I see 120 appearances, majority of them are just copy. I would rather add the position update on this place.

Yes, that would be another way to do it. But requires significant code changes every time something changes related to object construction changes.

> But we have to be careful with the granularity, if we add the filename for every lets say operator, this could result in enormous number of strings, if this is a case we would rather want to set [...]

Have you considered adding a string table implementation? There are a few open source implementation available -
https://github.com/farsightsec/mtbl
https://github.com/santazhang/sstable

Disclaimer - I haven't personally used any of the above implementations.

Nic30, 2019-11-11:
> Yes, that would be another way to do it. But requires significant code changes every time something changes related to object construction changes.

OK lets add another templatized constructor with ctx, or static method which will return `unique_ptr`

> Have you considered adding a string table implementation?

Yes, we discussed it many times, string table is a "must have" however the `line` directive has to be done first because otherwise after preprocessor code may migrate between files and so on.
The implementation is not a problem it is just set where each string has assigned id.

Thomasb81, 2019-11-11:
I think what have been previously discuss is symbol table. But string table in this context is different I think.

I understand "string table" a structure that would allow to record "string" that may be reference several time, in order to save memory foot print. I see that as a coding trick.
The symbol table is more direct application oriented since it record the symbol name and would allow to build a hierarchy tree.

Nic30, 2019-11-11:
@Thomasb81, true symbol table was discussed intensively. The string table was discussed when in some conversation about how to cache the parsed results and how to store them.

Anyway the string table is reason behind the  [HdlName class](https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/hdlAst/_expr.py#L20) in future it will be a reference to a string table.

Symbol table is also planned but we do not have work force to implement whole SV, VHDL type system right now.

Thomasb81, 2019-11-11:
Ok, I did not capture this.

Nic30, 2020-03-31:
positional data should be present on all objects, but is is not covered by tests, we need to pick a representative sources and write the tests, we can use https://github.com/Nic30/hdlConvertor/blob/verilog_pp_line_directive/hdlConvertor/to/hdl_ast_visitor.py#L10 to walk all objects easily.

Nic30, 2020-04-26:
Closing this issue as positional info is present and we probably forgot to close this issue.

closed 2020-04-26 by Nic30