Nic30/hdlConvertor #93: Symbol clash between Python 3.x and ANTLR 4.6+ runtime

fredrequin, 2019-11-14:
While trying to compile hdlConvertor under Cygwin 64-bit, I have encountered this problem:
`[93/95] Building CXX object hdlConvertor/CMakeFiles/_hdlConvertor.dir/_hdlConvertor.cxx.o
FAILED: hdlConvertor/CMakeFiles/_hdlConvertor.dir/_hdlConvertor.cxx.o
/usr/bin/c++.exe  -D_hdlConvertor_EXPORTS -I/usr/include/python3.6m -I../../../hdlConvertor -I../../../hdlConvertor/../include -I/usr/local/include/antlr4runtime -I../../../src/../include -Wa,-mbig-obj -O3 -DNDEBUG   -std=gnu++1z -MD -MT hdlConvertor/CMakeFiles/_hdlConvertor.dir/_hdlConvertor.cxx.o -MF hdlConvertor/CMakeFiles/_hdlConvertor.dir/_hdlConvertor.cxx.o.d -o hdlConvertor/CMakeFiles/_hdlConvertor.dir/_hdlConvertor.cxx.o -c hdlConvertor/_hdlConvertor.cxx
/usr/include/python3.6m/Token.h:15:17: error: expected identifier before numeric constant
 #define STRING  3
                 ^
/usr/local/include/antlr4runtime/tree/xpath/XPathLexer.h:13:13: note: dans l'expansion de la macro « STRING »
     ID = 7, STRING = 8
             ^~~~~~
`
I was running "python3 setup.py install" with ninja installed.
I have tried combinations of ANTLR v4.6, / v4.7 / v4.7.2 and Python 3.6 / 3.7.
It looks like the interface changes in ANTLR were done after v4.5.3 but the lack of CMake script in v4.5.3 prevented me from compiling the runtime.
Can you clarify which Python and ANTLR versions should be used ?
Regards,
Frederic Requin

fredrequin, 2019-11-14:
It looks like Py_LIMITED_API must be defined to avoid this issue.
In /usr/include/python6.6m/Token.h, we have:
`/* Token types */`
`#ifndef Py_LIMITED_API <---`
`#ifndef Py_TOKEN_H`
`#define Py_TOKEN_H`

Nic30, 2019-11-14:
Hello, the tested python and ANTLR versions are defined in https://github.com/Nic30/hdlConvertor/blob/master/.travis.yml and https://github.com/Nic30/hdlConvertor/blob/master/.appveyor.yml.

Note that on Windows you can build natively and you do not have to use Cygwin.
I do recommend using Python3.7 + ANTLRv4.7.2 (latest in both cases).

However, if definition of Py_LIMITED_API will help for Cygwin will help, let's add it for the Cygwin build and lets add Cygwin build to appveyor.
However I am not sure about what Py_LIMITED_API actually does and if it should be modified by user. It looks like broken include order or something.

fredrequin, 2019-11-14:
> 
> 
> Hello, the tested python and ANTLR versions are defined in https://github.com/Nic30/hdlConvertor/blob/master/.travis.yml and https://github.com/Nic30/hdlConvertor/blob/master/.appveyor.yml.
> 
> Note that on Windows you can build natively and you do not have to use Cygwin.
Well, I do not want to depend on MSVC and I already have a nice Cygwin environment with Verilator, GHDL and Yosys.
> I do recommend using Python3.7 + ANTLRv4.7.2 (latest in both cases).
With Cygwin, I am (officialy) stuck with Python3.6. I have installed Python3.7 with some hacks and it fails the same way.
> 
> However, if definition of Py_LIMITED_API will help for Cygwin, let's add it for the Cygwin build and lets add Cygwin build to appveyor.
I have already changed some CMakefile such as the fix for big objects in 64-bit mode. I did not commit the changes yet.
> However I am not sure about what Py_LIMITED_API actually does and if it should be modified by user. It looks like broken include order or something.
It did actually fix the issue in my case but I am not sure about which CMakefile to change.

Overall, it is true that Cygwin compile is a pain:
- A shell wrapper has to be written for java.exe and javac.exe
- ANTLR runtime API has to be compiled
- python36.a library must be created with dlltool
- scikit-build must be changed
- CMakefiles have to be changed too
But at the end, I am pretty happy to have this very promising tool.

Thank you

Frederic Requin

Nic30, 2019-11-14:
What has to be changed in scikit-build and CMakefiles?
Do you think you would be able to write steps how to build under Cygwin for others?
It would be also nice if we can build without Cygwin and just use it in Cygwin (I do not know if it is possible).

I also think that simple things should not require a Docker, but if I should choose docker or Cygwin...
Why do you need Cygwin and can not use native build?

fredrequin, 2019-11-14:
skbuild/platform_specifics/platform_factory.py does not have support for Cygwin yet.
Yes, I can write a README for each steps : Java install, Java wrapper, ANTLR install, ANTRL runtime, Python library, Python modules, scikit-build patch, CMakefile patches (can be added in your repo).
Most of those steps can be automated in a shell script, just the Java install (under Windows) has to be manually done.

Nic30, 2019-11-14:
How did you obtain the ANTLR dll for Cygwin isn't there a version without version string "4.7.2" inside?

About Py_LIMITED_API  can we add it using `-D` ? rather than in code? Because it seems to me that it has to be before each `Python.h, antlr*.h` include, and that is quite lot of places.

fredrequin, 2019-11-14:
Hello,
the first step was to re-install java in some folders without space in the name : 
I took the version 8u212-b10 from https://github.com/frekele/oracle-java/releases
and install it under C:\jdk1.8.0_212 and C:\jre1.8.0_212.
I set the %JAVA_HOME% and %CLASSPATH% environment variables accordingly.

The second step is to translate path names for java.exe and javac.exe using two scripts under /usr/local/bin : 

`$ cat /usr/local/bin/java`
`#!/bin/bash`
`NEW_ARGS[0]=''`
`IDX=0`
`for ARG in "$@"; do`
`    # Make sure this is a unix path`
`    if [[ $ARG == *"/"* ]]; then`
`        NEW_ARGS[$IDX]=$(cygpath -mp "$ARG")`
`    else`
`        NEW_ARGS[$IDX]="$ARG"`
`    fi`
`    let IDX=$IDX+1`
`done`
`ARGS="${NEW_ARGS[@]}"`
`/cygdrive/c/jdk1.8.0_212/bin/java.exe -Xmx500M $ARGS`

`$ cat /usr/local/bin/javac`
`#!/bin/bash`
`NEW_ARGS[0]=''`
`IDX=0`
`for ARG in "$@"; do`
`    # Make sure this is a unix path`
`    if [[ $ARG == *"/"* ]]; then`
`        NEW_ARGS[$IDX]=$(cygpath -mp "$ARG")`
`    else`
`        NEW_ARGS[$IDX]="$ARG"`
`    fi`
`    let IDX=$IDX+1`
`done`
`ARGS="${NEW_ARGS[@]}"`
`/cygdrive/c/jdk1.8.0_212/bin/javac.exe $ARGS`

Third step is to install the ANTLR4 tool : 
`mkdir /usr/share/java`
`cd /usr/share/java`
`curl -O https://www.antlr.org/download/antlr-4.7.2-complete.jar`
`mv antlr-4.7.2-complete.jar antlr4-complete.jar`

Fourth step is to compile the runtime library : 
`cd ~`
`git clone https://github.com/antlr/antlr4`
`cd antlr4/runtime/Cpp`
`mkdir build`
`cd build`
`cmake .. -DANTLR_JAR_LOCATION=/usr/share/java/antlr4-complete.jar`
`make`
`make install`

Quite a few head scratchings to get to this result...

Frederic

Nic30, 2019-11-14:
Do you know if there is some Java version for Cygwin which does work without such a hacks?
We can download the jar in cmake. But we can not fix java for Cygwin. What has changed in java executables in `/usr/local/bin/`?

fredrequin, 2019-11-14:
Nope, I did not find any but I found a lot of forum topics about the filepath issue of java.exe under Cygwin.
At first, I started patching the Makefiles and CMakefiles but I found easier to put this wrapper in my Cygwin install.
BTW, swapping the include order does work. One less hack to make it compile :-)

Frederic

Nic30, 2019-11-15:
OK so for the Cygwin support:

* include order swap https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/CMakeLists.txt#L9
* antlr jar wget
* cygpath for Java paths.
* Cygwin build on appveyor https://github.com/Nic30/hdlConvertor/blob/master/.appveyor.yml

Nic30, 2019-11-15:
Is the problem with " " inside Java path caused by Cygwin env or it is because some path in cmake is not wrapped with "" or something like this?

fredrequin, 2019-11-15:
It is caused by Cygwin.
Unix path should be changed to Windows path with cygpath because java.exe is a Windows app.

Nic30, 2019-11-21:
48ba93df961d648a251fba408fd4acfe6c61d3bf

Nic30, 2019-11-21:
OK, now it would be perfect if we can do a cygwin build on appveyor so we can check if everything is working fine. The cygwin is already installed so it is required to install python for cygwin, antlr and this library.

Nic30, 2020-04-26:
@fredrequin this should be fixed, we probably forgot to close this issue, please reopen it if there is still some problem.

closed 2020-04-26 by Nic30

t0mj0nes, 2020-06-18:
Not clear what the resolution was - should "python3 setup.py install" work now or are there a bunch of hand hacks still required?

Nic30, 2020-06-19:
> Not clear what the resolution was - should "python3 setup.py install" work now or are there a bunch of hand hacks still required?

Hello @t0mj0nes, `python3 setup.py install` should work as well as 
`python3 setup.py install -j$(nproc)` or
`pip3 install hdlConvertor`.

Just remember that you have to install apt dependencies (
`apt install build-essential uuid-dev cmake default-jre python3 python3-dev python3-pip libantlr4-runtime-dev antlr4`
) and `pip3 install scikit-build` if you are not using (new) pip.

Do you have any problem with an installation?

t0mj0nes, 2020-06-20:
I have nothing but problems installing on Win10 - with or without using cygwin.  Perhaps someone could write some detailed instructions. "Look at the Appveyor.yml" is not enough for me to fully grasp the process. Thanks.

Nic30, 2020-06-20:
@t0mj0nes , sure but what should we put in that tutorial?

* native installation:
   * installation of visual studio/MinGW/... cmake, python, antrlr, java
* docker installation
* installation in [ubuntu subsystem for windows](https://docs.microsoft.com/en-us/windows/wsl/install-win10) 

Docker and wsl installation is relatively straight forward but it may be only another complication as it is actually not installed natively in system.

t0mj0nes, 2020-06-20:
My preference would be native with cygwin, but visual studio community would also suit my need. I will need to deploy this to hundreds of machines eventually. Thanks.

Nic30, 2020-06-21:
@t0mj0nes what about conda and others do you use some package manager for windows?

* theoretically I can just create conda package and you can just install it (without need for any tutorial)