Nic30/hdlConvertor #95: Support for static library build

hs-apotell, 2019-11-14:
Add support for static library build - for use in native tools without bringing in python dependency.

Nic30, 2019-11-14:
```
$ mkdir build
$ cd build/
$ cmake ..
...
make -j8
...
[100%] Linking CXX shared library libhdlConvertor.so
[100%] Built target hdlConvertor_cpp_shared
```
On windows this should produce static library.
(Because public API is not marked for DLL export.)

Are you having troubles with this kind of build?

hs-apotell, 2019-11-14:
No that doesn't work because to build static libs ```HDLCONVERTOR_PYTHON``` has to be defined and that brings in dependency on Python. Moreover, hdlConverter/CMakeLists.txt has an explicit dependency on Cython and other python extensions that aren't wrapped in any conditional statement.

CMakeLists.txt will have to be updated to support building static libraries only.

Nic30, 2019-11-14:
Now I see.

Nic30, 2019-11-21:
@hs-apotell one small thing, you want a static build for windows or linux?

What I am trying to ask is if you actually need a static build so your users do not have to care about an extra lib or it is because there is no package in any public repo for C library and the more correct solution is to make a apt,conda, etc package.

hs-apotell, 2019-11-22:
We are building for Windows, Linux (multiple different flavors) and Mac, so we need for all these platforms. That said, I don't expect you to support all these platforms. If one is supported we can port for others. It's basically just a way to ensure this particular configuration is validated and tested in your release cycle.

We plan on linking it always as a static library to avoid dynamic library nightmares. Also, we use custom memory management and that's a whole lot more maintenance when dealing with dynamic modules. Also, we will always build it our self because of the number of different platforms we have to support and so release packages are not very attractive for our use case.