Nic30/hdlConvertor #96: Elaboration?

hs-apotell, 2019-11-14:
Are you planning on implementing elaboration as well. Is the ultimate goal to output a synthesizable design? Os is that outside the scope of this project?

> 1364-2005: Elaboration is the process that occurs between parsing and simulation. It binds modules to module instances, builds the model hierarchy, computes parameter values, resolves hierarchical names, establishes net connectivity, and prepares all of this for simulation. With the addition of generate constructs, the order in which these tasks occur becomes significant.

Nic30, 2019-11-14:
https://github.com/Nic30/hdlConvertor/issues/51#issuecomment-553518058
> analysis and visualization tools for HWT to VHDL/Verilog ecosystem

Yes. The deadline is Christmas. But I think this will could be actually very complicated and as it is not my PhD topic, I am working on this project during waiting on test results and so. So I bet it can take longer.

Thomasb81, 2019-11-14:
My interest is code analyze to extract what is useful and interesting for IP reuse and verification.

Now it's open source project. If someone find useful to build on top of this library a simulator, synthesis or any tool it would mean that we would have been not so bad at software design. ^^

Most of other major open source tool have hdl parser customized for their application making difficult to re-use for other even basic purpose.

Nic30, 2019-11-14:
Actually, after the update of HWT API for hdlConvertor it would be possible to simulate the HDL in https://github.com/Nic30/pycocotb/tree/master/pycocotb/basic_hdl_simulator and others. It is actually not so far.

hs-apotell, 2019-11-14:
My interest is primarily in the Verilog and SystemVerilog parsers. I intend to use it for building a verification tool. But I need a synthesizable netlist and that would need elaboration. I don't need any optimizations or simplification logic. I might explore doing it myself if my schedule permits.

There are few NotImplemented exceptions that still need to worked out in the parser. At the moment, the parser fails to parser some of our internal designs (these use ```fork/join``` which is not implemented yet). I am still validating the parser against all the other files.

Thomasb81, 2019-11-14:
Did we miss some` fork / join `construction syntax ?

Nic30, 2019-11-14:
Yes, it is not converted to python https://github.com/Nic30/hdlConvertor/blob/master/src/svConvertor/statementParser.cpp#L199

hs-apotell, 2019-11-15:
Would it be too much of an ask to finish the parser work independently of whether or not it is Python compatible?

Thomasb81, 2019-11-15:
@hs-apotell could you disclose your file or reproduce a testcase that you can attach to a new issue ?

We did your best to test all syntax that are depict in Ieee document. But unfortunately the grammar is so complex that not all possible syntax have example, and we probably miss some...

Nic30, 2019-11-15:
@hs-apotell On C++ level there are 2 things.

* It is possible to use ANTLR parsers directly and you will get full language support, all code positions etc.
* Or you can use `hdlConvertor::Convertor` 

All parsers in C++ are independent of Python. Conversion to Python is [straigforward](https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/toPy.h)

Currently the conversion throws error if it does not underestand the object. But we can "hotfix" that for new object so we can focus on C++ first.

I do not see problem in finishing C++ version firsts.

@Thomasb81 the syntax is parsed correctly, it is just missing in Python because the conversion is not implemented.

Thomasb81, 2019-11-15:
@Nic30  It is not so clear to me. 

By the past the big 3 simulator had some divergence of behavior on the same SV code. It is still possible that @hs-apotell code use a simulator specific code or that we miss something... Remember the BNF was not 100% correct.

To judge the case, we need a testcase.

Nic30, 2019-11-15:
@Thomasb81 it is parsed correctly, but it is missing in the parsed result.

hs-apotell, 2019-11-15:
> could you disclose your file or reproduce a testcase that you can attach to a new issue ?

Created Issue #98

Nic30, 2020-08-15:
@mewais is now working on elaboration (as a hobby project)

hs-apotell, 2020-08-26:
> @mewais is now working on elaboration (as a hobby project)

Does this extend to inferencing?

mewais, 2020-08-27:
@hs-apotell depends on what you mean by inference?

I plan to do constant and generic substitution, expansion of generate statements and loops, expansion of subprograms/functions, optional hierarchy flattening. Basically take a complex HDL and spit out a netlist. However, any netlist simplification is not part of the plan. So no FSM optimization, no boolean minimization, etc.

By the way, just like Nic I'm also a PhD student and this is my side project. Which means my progress won't be as quick as I'd like it to be.

hs-apotell, 2020-08-27:
@mewais By inferencing I meant identifying higher level constructs like flip-flops, inverters, etc. From your answer, it doesn't look like that is in your plan for now. This functionality is required to be able to visualize the design as a meaningful schematic diagram.

Generating a netlist is a good start but there is lot more to be done.

Nic30, 2020-08-27:
I could potentially hire someone or make thesis topic out of it.
However I can not do it just because it would be nice.
I need some company to explicitely request for this feature.
(There are several users asking for this but I do not know if they represent some larger entity or not. )

Nic30, 2020-08-27:
@hs-apotell About construct identification for schematic: https://github.com/Nic30/hwtGraph/blob/master/hwtGraph/elk/fromHwt/statementRenderer.py

After elaboration is somehow working I possibly can reuse some code for detection you require.

mewais, 2020-08-27:
@hs-apotell my plan definitely includes inference of constructs like muxes, decoder, FFs, etc. Otherwise you cannot reach an output netlist correctly. 
However, given the order of other things I mentioned, which kinda count as preprocessing and have to happen before inference, it's going to be a while before I get there.