Nic30/hdlConvertor #97: Implement __str__ /__repr__

Thomasb81, 2019-11-14:
It would be nice to retrieve something like the initial json output.
It was very useful to discover what is inside object...

Actually such script:

```
import sys
from hdlConvertor.language import Language
from hdlConvertor import HdlConvertor

include_dirs = []
c = HdlConvertor()

filenames = ['tests/verilog/uart.v']

d = c.parse(filenames, Language.VERILOG, include_dirs, hierarchyOnly=False, debug=True) 

print(d)
print(d.objs)
```
produce
```
<hdlConvertor.hdlAst._structural.HdlContext object at 0x7f412f974550>
[<hdlConvertor.hdlAst._structural.HdlModuleDec object at 0x7f412edf5f48>, <hdlConvertor.hdlAst._structural.HdlModuleDef object at 0x7f412edde828>]
```
A prettyPrint would be excellent.

Nic30, 2019-11-15:
Ah, this was postponed so many times. I am thinking if there is some automatic way how to do this. It should be possible to use __slots__ and __dict__ in universal method in https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/hdlAst/_bases.py#L5

Thomasb81, 2019-11-15:
Yes I start to look at re-use of __slot__ and __dict__ 
But I had some difficulties or did not manage to find the good implementation for the time being. So for the moment it does not produce the expected result.

Nic30, 2019-11-15:
OK, I will look in to it tomorrow if I will be able to find some time.

Nic30, 2020-04-02:
> I will look in to it tomorrow

@Thomasb81 it seems that the tomorrow is no different for me than next year, but at least it is done.

Thomasb81, 2020-04-03:
No problem,  software is available as is, no support guaranty ... 

I must give a trial. then close this ticket.

Nic30, 2020-04-26:
@Thomasb81 you can try it in one click https://mybinder.org/v2/gh/Nic30/hdlConvertor/master?filepath=notebooks%2F01_parse_and_dump.ipynb

Thomasb81, 2020-04-26:
Look cool! 
Thanks

closed 2020-04-26 by Thomasb81