Nic30/hdlConvertor #102: VHDL library statement not supported in Python

rhinton, 2020-02-21:
Pretty much every VHDL file will have "library IEEE;" near the beginning, but I get an error that conversion to a Python object isn't implemented.

[vhdl_lib_ex.tar.gz](https://github.com/Nic30/hdlConvertor/files/4236608/vhdl_lib_ex.tar.gz)

The attachment includes a simple VHDL file (adder.vhd) and Python script based on the "Usage" snippet on the README.

I know this project is under development.  I'm happy to see if I can provide a patch.  Do you have a "generic HDL" (Python) object corresponding to a library declaration?  Does one need to be created?  Any suggestions for how to help with this feature (and learn the code base)?

Thanks!

Nic30, 2020-02-21:
Hello, yes the `library IEEE;` is basically in each file. However it is usually followed by `use IEEE.*;` thus the library statement can be derived from use statement.

I am not sure if this is wrong or good. We can add a library statement it is not a problem. There is a list of things which has to be done.

1. create a c++ library statement class https://github.com/Nic30/hdlConvertor/tree/master/include/hdlConvertor/hdlObjects
2. parse it in https://github.com/Nic30/hdlConvertor/blob/master/src/vhdlConvertor/designFileParser.cpp#L127
(library_clausule is just list of names: https://github.com/Nic30/hdlConvertor/blob/master/grammars/vhdlParser.g4#L1129)
3. create a python library object https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/hdlAst/_statements.py
4. add conversion from c++ object to a python object https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/toPy.h#L143

However I still do not know it is necessary.
Do you know about any case where library statement can not be derived from use statements?

rhinton, 2020-02-21:
Well, I know the language *requires* it to be there for any library except `STD`.  That exception suggests to me that it's easier to model it than to add exceptions to the parsing and generation code.  Also VHDL-2008 context clauses can change which libraries are included or not by default or by declaration.

Thanks for the detailed to-do list. :-)  I'll take a look!

rhinton, 2020-02-28:
I created a pull request with code that works for me, but still needs a little work (esp. testing).  I look forward to that conversation there.

I've thought further about *why* to have the VHDL library construct.  

* VHDL is very explicit.  Before you can use a library named `IEEE`, it needs to be declared.  

* A VHDL library corresponds to a container of some sort (usually a file or directory in the file system) that holds the tool-specific compiled results for the design units in that library.  The effect is kind of like a Java JAR.

* VHDL libraries are a good place to store things that don't change very often like FPGA/ASIC libraries and third-party cores and collections of code like OSVVM or PoC.

* In order to properly generate typical VHDL code you need to be able to generate them.

I agree that a parser or AST tool can *infer* that the first element in a `use` statement is (usually) the library name.  But a correctly-formed design file needs the library declared first _except_ for the `STD` library.  And a selected name (e.g. from a use statement) need not always include the full path.  I'm pretty sure the following is valid, though I don't have a good compiler handy to test at the moment.

`library IEEE;
use IEEE.std_logic_1164;  -- this brings the package name into the current namespace
use std_logic_1164.all;  -- this pulls the package contents into the current namespace
-- notice that the second statement doesn't start with a library name`

Given the exceptions and corner cases, I think it's easiest to model the full language.

Nic30, 2020-02-29:
>  But a correctly-formed design file needs the library declared

A valid argument, I was blind and was thinking about VHDL interpretation and thus I did not see the need for validity checking.

rhinton, 2020-03-13:
I'm happy with the result, so closing this issue.  Let me know if you would rather me wait for a project member to do this.

closed 2020-03-13 by rhinton