Nic30/hdlConvertor #107: VHDL type declarations not being parsed

rhinton, 2020-03-13:
Looking at the code, it appears that VHDL type declarations are not being parsed.

[https://github.com/Nic30/hdlConvertor/blob/2d17b33991ffc0119ec7ded3c6526baba3077591/src/vhdlConvertor/packageHeaderParser.cpp#L109](url)

For VHDL, at least, data types are critical to even very basic operation (e.g. `std_logic_vector`).  So data type parsing needs to be implemented.

rhinton, 2020-03-13:
It looks like the SystemVerilog area has *some* provisions for parsing data types, but the interesting ones (enums, structs, unions) are not implemented.

[https://github.com/Nic30/hdlConvertor/blob/2d17b33991ffc0119ec7ded3c6526baba3077591/src/svConvertor/typeParser.cpp#L169](url)

Do you want VHDL data types and SV data types to be parsed into a common "data type" hdlObject, or should they have their own types?  From past conversations, I suspect the answer is for each language to have its own data type concept, and the convertor mechanism will know how to translate between them.

Nic30, 2020-03-13:
> https://github.com/Nic30/hdlConvertor/blob/master/src/svConvertor/typeParser.cpp#L169
> https://github.com/Nic30/hdlConvertor/blob/master/src/vhdlConvertor/packageHeaderParser.cpp#L109

Lets implement conversion at least for essential data types like arrays, enums, int ranges, subtypes, classes. This corresponds to following rules.

https://github.com/Nic30/hdlConvertor/blob/master/grammars/vhdlParser.g4
```
type_declaration:
      full_type_declaration
      | incomplete_type_declaration
;
full_type_declaration:
      KW_TYPE identifier KW_IS type_definition SEMI
;
type_definition:
      scalar_type_definition
      | composite_type_definition
      // | access_type_definition
      // | file_type_definition
      // | protected_type_definition
;
scalar_type_definition:
      enumeration_type_definition
      | integer_type_definition
     // | floating_type_definition
     // | physical_type_definition
;

enumeration_type_definition:
      LPAREN enumeration_literal ( COMMA enumeration_literal )* RPAREN
;
enumeration_literal: identifier | CHARACTER_LITERAL;
integer_type_definition: range_constraint;

composite_type_definition:
      array_type_definition
      | record_type_definition
;
array_type_definition:
      unbounded_array_definition | constrained_array_definition
;
```
subtype_declaration -> variable of TypeType
https://github.com/Nic30/hdlConvertor/blob/master/grammars/sv2017Parser.g4
```
    | class_declaration
    | interface_class_declaration

data_type:
    ( data_type_primitive
        | KW_ENUM ( enum_base_type )?
           LBRACE enum_name_declaration  ( COMMA enum_name_declaration )* RBRACE
        | struct_union ( KW_PACKED ( signing )? )?
            LBRACE ( struct_union_member )+ RBRACE
        | package_or_class_scoped_path
      ) ( variable_dimension )*
 ;
```

> Should VHDL data types and SV data types to be parsed into a common "data type" hdlObject, or should they have their own types?

It would be beneficial if we can use less code to do more work.
I think we can represent vhdl record, sv struct, union and class by a single object with just some flag set.
I think we can do the same for enums.
Some type type already have it's hdlObject https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/hdlAst/_typeDefs.py#L84.  
However I think this agenda requires more detailed analysis, so think of this comment as a brainstorming.

rhinton, 2020-03-13:
Are there any current C++ classes to represent data types?  I haven't been able to find any.

Assuming not, do you have any preference between 1 class to represent any data type, a base class for all data types and subclasses for each class of data types (enum, struct, etc.), or simple independent classes for each class of data type?

rhinton, 2020-03-13:
Also, is there any pattern to why some objects have an 'i' prefix and some don't?  The comment in `iHdlObject.h` says that these objects may appear in the AST.  But only a few of the classes that inherit from `iHdlObject` (and do appear in the AST) have this same prefix.  Am I missing something?

I ask to better guess whether a new data type (base?) class should have the 'i' prefix. :-)

Nic30, 2020-03-14:
> 'i' prefix 

It was meant for interfaces. However I have to admit that the iHdlExpr is far from being an interface.

> Are there any current C++ classes to represent data types?

https://github.com/Nic30/hdlConvertor/tree/master/src/hdlObjects

No there are not.

> 1 class to represent any data type X a base class for all data types and subclasses for each class of data types (enum, struct, etc.), or simple independent classes for each class of data type?

I would prefer some base class lets say iHdlType and several type classes for a different type types (HdlEnum, HdlClass (for classes, records, structs, union)).

Nic30, 2020-05-13:
@rhinton Shall we close this issue now as the types are being parsed?

rhinton, 2020-05-17:
Absolutely.  I'll close this ticket for the current implementation.  If I find a further issue, I'll open a new ticket.  Thanks!

closed 2020-05-17 by rhinton