Nic30/hdlConvertor #108: Should hierarchy-only mode explore dependencies?

rhinton, 2020-03-13:
I'm not quite sure how the `hierarchyOnly=true` mode is supposed to work.  Is it simply parsing enough to identify the design units contained in each file (module, entity, package, etc.)?  Or is it *also* intended to identify dependencies of each design unit?

If the `hierarchyOnly=true` mode is supposed to identify dependencies, then the VHDL package parsing (package and package body) needs to be modified to include `use` clauses in the package (header) and package body.

Nic30, 2020-03-14:
> hierarchyOnly=true

It was meant for discovery of dependencies. Yes use clauses are missing as it was originally implemented only to discover instances without parsing anything else.

rhinton, 2020-04-01:
I'm closing this ticket since my question is answered.  When we're ready to fix the hierarchy mode, an appropriate ticket should be opened.

closed 2020-04-01 by rhinton