Nic30/hdlConvertor #112: Trying to get function names in a core dump

rhinton, 2020-03-28:
How can I get function names and line numbers in a core dump?

BACKGROUND:

I am getting a core dump when running the tests.

> $ python3-dbg -m tests.test_vhdl_conversion
> test_arch_with_assig (__main__.VhdlConversionTC) ... ok
> test_call (__main__.VhdlConversionTC) ... Segmentation fault (core dumped)

I have setup the core dumps to go to /tmp, which works.  But loading the core in GDB gives a useless call stack:

> gdb -c /tmp/core-python3-dbg.23539.mcube2.1585368553
> Core was generated by `python3-dbg -m tests.test_vhdl_conversion'.
> Program terminated with signal SIGSEGV, Segmentation fault.
> #0  0x00007f5aaf15e3c8 in ?? ()
> (gdb) where
> #0  0x00007f5aaf15e3c8 in ?? ()
> #1  0x00007fff6aae1800 in ?? ()
> #2  0x00007f5aaf117ae8 in ?? ()
> #3  0x00007fff6aae1800 in ?? ()
> #4  0x0000000000000010 in ?? ()
> #5  0x00007fff6aae1810 in ?? ()
> #6  0xfe2737c93dbb7100 in ?? ()
> #7  0x00007fff6aae1860 in ?? ()
> #8  0x00007f5aaf328b2a in ?? ()
> #9  0x0000000000000000 in ?? ()

INSTRUCTIONS:

I am doing the following, and still getting this useless core dump.

> python3 setup.py clean
> rm -R build/*
> cd build
> cmake ..
> cd ..
> python3-dbg setup.py build --build-type Debug -j 20 && sudo python3-dbg setup.py install --build-type Debug -j 20
> cd hdlConvertor
> rm _hdlConvertor*.so
> ln -s ../_skbuild/linux-*/setuptools/lib/hdlConvertor/_hdlConvertor.*.so . 
> cd ..
> python3-dbg -m tests.test_vhdl_conversion

I know to add "-g -O0" on a g++ command line to get great debug information, but I know very little about `cmake` and `setup.py`.

Nic30, 2020-03-28:
Try: https://github.com/Nic30/hdlConvertor/blob/master/doc/dev_minitutorial.md#debuging-in-eclipse

If there is a segfault in python during module load it is probably because the module does not define some c++ symbol (= c++ function definition etc). For this you need to use the python3-dbg in order to see an error msg. If it breaks after some function execution it is probably segfault and you can use stacktrace, valgrind etc.

There is also the scikit-build option `--build-type Debug` https://scikit-build.readthedocs.io/en/latest/usage.html but it is not reliable as adding the -g manually.

Thomasb81, 2020-03-28:
Hello

I think python-dbg is just a python version with debugging symbol and no compiler optimization.
I don't think you are tracking an issue in the python binary.  

What I did to debug:

I build hdlConvertor library with "--build-type Debug" and VERBOSE=1 environment variable
This should compile with "-g -O0" option and print compilation line in the terminal. So you can check options are ok.

Then my debug command line was:
gdb --args <path_to_python> <path to myscript.py with import hdlConvertor>

rhinton, 2020-04-03:
Last week the `gdb --args` approach failed.  Today it works. :-)  Thanks!

Using the `gdb --args` approach, GDB stops at the segfault including full context.  To be explicit, my command line is

> gdb --args /usr/bin/python3-dbg -m tests.test_vhdl_conversion

closed 2020-04-03 by rhinton