Nic30/hdlConvertor #114: Possible duplication in expression and value

rhinton, 2020-04-03:
I see that `HdlValue` can be a `BigInteger`, and `iHdlExpr` can be initialized with a `BigInteger` or `HdlValue`.  Is there a reason for this duplication?

rhinton, 2020-04-03:
Update: Looking in the code, the `iHdlExpr` constructor that takes a `BigInteger` simply wraps it in an `HdlValue`.  I assume the duplicate code paths are for convenience, right?

https://github.com/Nic30/hdlConvertor/blob/83ed705d37e4c5962e75b3b4a1e5a255230c44be/src/hdlObjects/iHdlExpr.cpp#L48

Nic30, 2020-04-03:
>  for convenience

Yes, but also I do see it as a suboptimal solution, let's merge all in progress work in some branch and let's create a subclass of `iHdlExpr` for each value type and use it instead of `iHdlExpr.data = new HdlValue...` The iHdlValue was meant for something different and currently it only shadows the things.

Nic30, 2020-04-26:
depends on #115

Nic30, 2020-04-28:
closing as the referenced code does not exists any more after 4ee2378

closed 2020-04-28 by Nic30