Nic30/hdlConvertor #116: Architecture and toX discussion

rhinton, 2020-04-10:
I like the new `toVerilog` and `toVhdl` implementations based on running the Python classes through `toJson`.  Just to be clear, here is my current understanding of the parser chain.

Current: 
HDL source -> ANTLR lex/parser -> C++ parser -> C++ data types -> `toPy` Python classes -> `toJson`  Python list, dict, string -> `toVerilog` or `toVhdl` source.

As I'm working right now to create the `toPy` infrastructure for several new classes, my next thought is: why not move the `toJson` back a step?  Why not have the C++ code create the JSON (or at least generic list, dict, string structures) instead of creating Python classes first?  As far as I can tell, the Python classes don't actually *DO* anything.  They're just containers.  If we did this, the process would look as follows.

C++ data types directly to JSON:
HDL source -> ANTLR lex/parser -> C++ parser -> C++ data types -> `toJson` C++ list, dict, string -> `toPyJson` Python list, dict, string -> `toPy` Python classes or `toVerilog` or `toVhdl` source.

This approach could greatly simplify the `toPy` implementation in particular.  We wouldn't need custom Python extension C++ code to convert every C++ class to its corresponding Python class and code all the Python classes in the extension, etc.  Instead, the Python extension C++ code could simply convert the strings, dicts, and lists to Python equivalents.  Then we can create the Python classes in Python, and translate "JSON" structures to Python classes *in Python*.  What do you think?

Of course, the next question is, why not move the `toJson` step back even a step further?  As far as I can tell, the C++ data types don't actually *DO* anything.  They're just containers.  If we did this, the process would look as follows.

C++ parser directly to JSON:
HDL source -> ANTLR lex/parser -> C++ parser -> `toJson` C++ list, dict, string -> `toPyJson` Python list, dict, string -> `toPy` Python classes or `toVerilog` or `toVhdl` source.

This again would get rid of gobs of code defining and creating all of the C++ classes.  

I expect that converting C++ JSON dict/list to Python equivalents (`toPyJson`), then implementing `toPy` in Python would cut the Python extension C++ code by 90% -- the most annoying 90%, too!  Including the `toPy` code, I expect the two-stage conversion to Python objects would take only 25% as much code.

Question: Do we need the Python classes if the `toVerilog` and `toVhdl` aren't using them?

I expect that skipping the C++ classes and parsing directly to generic list/dict objects instead of custom C++ classes would reduce the C++ code base by 50%.

Thoughts?

Nic30, 2020-04-10:
> HDL source -> ANTLR lex/parser -> C++ parser -> C++ data types -> toPy Python classes -> toJson Python list, dict, string -> toVerilog or toVhdl source.

toJson **or** toVerilog or toVhdl  json is independent thing and the Python objects are transpiled directly to a target language
Also note that "C++ data types" should be equivalents of Python classes

> Why not have the C++ code create the JSON

I clearly see how much of additional work is to use a C++/Python classes rather than just "JSON" however the goal is to have structure of data checked and without anything unexpected. It is hard to achieve with just a simple JSON. Python objects are dynamic but the properties  are checked using `__slots__`. Types can also by specified in the code, JSON schema can not.

On the other hand it should be fairly easy to write a JSON equivalent of toPy as you say.

> Do we need the Python classes if the toVerilog and toVhdl aren't using them?

https://github.com/Nic30/hdlConvertor/blob/verilog_pp_line_directive/hdlConvertor/to/verilog/verilog2005.py#L106

All "to" transpilers are using them. We do have Python classes so there is at least some type checking.

> I expect that skipping the C++ classes and parsing directly to generic list/dict objects instead of custom C++ classes would reduce the C++ code base by 50%.

I also see the opportunity however I am really scared that if we remove a type system from this we wont be able to debug anything.

However I see a different problem. I would like to extract python classes for HDL object to a separate library. In order to test that library it is required to init the data structures somehow. As it is very time consuming to write AST by hand and we do not want to use c++ part of library we probably need to construct the Python HDL objects from JSON. And the direct json output from c++ coud be used to test the conversion to python objects.

Thomasb81, 2020-04-10:
> Why not have the C++ code create the JSON

If you return enough in the code history, there was a define maco for that. I never try it.

> As it is very time consuming to write AST by hand and we do not want to use c++ part of library we probably need to construct the Python HDL objects from JSON. And the direct json output from c++ coud be used to test the conversion to python objects.

What the advantage compare to write verilog, systemVerilog and vhdl code snippet for that ?
I think It will be as difficult and time consuming to write those json file than hdl source code for unit test purpose.

rhinton, 2020-04-10:
> Test the Python classes without the C++ part.

That sounds like a good idea, and could be a great use of a direct C++ -> JSON conversion.

> I am really scared that if we remove a type system from this we won't be able to debug anything.

I've used Python and Ruby enough that I'm not scared of this.  I have been impressed at how much good, correct, simple code I can write with just dict, list and string types.  Debugging the current design requires GDB/Eclipse to set breakpoints and open half a dozen files to have all of the types and parsers available.  Debugging the dict/list/string system could pretty-print the current node at any point in the parser.  As far as I can tell, the current type system makes 10x the code to debug, and makes the debugging twice as hard because you have to understand the parser *and* the type structure (*and* the Python type structure, *and* how to write a Python C++ extension, *and* ...).

The way I see it, the ANTLR parser converts the source code into a highly-specific data structure.  The C++ code converts this into a slightly-less specific data structure.  The JSON really has all of the same information as the C++ data structure: the class, doc, position, name, and all of the class-specific data by name in a dict.  The current C++ classes with constructor, destructor, and public member variables aren't much different from a dict/list/string approach.

Really, I don't see much difference between the JSON approach and `iHdlExpr` where any kind of expression gets slammed into a class with a 'type' field and a bunch of member variables.  It's kind of like a C++ class-union hybrid.  If anything, the JSON approach is cleaner.  It becomes easy to make the 'class' field as specific as you want.  For example, instead of of creating an `iHdlExpr` and marking it as 'others' or 'any type' or a `BigInteger` or something else, we could create the dict with an 'expression' class, and fill in the `'others' => True` or `'anyType' => True` or `'integer' => '267846'` or whatever.  This would make the C++ parsing code (that interacts with the ANTLR result) very easy to write!

The dict/list/string structure also seems perfect for manipulation -- possibly multiple "passes".  For example, perhaps we can find an easy way to auto-convert the ANTLR result to the dict/list/string structure (maybe, maybe not).  If we had this highly-specific structure (a different class for every BNF production in every language), the C++ "parser" would then be a simple compiler "pass" to translate the too-detailed, language-specific dict/list/string structure into a HDL-agnostic dict/list/string structure.  The "hierarchy only" function could be implemented as another compiler pass.  (Although that might defeat the purpose of having a super-fast "hierarchy-only" parser mode that doesn't need to create the data structure at all.  It would be interesting to benchmark.)  The `toX` outputs could be implemented as compiler passes.  Making the data structures more generic (dict/list/string) seems like it adds power, not removes it.

I see a few disadvantages to the dict/list/string approach.

* No subclassing.  We may need to invent and enforce a mechanism for "subclassing".  Going back to the `iHdlExpr` example, we could make the '__class__' field as specific as we want.  But we almost certainly want to be able to handle all expressions in a particular way.  Or to know which kinds of object classes have a name.  It may be that the Ruby concept of "duck typing" may provide all the "class"-like information we need.  The basic idea is that, if this dict has a name, then whatever kind of object it is must be one of those that has a name.  And so code that operates on it, instead of `dynamic_cast<HdlObjWithName*>` and check for null, would check to see if the 'name' attribute is defined.  This concept goes a long way in Ruby, and I've really enjoyed using it.

* Documentation.  Right now the C++ and Python classes are a big pile of code that has to be written, tested, maintained, converted, etc.  But at least it documents what structures are available.  With the dict/list/string approach, how does a maintainer know which '__class__' fields are allowed?  Especially if compiler passes as described above *change* the '__class__' value?  The obvious answers are (a) documentation and (b) testing.  And good testing can serve as documentation.  Personally, I would rather write more tests and maintain a Markdown page or two that documents the classes rather than maintain dozens of C++ header and source files.

Thoughts?

rhinton, 2020-04-11:
> What the advantage compare to write verilog, systemVerilog and vhdl code snippet for that ?

I don't understand.  What two things are you comparing here??

> I think It will be as difficult and time consuming to write those json file than hdl source code for unit test purpose.

Interesting, I hadn't thought about unit tests.  Still, I think this might be an improvement.  Currently the C++ can only be tested with the Python.  But if the parsed result can be written to a JSON file, it should be fairly easy for a human to examine and compare for correctness.

FYI, in the discussion above, when I say "JSON", what I really mean is an in-memory data structure composed of dictionaries, lists, and strings.  The current C++ classes would turn into dictionaries, and their member variables into keys in the dict.  I'm proposing that it could serve as the primary in-memory representation, at least for the C++ side.  A happy coincidence is that it would be easy to write to a JSON file.

Thomasb81, 2020-04-11:
> > >And the direct json output  from c++ coud be used to test the conversion to python objects.

> > What the advantage compare to write verilog, systemVerilog and vhdl code snippet for that ?
> 
> I don't understand. What two things are you comparing here??

Here, I was answering to the proposition to use json code to test conversion to python object. It is a significant work of specification to define intermediate representation and at the end you will end up to write json code by hand for unit test, which I expect is not easy. 

Aren't we looking for [reflection](https://en.wikipedia.org/wiki/Reflection_(computer_programming)) properties ?
My understanding is that we have it on python side, but it's costly to develop and maintain, because it's mainly mirror C++ class and we plan to change to keep structure of what is capture from hdl code. While it could be smart to have reflection directly in C++.

That's what seduce me in the early hour after discovering this project, with a simple pretty print I can observe my verilog module prototype captured in a "simple" and iterable structure  without having to dive in the c++ class tree structure. It allow quick development on top of the library.

Nic30, 2020-04-11:
> use json code to test

I was thinking about the way how to test the Python part without C++ part, it may be useful to be able to just load HdlObjects from json without parsing the Verilog/VHDL because this things may break and then it is not clear whenever the problem is in c++ or Python path.

> Aren't we looking for reflection properties ?

https://bytemaster.github.io/boost_reflect/group__boost__reflect__quickstart.html

>  with a simple pretty print I can observe my verilog module

Current __repr__ should provide that.

@ rhinton 

> duck typing in json instead type system, type system as a form of doc.

I am sure that the JSON can be used as an internal representation in C++ and/or in Python. HdlConvertor was intially implemented in that way. However conversion to a Python JSON (dict/str/int/list) was not fully automatic. Also the json is quite slower than C++ native objects (however the conversion to Python objects is even slower...)

The class based type system for HDL objects allows for quick modifications of data specification. The data format is also checked compile time. For me the biggest "+" is that I can quickly check tools that are using HdlConvertor and update the HdlConvertor API used in them. With JSON each user application was quite unique and it was harder to find out what did change and why it is not working any more.

rhinton, 2020-04-11:
I can see how the detailed model is faster and smaller.  But there are good ways to minimize these impacts if we decided to go this route.

I just took a quick look at Clang, and it appears to have a more detailed object model like HdlConvertor.  To me this is a Perhaps it's just that I'm fiddling with the C++ types that it seems like so much work!  

I assume external tools and compiler passes will be implemented with some kind of [Visitor pattern](https://en.wikipedia.org/wiki/Visitor_pattern) -- a pattern that I know and love.  The main limitation of the Visitor pattern is that the class hierarchy quickly ossifies as more and more visitor code is written than depends on it.  In other words, the more successful this project becomes, the harder it will be to improve it. :-)  As @Nic30 pointed out, this mismatch is easy to detect with the detailed object model because the visitor code no longer compiles.  Ultimately, though, there is no replacement for good tests, which would catch mismatches in either case (either detailed object model or generic dict/list/string).

So it seems critical to get the "right" object model as early as possible.  How do we know when we've found this "right" object model?  How can we tell if one model is better than another?

How does our current object model compare with other compilers like Clang or GCC?  Maybe the comparison between SW and HDL languages isn't particularly meaningful.  I wish I had a bulk of compiler experience I could bring to bear, but I've always been a compiler user instead of compiler writer.

Is the goal to make the HdlConvertor C++ object iterable/visitable?

Nic30, 2020-04-11:
> class hierarchy quickly ossifies

Do uou means that the json schema would ossificate quickly or that the visitor pattern is add only code?

object model of typical c compiler is quite different
https://llvm.org/devmtg/2017-06/1-Davis-Chisnall-LLVM-2017.pdf

The problem with verilog/vhdl is that there is too much of specific things event descriptions, SV verification stuff, assignment types,....

The object model of tools like this one is usually just verilog/VHDL AST

* https://github.com/verilator/verilator/blob/master/src/V3AstNodes.h 
* https://github.com/freechipsproject/firrtl/blob/master/spec/spec.pdf
* https://github.com/YosysHQ/yosys/blob/master/kernel/rtlil.h

>  How can we tell if one model is better than another?

Good question.

> Is the goal to make the HdlConvertor C++ object iterable/visitable?

There is a visitor class. It is already done and used in translators/to classes. 

https://github.com/Nic30/hdlConvertor/blob/verilog_pp_line_directive/hdlConvertor/to/hdl_ast_visitor.py#L10

rhinton, 2020-04-11:
Another interesting object model:
* https://ghdl.readthedocs.io/en/latest/internals/AST.html
I agree that a C/C++ compiler AST is going to be a very poor fit for an HDL.

Thanks for the Python visitor link.  How about the C++ object hierarchy?  Is it intended to be iterable/visitable?  If so, shouldn't the `toPy` framework be iterated as a visitor?

Thomasb81, 2020-04-11:
> > How can we tell if one model is better than another?
> 
> Good question.

It is difficult to think to a data model working for all problems. That's probably why EDA industry still propose different tools for each purpose: simulation, synthesis ... Data are process differently and required to organized differently for better run-time.

rhinton, 2020-04-13:
I just read [the GHDL AST doc](https://ghdl.readthedocs.io/en/latest/internals/AST.html) that I reference above a little more thoroughly.  The author says that, for each node type, the attributes and their types are fixed.  But writing code becomes much easier with a "meta-model" that allows describing and handling all nodes in a consistent way.  The concept is similar to @Thomasb81 reference to the RTTR library.  The GHDL document says that at least some of the source is generated from a Python script.

I plan to finish my current pull request (pull request #115), then look further into these issues.

This seems like a good ticket to continue the discussion.  I see a couple of possible directions for improvement.  The best solution may be a combination of several ideas.

* Python code and "meta-model" to generate some C++ and/or Python code.
* Reflection (RTTR or Boost::reflect or other) to enable a generic mechanism to visit/iterate over the C++ object model.  

Between these two ideas, my hope is to replace most of the `toPy` code with a few short functions/methods.

rhinton, 2020-04-13:
I just noticed a question from @Nic30 that I missed.

> Do you mean that the json schema would ossify quickly or that the visitor pattern is add only code?

I mean that a typical GoF visitor approach (not using reflection) will break if the object model (C++/Python/whatever) changes.  So as visitor code multiplies, it becomes difficult to impossible to modify the object model to better fit the data as our understanding improves, HDL specs change, etc.  So the object model becomes ossified, frozen at the point that visitors begin to be widely used.

It looks to me like a reflection-based visitor scheme can remain more flexible over time and change, so I'm favoring this approach.

Nic30, 2020-04-13:
It seems to me that we can relatively effortlessly decorate c++ hdl object classes with  boost::reflect.
We can also detect python HDL object classes, constructor args etc.

Based on this we can probably fully automatize c++ toX (we are most probably interested in c++ to Py/json). However I am guessing that it will save us around 500 code lines in toPy. That means not relatively much it would be good it has some another benefit as well. Maybe it can also allow as to create binding to other languages effortlessly. There we several users which asked for java/scala support, for example.

Nic30, 2020-04-13:
@Thomasb81  #115 

rttr looks cool, it has also package in Linux repositories which is nice. From my point of view it is probable that the user will already have libboost installed. But most important question is what exactly we can do with reflection, automatically translate properties?

For c++ toPy we need to resolve a python class from class name, for c++ toJson we need to add class name for some object. There will be also some non class values which may require different handling on context. So we may actually still end up with a method for each object type.

Thomasb81, 2020-04-14:
I see reflection as some kind of documentation substitution, since you can just query or print what is available in your data structure on your particular example.

Data structure update is simplified, wrapper can be generated.

Nic30, 2020-04-26:
I just added a https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/parse_hdlConvertor_json.py#L71 it loads hdl objects from json produced by ToJson, it is really small piece of code but it can load everything. We may want to do something similar in C++ toPy if we had a reflection.

Nic30, 2020-04-28:
I have found an easy way how to implement property iterator in C++ and thus automatically convert hdl ast classes to python/json etc. http://coliru.stacked-crooked.com/a/a5902693e51b59b9

rhinton, 2020-05-17:
We can probably close this ticket as well since @Nic30 reworked the toX infrastructure.  If we want to add reflection or such, it probably belongs on another ticket.

Nic30, 2020-05-17:
ok

closed 2020-05-17 by Nic30