Nic30/hdlConvertor #119: The example runs failed in read.md

lionheart117, 2020-04-30:
Hi, Nic
Yesterday I followed every instructions from READ.md to install this lib. But when I ran the example of usage, the Python3 interpretor told me that it could't find a lib called "hdlConvertor.to" from some imports showing below:

"
from hdlConvertor.language import Language
from hdlConvertor.to.verilog.verilog2005 import ToVerilog2005
from hdlConvertor import HdlConvertor
"

Nic30, 2020-04-30:
Hello, are you using pip version or version from git ?
hdlConvertor.to was named differently because of this it does not work (Or there is something wrong with packaging and hdlConvertor.to is not installed and I need to fix it if this is a case).

I made a binder (jupyter notebook running in docker with hdlConvertor installed)

https://mybinder.org/v2/gh/Nic30/hdlConvertor/master?filepath=notebooks%2F01_parse_and_dump.ipynb

There everything seems to be working there.

lionheart117, 2020-04-30:
> Hello, are you using pip version or version from git ?
> hdlConvertor.to was named differently because of this it does not work (Or there is something wrong with packaging and hdlConvertor.to is not [...]

I tried the both ways to install the lib@v1.5. But both of them seemed to fail and occur the same problem.

Nic30, 2020-04-30:
ok, I try it as well,

but maybe you do have installed the "old" version. (just reinstall does nothing as the version number is same) try https://stackoverflow.com/questions/19133831/python-how-to-force-overwriting-of-files-when-using-setup-py-install-distutil

lionheart117, 2020-04-30:
> ok, I'll try it as well,
> 
> but maybe you do have installed the "old" version. (just reinstall does nothing as the version number is same) try https://stackoverflow.com/questions/19133831/python-how-to-force-overwriting-of-files-when-using-setup-py-install-distutil

OK, I'll try so. Thank you😀

Nic30, 2020-04-30:
I just tried it and it works for me. It seems to me that you really have installed pip version.
(pip package will be updated in eta 2 weeks)

Maybe try look in to /home/$USER/.local/lib/python3.7/site-packages or https://stackoverflow.com/questions/122327/how-do-i-find-the-location-of-my-python-site-packages-directory to see what is actually installed.

lionheart117, 2020-04-30:
> I just tried it and it works for me. It seems to me that you really have installed pip version.
> (pip package will be updated in eta 2 weeks)
>
> Maybe try look in to [...]

Do you mean that the pip version would make this problem？
Should I use the git version instead？

Nic30, 2020-04-30:
yes, get git version (because it is better and because you are trying an example for a git version)

lionheart117, 2020-04-30:
> yes, get git version (because it is better and because you are trying an example for a git version)

Do I need to remove the pip version before installing the git version？

Nic30, 2020-04-30:
Because pip and git version do have same version number pip/setup.py does not reinstall the library automatically because it thinks that it is already installed. In order to avoid that use `pip uninstall hdlConvertor` and then install from git.

lionheart117, 2020-04-30:
> Because pip and git version do have same version number pip/setup.py does not reinstall the library automatically because it thinks that it is already installed. In order to avoid that use `pip uninstall hdlConvertor` and then install from git.

It works! Thanks😀

closed 2020-04-30 by lionheart117