Nic30/hdlConvertor #120: Where is the specific document of this lib?

lionheart117, 2020-04-30:
Hi, Nic
   This project is amazing! I hope that I could find more details about it. So I would like to know where I can find the specification about it?

Nic30, 2020-04-30:
Hello @lionheart117,

i am afraid that the old doc is not relevant to a current version. You can take look at the AST nodes which are used for representation. https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/_hdlConvertor.pyx

And for API of a main module https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/_hdlConvertor.pyx#L128 

Or some .to classes https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/to/verilog/verilog2005.py#L195

There was not much pressure on writing doc because it seems like if you do not know something you just execute the thing and you will see.
if you have something concrete which you are searching for in doc I would be glad for any input. 

I will start working on doc soon after merge of #115.
The original wiki always become deprecated very quickly and there was no one to update.
That's why I would like to use [jupyter notebooks](https://mybinder.org/v2/gh/Nic30/hdlConvertor/master?filepath=notebooks%2F01_parse_and_dump.ipynb) with tutorials rather than regular doc.

lionheart117, 2020-05-05:
> if you have something concrete which you are searching for in doc I would be glad for any input.

OK, if I get something concrete, I will make some input.

closed 2020-05-05 by lionheart117