Nic30/hdlConvertor #121: Do this lib support to parser module nestedly？

lionheart117, 2020-05-02:
Hi Nic,
    I have got this lib working after your help. But now I have a further question: does it support to parser a module nestedly as far as it reaches the deepest level of this module？

Thomasb81, 2020-05-02:
The tool read hdl source file and fill a structure available to user. 
Today there is, as far as I know, no elaboration that would allow user to access to a structure representing the hierarchy of the design.

Surlog has such feature.

lionheart117, 2020-05-02:
> The tool read hdl source file and fill a structure available to user.
> Today there is, as far as I know, no elaboration that would allow user to access to a structure representing the hierarchy of the design.
> 
> Surlog has such feature.

You mean such job need to be done by users themselves using this lib？ And the lib hasn't such feature now？ 

And what is the meaning of surlog？

Nic30, 2020-05-02:
https://github.com/alainmarcel/Surelog , they do have elaboration partially implemented, in hdlConvertor the elaboration is possible, but users do have to implement it themself, I do have some elaboration engine but I would like to tune it up a little bit before someone else relying on it.

I do have several tasks which I do have to do first however.

Nic30, 2020-05-02:
> does it support to parser a module nestedly until it reaches the deepest level of this module?

I am not sure what do you mean, yes instances in modules are parsed, but they are instances and original module is a different. The module is object accessible by name in top name scope. Maybe try to explain what you are trying to do.

lionheart117, 2020-05-02:
> I am not sure what do you mean, yes instances in modules are parsed, but they are instances and original module is a different. The module is object accessible by name in top name scope. Maybe try [...]

Your previous understanding is right. I would like to know whether I could use your lib to elaborate a design, not just parser a module.

Nic30, 2020-05-02:
Tricky thing about elaboration is that in most of the cases is required only to evaluate some expressions. However in order to do a full elaboration we need to implement all build-in functions, special properties, atributes, casts and datatypes.On top of that all the usual compiler stuff like dead code elimination, loop unroling and many others.

I mean the complete solution is full blown compiler, but in most of the cases simple script would do.

lionheart117, 2020-05-03:
> Tricky thing about elaboration is that in most of the cases is required only to evaluate some expressions. However in order to do a full elaboration we need to implement all build-in functions, [...]

If only elaborating a RTL design, is it much easier than what you said above？

lionheart117, 2020-05-03:
> I am not sure what do you mean, yes instances in modules are parsed, but they are instances and original module is a different. The module is object accessible by name in top name scope. Maybe try [...]

I think if this lib is gonna support elaborate, it would need to be modified in somewhere, for example, class HdlComponentInst perhaps needs to support to link its module definition.

Nic30, 2020-05-03:
The name symbol (HdlName) does support that 
https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/hdlAst/_structural.py#L87
https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/hdlAst/_expr.py#L28
And this is an example code which is discovering the module for component instance 
https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/translate/common/resolve_names.py#L76
https://github.com/Nic30/hdlConvertor/blob/f554854390ebea2ef71604a499be9a0dd28958ee/hdlConvertor/translate/verilog_to_basic_hdl_sim_model.py#L20

Nic30, 2020-05-11:
I hope the question was answered, let's have only issue #96 to talk about elaboration.

closed 2020-05-11 by Nic30

lionheart117, 2020-05-11:
> I hope the question was answered, let's have only issue #96 to talk about elaboration.

OK, if I have any question about elaboration, I will follow this issue.