Nic30/hdlConvertor #122: ModuleParser.module_item.generate_region Conversion to Python object not implemented

lionheart117, 2020-05-03:
`picorv32.v:200:1: ModuleParser.module_item.generate_region Conversion to Python object not implemented
    ...generateif(ENABLE_FAST_MUL)beginpicorv32_pcpi_fast_mulpcpi_mul(.clk(clk),.resetn(resetn),.pcpi_valid(pcpi_valid),.pcpi_insn(pcpi_insn),.pcpi_rs1(pcpi_rs1),.pcpi_rs2(pcpi_rs2),.pcpi_wr(pcpi_mul_wr),.pcpi_rd(pcpi_mul_rd),.pcpi_wait(pcpi_mul_wait),.pcpi_ready(pcpi_mul_ready));endelseif(ENABLE_MUL)beginpicorv32_pcpi_mulpcpi_mul(.clk(clk),.resetn(resetn),.pcpi_valid(pcpi_valid),.pcpi_insn(pcpi_insn),.pcpi_rs1(pcpi_rs1),.pcpi_rs2(pcpi_rs2),.pcpi_wr(pcpi_mul_wr),.pcpi_rd(pcpi_mul_rd),.pcpi_wait(pcpi_mul_wait),.pcpi_ready(pcpi_mul_ready));endelsebeginassignpcpi_mul_wr=0;assignpcpi_mul_rd=32'bx;assignpcpi_mul_wait=0;assignpcpi_mul_ready=0;endendgenerate...`

Hi Nic,
   I am trying to do some tests about the lib by some codes. For example, parsing this verilog code showing below:
 [https://github.com/cliffordwolf/picorv32/blob/master/picorv32.v](url)

   According to the feedback from the lib, how could I convert "**generate_region**" to a python object?

And there are much more info showing below, I don't know what's the meaning of these info:

`picorv32.v:1157:2: AttributeParser.visitAttribute_instance Conversion to Python object not implemented
    ...(*parallel_case,full_case*)...

picorv32.v:1174:2: AttributeParser.visitAttribute_instance Conversion to Python object not implemented
    ...(*parallel_case,full_case*)...

picorv32.v:1216:3: AttributeParser.visitAttribute_instance Conversion to Python object not implemented
    ...(*parallel_case*)...

picorv32.v:1938:33: VerExprParser.visitPackage_or_class_scoped_hier_id_with_select - operator_plus_minus Conversion to Python object not implemented    ...next_rd[j+:CARRY_CHAIN]...

picorv32.v:1939:15: VerExprParser::visitArray_range_expression - operator_plus_minus Conversion to Python object not implemented
    ...j+:CARRY_CHAIN...`

Nic30, 2020-05-04:
Hello @lionheart117 ,

now it should work properly (you need to pull and rebuild I also recommend to delete _skbuid build directory, just in case)

```python
    from os import path
    import sys
    from hdlConvertor import HdlConvertor
    from hdlConvertor.language import Language
    from hdlConvertor.to.verilog.verilog2005 import ToVerilog2005
 
    f = path.expanduser("~/Downloads/picorv32.v")
    c = HdlConvertor()
    res = c.parse([f, ], Language.SYSTEM_VERILOG, [], debug=True)
    tv = ToVerilog2005(sys.stdout)
    tv.visit_HdlContext(res)
```
You should see the same thing as is in attached picorv32.res.zip 
[picorv32.res.zip](https://github.com/Nic30/hdlConvertor/files/4574802/picorv32.res.zip)

For next time do you think you would be able to create a minimal example for a specified issue? I am asking because if you do (or better you directly commit the test) I (we) can use it directly as a test and it saves the time.

ace192f1afe67d00db63e85875091cf756f5e691

lionheart117, 2020-05-04:
> Hello @lionheart117 ,
> 
> now it should work properly (you need to pull and rebuild I also recommend to delete _skbuid build directory, just in case)
> 
> ```python
>     from os import path
>     import sys
>     from hdlConvertor import HdlConvertor
>     from hdlConvertor.language import Language
>     from hdlConvertor.to.verilog.verilog2005 import ToVerilog2005
>  
>     f = path.expanduser("~/Downloads/picorv32.v")
>     c = HdlConvertor()
>     res = c.parse([f, ], Language.SYSTEM_VERILOG, [], debug=True)
>     tv = ToVerilog2005(sys.stdout)
>     tv.visit_HdlContext(res)
> ```
> 
> You should see the same thing as is in attached picorv32.res.zip
> [picorv32.res.zip](https://github.com/Nic30/hdlConvertor/files/4574802/picorv32.res.zip)
> (note that I did not check the rest of file)
> 
> For next time do you think you would be able to create a minimal example for a specified issue? I am asking because if you do (or better you directly commit the test) I (we) can use it directly as a test and it saves the time.
> 
> [ace192f](https://github.com/Nic30/hdlConvertor/commit/ace192f1afe67d00db63e85875091cf756f5e691)

OK, next time I will make more targeted tests and commit it into the test folder directly.
But there are still some issues left behind, e.g.

```
picorv32.v:1157:2: AttributeParser.visitAttribute_instance Conversion to Python object not implemented
    ...(*parallel_case,full_case*)...

picorv32.v:1174:2: AttributeParser.visitAttribute_instance Conversion to Python object not implemented
    ...(*parallel_case,full_case*)...

picorv32.v:1216:3: AttributeParser.visitAttribute_instance Conversion to Python object not implemented
    ...(*parallel_case*)...

picorv32.v:1938:33: VerExprParser.visitPackage_or_class_scoped_hier_id_with_select - operator_plus_minus Conversion to Python object not implemented    ...next_rd[j+:CARRY_CHAIN]...

picorv32.v:1939:15: VerExprParser::visitArray_range_expression - operator_plus_minus Conversion to Python object not implemented
    ...j+:CARRY_CHAIN...
```

I don't know what's the meaning of these warnings ?

lionheart117, 2020-05-04:
> Hello @lionheart117 ,
>
> now it should work properly (you need to pull and rebuild I also recommend to delete _skbuid build directory, just in case)
>
> ```python
>     from os import path
>     import [...]

Hi Nic,
   It seems like that [ace192f] can only deal with "generate if" **but not "generate for",** 
e.g.  parsing the code showing below:

```
module top
  (
   input CLK, 
   input RST,
   input enable,
   input [31:0] value,
   output [7:0] led
  );
  reg [31:0] count;
  reg [7:0] state;
  assign led = count[23:16];

  wire[2:0] out;
  genvar i;

  generate
    for(i=0;i<3;i=i+1)
      assign  out[i]=i;
  endgenerate

  always @(posedge CLK) begin
    if(RST) begin
      count <= 0;
      state <= 0;
    end else begin
      if(state == 0) begin
        if(enable) state <= 1;
      end else if(state == 1) begin
        state <= 2;
      end else if(state == 2) begin
        count <= count + value;
        state <= 0;
      end
    end
  end
endmodule
```

Then the lib will feed back some warnings:

```
test.v:13:2: VerGenerateParser.visitModule_or_generate_item.genvar_declaration Conversion to Python object not implemented
    ...genvari;...
test.v:16:4: VerGenerateParser.visitModule_or_generate_item.loop_generate_construct Conversion to Python object not implemented
    ...for(i=0;i<3;i=i+1)assignout[i]=i;...
```

Nic30, 2020-05-04:
The warning means that the specified object is not in python objects received from conversions. This is the case for attribute instances which are parsed correctly on c++ level but they are not available in Python.

lionheart117, 2020-05-05:
Hi Nic,
Actually I am aware of the meaning of this type of warnings, but I don't know what these specific objects are.

And it seems like that ace192f can only deal with "generate if"， but not "generate for".

Nic30, 2020-05-06:
Hello @lionheart117 , the reason for missing generate for impl. is that just let not implemented message there instead of calling proper function https://github.com/Nic30/hdlConvertor/blob/master/src/svConvertor/generateParser.cpp#L357 

generate for loop is nearly same as normal for
https://github.com/Nic30/hdlConvertor/blob/master/grammars/sv2017Parser.g4#L2180

thus we already have a code to do the job.

However I am in quite hurry so I will look into it on 05:07 16:00+

lionheart117, 2020-05-07:
Hi Nic,
    As I couldn't find any objects describing generate_region in the python equivalent of the universal HDL AST, I would like to know wheather this feature have been included in the universal HDL AST？

Nic30, 2020-05-08:
Hello @lionheart117 ,
there is not an explicit generate_region object, instead each statement has `in_preproc` flag.
https://github.com/Nic30/hdlConvertor/blob/master/hdlConvertor/hdlAst/_bases.py#L59

for `generate if` HdlStmIf will have  `in_preproc=True`, note that the begin-end is `HdlStmBlock` etc.

lionheart117, 2020-05-08:
```
Extracting hdlConvertor-1.5-py3.6-linux-x86_64.egg to /usr/local/lib/python3.6/dist-packages
hdlConvertor 1.5 is already the active version in easy-install.pth

Installed /usr/local/lib/python3.6/dist-packages/hdlConvertor-1.5-py3.6-linux-x86_64.egg
Processing dependencies for hdlConvertor==1.5
Searching for hdlConvertorAst>=0.1
Reading https://pypi.org/simple/hdlConvertorAst/
Couldn't find index page for 'hdlConvertorAst' (maybe misspelled?)
```

compiling the latest commit will cause this error

Nic30, 2020-05-08:
https://github.com/Nic30/hdlConvertor/blob/master/README.md

```
sudo pip3 install git+https://github.com/Nic30/hdlConvertorAst.git
```

I am currently preparing hdlConvertorAst so it will get installed automatically, but it will take 4h or so.

lionheart117, 2020-05-09:
> https://github.com/Nic30/hdlConvertor/blob/master/README.md
> 
> ```
> sudo pip3 install git+https://github.com/Nic30/hdlConvertorAst.git
> ```
> 
> I am currently preparing hdlConvertorAst pip package so it will get installed automatically, but it will take 4h or so (until then you can install hdlConvertorAst from git using command mentioned).

It works, and then I'll do more test……

lionheart117, 2020-05-10:
Hi Nic
I found there were something wrong with generate-for when I add a label in a "begin" of generate-for.
But I don't know whether this error is caused by parser or code-gen. 
For this issue, I issue a pull request #125 , pls check

lionheart117, 2020-05-10:
> But I don't know whether this error is caused by parser or code-gen.

I guess it is due to the parser.
Because the result of the parser added an extra "in_preproc" in the begin-end blockstatement after I added the label. 
This would misguide the code-gen,

Nic30, 2020-05-11:
Seems to be working now

lionheart117, 2020-05-11:
I think a label of the 'begin' is a kind of mark for the begin-end-statement.
why do you link this to the GENERATE?

Nic30, 2020-05-11:
I think a label of the 'begin' is a kind of mark for the begin-end-statement as well.
Where do I link this to GENERATE?

lionheart117, 2020-05-11:
> I think a label of the 'begin' is a kind of mark for the begin-end-statement as well.
> Where do I link this to GENERATE?

maybe the last bug give me this false sense because of the extra GENERATE related to the label.

Nic30, 2020-08-15:
Hello, the generate regions are now converted, that is why I am closing this issue, if there is still some problem, please reopen it.

closed 2020-08-15 by Nic30