Nic30/hdlConvertor #123: New release 1.6 discussion.

Nic30, 2020-05-04:
Next release is very close, missing things:

* extract hdlConvertor.ast,to,translate as a separate library
  * in order to avoid C++ dependency for python based code generators
* examples/doc:
  * notebook which will show how specific object are represented in HdlAst
    * hdl input -> hdlConvertor json
  * notebook for translator api
    * verilog -> vhdl
    * vhdl -> verilog
    * optionaly
      * verilog -> pycocotb + example sim
      * verilog/vhdl function evaluation
      * verilog -> MyHdl + example sim
      * elaboration

This issue is a discussion about what should be done before release and about potential pull request. if you have any idea/critic do not be afraid to share.

lionheart117, 2020-05-05:
As far as I am concerned,  the following content is important:
1. notebook which will show how specific object are represented in HdlAst；
2. verilog/vhdl function evaluation；
3. elaboration；

Nic30, 2020-05-07:
2. Function evaluation has a problem, currently I do have only a library which does support at most 3 state values (0,1,x), but verilog needs 4 values (0,1,x,z) and vhdl 9 values (0, 1, X, Z, U, W, L, H, -).
3. needs 2

I will try to make some simple example. Looking at my current vhdl/verilog interpret, I hope I can create something understandable.

lionheart117, 2020-05-10:
> 1. Function evaluation has a problem, currently I do have only a library which does support at most 3 state values (`0,1,x`), but verilog needs 4 values (`0,1,x,z`) and vhdl 9 values (`0, 1, X, Z, U, W, L, H, -`).

@Nic30 is it related that the current lib doesn't support 'casez' and 'casex' ? 
Yesterday, I found the problem.

> 2. needs 2
> 
> I will try to make some simple example. Looking at my current vhdl/verilog interpret, I hope I can create something understandable.

Nic30, 2020-05-11:
@lionheart117 no it is related only to evaluation of expressions.
I also added the conversions for casex, casez.

lionheart117, 2020-05-11:
Hi @Nic30 
Currently, if I wanna try your latest commit, shall I need to go through the whole installation guidance completely? 
Or update some parts of the lib?

Nic30, 2020-05-11:
Depending on when you did update last time you do not need to update hdlConvertorAst.
```
sudo pip3 install git+https://github.com/Nic30/hdlConvertorAst.git --update
git pull
sudo python3 setup.py install -j$(nproc) --update
```
If some c++ files were added/deleted the cmake can not detect this correctly and it is requred to `rm _skbuild/ -rf`however this should not be the case now.

Nic30, 2020-05-11:
(note that if your build takes > 2min something is wrong)

closed 2020-06-19 by Nic30