Nic30/hdlConvertor #129: hdlConvertor._hdlConvertor.ParseException: test.v:90:1:Error: protect is not defined

lionheart117, 2020-05-14:
Hi @Nic30 
I find that the 'protect will cause the error of parser.

lionheart117, 2020-05-18:
Hi @Nic30，
  Recently I used your lib trying to parser a lot of commercial IP/project's RTL codes as a kind of test of ability about the lib. 

The result is amazing! It could almost parse  them all. Only a few codes can't be parsed correctly, and the problems can be  classified into several categories.

Nic30, 2020-06-19:
Hello @lionheart117 ,

1. can you send me an  example with protect? Do you mean `pragma protect` or `protected` ?
2. non utf-8 character probably, it is not hard to add support for different encoding but it is not done yet, I can write a detailed task with changes required for per file encoding specifications.
3. ah, you are right it needs to be added  into 
https://github.com/Nic30/hdlConvertor/blob/master/include/hdlConvertor/hdlAst/hdlOpType.h#L6 
and returned there https://github.com/Nic30/hdlConvertor/blob/4a26ca05e0e2080b91b0a1e426b8b0b10d1f465e/src/svConvertor/literalParser.cpp#L272
4. same type of problem as in 3.

lionheart117, 2020-06-19:
Could all these problems be solved in v1.6？

Nic30, 2020-06-19:
In v1.7 because v1.6 was released prematurely https://github.com/Nic30/hdlConvertor/tags (doc and examples are missing)
It seems to me as a 2 man-hours so yes.

Thomasb81, 2020-06-19:
>     2\. non utf-8 character probably, it is not hard to add support for different encoding but it is not done yet, I can write a detailed task with changes required for per file encoding specifications.

I think verilog,systemVerilog,VHDL must be ASCII. It should be an issue if utf-8 encoding is used in source code.

Thomas

Nic30, 2020-08-13:
Hello @lionheart117 ,

I fixed probably all issues you mentioned, (you need to reinstall with `--upgrade --force-reinstall --no-cache-dir` as described in readme, because the pip packages are not released yet)

 > 1.   'protect will cause parser error;

should not happen

>    Some codes will raise: ParseException:wstring_convert::from_bytes
    I guess there must be some characters in comment leading to this kind of error, but I haven't got a clue yet.

file encoding error (non utf-8)

>    Op"===" has not been implemented in python;

https://github.com/Nic30/hdlConvertorAst/blob/master/hdlConvertorAst/hdlAst/_expr.py#L162

IS, IS_NOT

>    Op"i*P +: P" has not been implemented in python;

PART_SELECT_PRE/POST

https://github.com/Nic30/hdlConvertorAst/blob/master/hdlConvertorAst/hdlAst/_expr.py#L127

Can you please check if it works for you?

lionheart117, 2020-10-06:
> Hello @lionheart117 ,
> 
> I fixed probably all issues you mentioned, (you need to reinstall with `--upgrade --force-reinstall --no-cache-dir` as described in readme, because the pip packages are not released yet)
> 
> > 1. 'protect will cause parser error;
> 
> should not happen

Hi, @Nic30 

Unfortunately，the "`protect" issue seems not to be solved yet in the newest version. 
I pull a request #139 for adding a test file, and you can test this file by yourself.

> 
> > Some codes will raise: ParseException:wstring_convert::from_bytes
> > I guess there must be some characters in comment leading to this kind of error, but I haven't got a clue yet.
> 
> file encoding error (non utf-8) (encoding now can be specified in parse methods)
> 
> > Op"===" has not been implemented in python;
> 
> https://github.com/Nic30/hdlConvertorAst/blob/master/hdlConvertorAst/hdlAst/_expr.py#L162
> 
> IS, IS_NOT
> 
> > Op"i*P +: P" has not been implemented in python;
> 
> PART_SELECT_PRE/POST
> 
> https://github.com/Nic30/hdlConvertorAst/blob/master/hdlConvertorAst/hdlAst/_expr.py#L127
> 
> Can you please check if it works for you?

Nic30, 2020-10-07:
@lionheart117 thank you for the example. 
Now I see, the protect/endprotect is a Verilog-XL extension
https://www.cs.utexas.edu/users/moore/acl2/manuals/current/manual/index-seo.php/VL____PREPROCESSOR

Now my question is:

Do you know the exact standard where this is defined?
Currently we do not need (but it would be nice to have) an extra support for this extension as it is possible just add empty defines like this:
```python
c = HdlConvertor()
c.preproc_macro_db.update({
    "protect": "",
    "endprotect": ""
})
```

Thomasb81, 2020-10-07:
This construction seems to be specific to verilog-XL.

Icarus has one testcase : https://github.com/steveicarus/ivtest/blob/master/ivltests/pr478.v

[Verilog-XL Compiler Directives](https://www.csee.umbc.edu/portal/help/VHDL/verilog/compiler_directives.html)
[List_of_HDL_simulators](https://en.wikipedia.org/wiki/List_of_HDL_simulators)

If it is just a matter of parsing existing code. We could just add an additional mode or extension (verilog-XL) to the preprocessor grammar to support the additional specifc verilog-XL preprocessor directive.

lionheart117, 2020-10-08:
@Nic30 @Thomasb81  I thought these directives come from IEEE standard. But now, it is just a matter of parsing existing code indeed.

Thomasb81, 2020-10-08:
>@Nic30 @Thomasb81 I thought these directives come from IEEE standard. But now, it is just a matter of parsing existing code indeed.

I check, It has never been in any ieee364 version.

lionheart117, 2020-10-09:
> > @Nic30 @Thomasb81 I thought these directives come from IEEE standard. But now, it is just a matter of parsing existing code indeed.
> 
> I check, It has never been in any ieee364 version.

yeah, you're right

Nic30, 2020-10-09:
Now there is a question if we should add a support for Verilog-XL somehow. It seems to me that it could be a dictionary of defines which user can add to macro define database before running the parser. The reason for doing it like this is that this extension was for v1995 but it seems to me that it was used also with different standards. That means that we can not simply add it as another version.

Could anyone prepare this define dictionary or somehow check what is really required and potentially prepare some examples?

Thomasb81, 2020-10-09:
I guess some tool have still this legacy support to manage at some point verilog-XL migration to more modern simulator.
But yes code compatibility can be achieve with pre-definition of protect and endprotect macro.

A command line helpper to preload the define map that would accept something line `-Dprotect -Dendprotect` can also do the job.

lionheart117, 2020-10-12:
> > @Nic30 @Thomasb81 I thought these directives come from IEEE standard. But now, it is just a matter of parsing existing code indeed.

@Nic30 @Thomasb81 At the beginning, I thought these directives would come from IEEE standard(but they don't). And now, it is just a matter of parsing existing code indeed.

Sorry for my incorrect expression in previous reply.

> 
> I check, It has never been in any ieee364 version.

Nic30, 2020-11-02:
OK, it seems to me that the issue is solved, so I am closing it, please reopen if you thing otherwise.

closed 2020-11-02 by Nic30