Nic30/hdlConvertor #132: BLIF support

mewais, 2020-07-27:
Hi, any plans for adding support for BLIF? for importing and/or exporting?

Nic30, 2020-07-30:
Hello,

do you need only parsing and export of BLIF or do you also need to convert arbitrary Verilog/VHDL to BLIF and back?
BLIF is a simple format, It is possible to add it in few days. 

Currently I am seeing some serious deadline walls from very close distance while having serious velocity. So even a small help would be appreciated. 

Can you help me to check if there is a BLIF grammar for ANTLR4?

There is also a branch with EDIF 4.0.0 support, however it currently depends on elaboration machinery, which is not public yet (it is not public, because it is just proof of concept yet).

------------------------------------------------------------------------------
https://ddd.fit.cvut.cz/prj/Circ_Gen/blif.pdf

Libraries of examples:
https://github.com/whereyoufindthis/blifparse/tree/master/blifs
https://github.com/florianjomrich/BlifParser/tree/master/blif

mewais, 2020-07-30:
Unfortunately, as far as I know there are no grammars for BLIF, neither in ANTLR4 nor in lex/flex and yacc/bison. There is a ton of BLIF parsers out there but none of them use grammars. I think the format is so simple that people just opt to parse it directly.

I understand there are some converters between BLIF and EDIF out there, so if there's already a branch for EDIF maybe that will be enough?!

PS: This was a gentle request/check, by no means should it load you up with more work when you have other deadlines to meet.

Nic30, 2020-07-30:
It really seems that BLIF is so trivial that there is no need to use any lexer/parser generator.
If I understand correctly things like Yosys can work with BLIF.  Can you briefly explain why you need it?

mewais, 2020-07-30:
I am trying to build a standalone tool that does FSM detection and optimization and boolean minimization. Originally, it is supposed to feed on Verilog and VHDL and dump back simpler VHDL or Verilog.

Unfortunately I found that some benchmarks are only available in BLIF, that's why I was checking.

Nic30, 2020-07-30:
_I just looking at summary in your resume and if I replace university name it is basically same as mine._

FSM detection and boolean function optimization is well studied topic, is there something to improve or you need it for something bigger?

I think that your library could be also useful for me.
(I am also working on compilers but for packet processing. Currently I just call espresso logic minimiser. I do not actually care much as I am just producing an input for synthesis tool.)

In 5 days I will have some time, I think it should be easy, but I can not promise anything. 
Anyway, if you like something from this project I can extract something to an extra library. Or you can always send a pull request with BLIF support etc., just tell me in advance.

mewais, 2020-07-30:
interesting :) Very similar indeed. I don't work on compilers any more though, I kinda switched topics. But I still have interest in CAD tools :)

That said, you're correct about FSM and boolean minimization being well studied, but they have never been revisited since the 90s. My logic is this is the first step in any synthesis flow, and if this one is optimized more then there's enhancements for all next steps of synthesis. And I think I might have a different idea for how to do boolean minimization that may or may not give better optimization. But until I test I can't know for sure. Either way, this is not my work, I'm doing this for fun, so even if it doesn't work or give better optimization I will have implemented something nice, which could lead to the start of something else that is also potentially fun.

I have a deadline in a month too, much later than your five days, but if by then it still requires work, I'm more than happy and willing to help.

Nic30, 2020-07-30:
Nice, it also could be nice to have some predictor which can tell if there is a high probability that the function can be greatly optimized. I mean it would be nice to do a logic minimisation in the steps of high-level synthesis if it is useful. Currently it is not possible to do logic minimisation as much as it is required because of time.