Nic30/hdlConvertor #136: Package Instantiation in VHDL

mewais, 2020-08-27:
Hi,

VHDL 2008 allows for instantiation of packages that have generics. For example:
```vhdl
library IEEE;

package fixed_pkg is new IEEE.fixed_generic_pkg
  generic map (
    fixed_round_style    => IEEE.fixed_float_types.fixed_round,
    fixed_overflow_style => IEEE.fixed_float_types.fixed_saturate,
    fixed_guard_bits     => 3,
    no_warning           => false
    );
```

This unfortunately does not get compiled correctly, or rather, does not get compiled at all. My AST for this file has only the following:
```json
"fixed_pkg.vhdl": [
    {
        "__class__": "HdlLibrary",
        "position": [
            43,
            9,
            43,
            12
        ],
        "name": {
            "__class__": "str",
            "val": "IEEE"
        }
    }
]
```

mewais, 2020-08-27:
If you want the original `fixed_generic_pkg` it is as follows:

```vhdl
-- --------------------------------------------------------------------
--
-- Copyright � 2008 by IEEE.
--
-- This source file is an essential part of IEEE Std 1076-2008, 
-- IEEE Standard VHDL Language Reference Manual. Verbatim copies of this 
-- source file may be used and distributed without restriction. 
-- Modifications to this source file as permitted in IEEE Std 1076-2008
-- may also be made and distributed. All other uses require permission 
-- from the IEEE Standards Department(stds-ipr@ieee.org). 
-- All other rights reserved.
--
-- This source file is provided on an AS IS basis. The IEEE disclaims ANY 
-- WARRANTY EXPRESS OR IMPLIED INCLUDING ANY WARRANTY OF MERCHANTABILITY 
-- AND FITNESS FOR USE FOR A PARTICULAR PURPOSE. The user of the source file 
-- shall indemnify and hold IEEE harmless from any damages or liability 
-- arising out of the use thereof.
--
--   Title     :  Fixed-point package (Generic package declaration)
--             :
--   Library   :  This package shall be compiled into a library
--             :  symbolically named IEEE.
--             :
--   Developers:  Accellera VHDL-TC and IEEE P1076 Working Group
--             :
--   Purpose   :  This packages defines basic binary fixed point
--             :  arithmetic functions
--             :
--   Note      :  This package may be modified to include additional data
--             :  required by tools, but it must in no way change the
--             :  external interfaces or simulation behavior of the
--             :  description. It is permissible to add comments and/or
--             :  attributes to the package declarations, but not to change
--             :  or delete any original lines of the package declaration.
--             :  The package body may be changed only in accordance with
--             :  the terms of Clause 16 of this standard.
--             :
-- --------------------------------------------------------------------
-- $Revision: 1220 $
-- $Date: 2008-04-10 17:16:09 +0930 (Thu, 10 Apr 2008) $
-- --------------------------------------------------------------------

library STD;
use STD.TEXTIO.all;

library IEEE;
use IEEE.STD_LOGIC_1164.all;
use IEEE.NUMERIC_STD.all;
use IEEE.fixed_float_types.all;

package fixed_generic_pkg is
  generic (
    -- Rounding routine to use in fixed point, fixed_round or fixed_truncate
    fixed_round_style    : fixed_round_style_type    := fixed_round;
    -- Overflow routine to use in fixed point, fixed_saturate or fixed_wrap
    fixed_overflow_style : fixed_overflow_style_type := fixed_saturate;
    -- Extra bits used in divide routines
    fixed_guard_bits     : NATURAL                   := 3;
    -- If TRUE, then turn off warnings on "X" propagation
    no_warning           : BOOLEAN                   := false
    );

  -- Author David Bishop (dbishop@vhdl.org)
  constant CopyRightNotice : STRING :=
    "Copyright 2008 by IEEE. All rights reserved.";

  -- base Unsigned fixed point type, downto direction assumed
[... 1351 lines cut ...]

  function from_hstring (
    hstring : STRING)                   -- hex string
    return UNRESOLVED_ufixed;
  alias from_hex_string is from_hstring [STRING return UNRESOLVED_ufixed];

  function from_string (
    bstring : STRING)                   -- binary string
    return UNRESOLVED_sfixed;
  alias from_bstring is from_string [STRING return UNRESOLVED_sfixed];
  alias from_binary_string is from_string [STRING return UNRESOLVED_sfixed];

  function from_ostring (
    ostring : STRING)                   -- Octal string
    return UNRESOLVED_sfixed;
  alias from_octal_string is from_ostring [STRING return UNRESOLVED_sfixed];

  function from_hstring (
    hstring : STRING)                   -- hex string
    return UNRESOLVED_sfixed;
  alias from_hex_string is from_hstring [STRING return UNRESOLVED_sfixed];

end package fixed_generic_pkg;

```

Nic30, 2020-08-29:
This will require some API changes.

1. Extract shared code from `HdlComponentInstance` (name, parameter map, ...) to `HdlTemplateInstance` and then create `HdlPackageInstance`
2.  Extract shared code from `HdlModuleDec` (name, parameter defs ...) to `HdlTemplate`, represent vhdl package/package body with `HdlTemplate` of `HdlClassDef` with [type](https://github.com/Nic30/hdlConvertorAst/blob/master/hdlConvertorAst/hdlAst/_typeDefs.py#L142) [package/package body](https://github.com/Nic30/hdlConvertorAst/blob/master/hdlConvertorAst/hdlAst/_typeDefs.py#L98) instead of [HdlValueIdspace](https://github.com/Nic30/hdlConvertorAst/blob/master/hdlConvertorAst/hdlAst/_structural.py#L27)
3. `HdlValueIdspace` -> `HdlNamespace` as now it represents only namespaces