Nic30/hdlConvertor #138: primary_unit_declaration representation in HDL AST

Nic30, 2020-09-02:
An example:
```vhdl
type DISTANCE is range 0 to 1E5
  units
    um;
    mm = 1000 um;
    in_a = 25400 um;
  end units;
```

Grammar:
```g4
physical_type_definition:
      range_constraint
          KW_UNITS
              primary_unit_declaration
              ( secondary_unit_declaration )*
          KW_END KW_UNITS ( identifier )?
;
primary_unit_declaration: identifier SEMI;
secondary_unit_declaration: identifier EQ physical_literal SEMI;
```

Currently there is not a direct representation of such a type. Physical type is basically enum with base type specified.
```verilog
enum bit[1:0] {RED, YELLOW, GREEN}         light_2; 
```
So the previous `DISTANCE` type example can be represented as:
```python3
HdlEnumDef(
   base=[HdlOp(RANGE, 0, 1e5), ],
   values=[HdlId("um"),
                HdlOp(MAP_ASSOCIATION, HdlId("mm"), HdlOp(MUL, 1000, HdlId("um"))),
                HdlOp(MAP_ASSOCIATION, HdlId("in_a"), HdlOp(MUL, 25400, HdlId("um")))
  ]
)
```

The question is if we should add some flag to explicitly note that this is a physical type definition or we can always recognize it by  `base` specification.
Another potential issue is that the unit specification in expression is currently represented as `*` maybe we need an extra operator for unit specification.

@mewais

mewais, 2020-09-02:
1. That works, I was thinking of creating a new data structure for it, `HdlPhysicalDef` which would have a vector of pairs for the names and values. This would have the advantage of having `"__class__": "HdlPhysicalDef"` in the ast, so no need for extra flags and whatnot. The choice is up to you though.
2. What's wrong with using `*` as the operator for representing units? That's exactly what I'd use. The first unit has an implicit `1 * `, everything else is multiples of it. This is what it naturally should be.

Nic30, 2020-09-02:
1. ok, let's make `HdlTypePhysicalDef` which will inherit from `HdlEnumDef` and use values to store unit definitions. I can possibly do this during the weekend, in the case you start work on this please notify me somehow, or push to your fork regularly so I can see.
 
2. if `1 ns` is represented as `1 * ns` it prevents from direct translation back to VHDL as `1*ns` will not be correct and without syntax. analysis we do not know that `ns` is unit id

mewais, 2020-09-02:
1. Sure, I will try to work on it today and/or tomorrow. If it works I'll always commit.
2. It implicitly has `1*` but that does not mean we have to include it, we can work around it easily by having a separate base type.

Nic30, 2020-09-06:
solved in #137

closed 2020-09-06 by Nic30