Nic30/hdlConvertor #144: Can the normal mode and the preprocessing mode SystemVerilog Grammar be combined into one file?

itviewer, 2020-12-11:
I want to use antl4 and your SystemVerilog grammar to make a simple SystemVerilog analysis tool, I need to combine the [preprocessing](https://github.com/Nic30/hdlConvertor/blob/master/grammars/verilogPreprocLexer.g4) and [normal](https://github.com/Nic30/hdlConvertor/blob/master/grammars/sv2017Lexer.g4) mode grammar together into one file, and then use antl4 to generate the lexical analyzer.

Is it possible(or how can it be done)? 
thanks.

Nic30, 2020-12-11:
Hello itviewer,

> Is it possible(or how can it be done)?

Yes, but do not do it because you do not need it. It would also significantly degrade performance of the parser.

In hdlConvertor the preprocessor is run in advance and the meta-information for later use (`__LINE__`, `timescale`) are passed from a result of preprocessor on c++ level.

It is done in this way, because it is less complicated.

You can not merge grammars just by using some antrl4 directive. You need to do it manually. Generally speaking it is possible but is probably 1week of work due preprocessor directives may appear on any place and you will need to dynamically rewrite the code. The performance hit for antlr4 will be so large that it is probably not worth it.

itviewer, 2020-12-11:
Here is a simple Verilog file [GCD.v.txt](https://github.com/Nic30/hdlConvertor/files/5678715/GCD.v.txt) ,which I need to convert to abstract syntax tree.
I am not familiar with antlr4. According to the documents, I need a lexer grammar and a parser grammar, and then generate the corresponding lexical analyzer and grammar analyzer (with JAVA), If the preprocessing grammar located in a stand alone file, what do I need to do to generate the correct parser(so as to distinguish statements such as `ifdef` `endif`)?

Nic30, 2020-12-11:
> I need to convert to abstract syntax tree. I am not familiar with antlr4. According to the documents, I need a lexer grammar and a parser grammar, and then generate the corresponding Lexer and Parser (with JAVA),

For example in C language, if you want to get AST you need to run preprocessor, then you get the code without preprocessor directives (`ifdef`, ....). Then you run the compiler, which have lexer/parser and generates AST which is then translated to binary ...

[image]

Similar for a Verilog you need to run preprocessor first to expand preprocessor directives.
You can use [HdlConverot.verilog_pp](https://github.com/Nic30/hdlConvertor/blob/master/tests/test_verilog_preproc.py#L21) or the one from [surelog](https://github.com/alainmarcel/Surelog) or any other preprocessor.

**Preprocessor is not just a antrl4 grammar** it is a program which performs the directive expansion so I recommend you to use something existing because there are plenty of obscure things in Verilog preprocessor.

Once you obtained the preprocessed verilog with all directives expanded you can now parse it in to AST.
You can use an existing antlr4 grammars to generate a lexer+parser (from lexer and parser grammar) and then you can use them to load a preprocessed verilog to a AST.

> I am not familiar with antlr4. 
(Lexer grammar is a grammar which is used to parse keywords, literals, etc, Parser grammar is using lexer grammar and contains the rules for more complicated code construct like statements etc.  It is separated for performance reasons.)

Note that antlr4 can generate lexer/parser for multiple languages not just for java.

hdlConvertor is a library which contains preprocessor and parser and AST obscurity remover.
You can use it to get Verilog AST in python or C++ without the need to go through the hell of the verilog deprecated features and legacy. 
 
> statements such as ifdef endif

 `ifdef`, `endif` is not a statement it is a compiler directive, of course you can use antlr grammars to generate lexer/parser but then you are still left with lots of work because you need to manually expand (replace with a string corresponding to that macro/directive) each appearance. Each directive does something different with some obscure behaviour different in each version of the verilog. Really if you can use something existing.

Nic30, 2020-12-11:
Note that in SV preprocessor is a part of the language, but that does not mean that it needs to be written in a single grammar and from performance and complexity reasons it is not preferred solution.

itviewer, 2020-12-11:
Thanks, I learned a lot!

closed 2020-12-11 by itviewer